raspberry-vanilla/android_local_manifest #1: build fail

goranstupar, 2022-12-17:
Hi android-13 build is failing

```
build/make/core/soong_config.mk:209: warning: BOARD_PLAT_PUBLIC_SEPOLICY_DIR has been deprecated. Use SYSTEM_EXT_PUBLIC_SEPOLICY_DIRS instead.
build/make/core/soong_config.mk:210: warning: BOARD_PLAT_PRIVATE_SEPOLICY_DIR has been deprecated. Use SYSTEM_EXT_PRIVATE_SEPOLICY_DIRS instead.
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=13
TARGET_PRODUCT=aosp_rpi4
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a72
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a72
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.15.0-56-generic-x86_64-Ubuntu-22.04.1-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=TQ1A.221205.012
OUT_DIR=out
PRODUCT_SOONG_NAMESPACES=device/brcm/rpi4 vendor/brcm/rpi4 external/v4l2_codec2
============================================
[100% 397/397] analyzing Android.bp files and generating ninja file at out/soong/build.ninja
FAILED: out/soong/build.ninja
cd "$(dirname "out/host/linux-x86/bin/soong_build")" && BUILDER="$PWD/$(basename "out/host/linux-x86/bin/soong_build")" && cd / && env -i  "$BUILDER"     --top "$TOP"     --soong_out "out/soong"     --out "out"     -o out/soong/build.ninja --globListDir build --globFile out/soong/globs-build.ninja -t -l out/.module_paths/Android.bp.list --available_env out/soong/soong.environment.available --used_env out/soong/soong.environment.used.build Android.bp
Killed
14:34:53 soong bootstrap failed with: exit status 1

#### failed to build some targets (03:10 (mm:ss)) ####
```

KonstaT, 2022-12-18:
There isn't actual error message so difficult to say. How much RAM do you have? You need at least 32GB (or 16GB and 16GB swap).

Google still recommends Ubuntu 18.04 for building AOSP. It looks like you're using 22.04 and there could be changes in the required packages you need to install for that version (see https://source.android.com/docs/setup/start/initializing#installing-required-packages-ubuntu-1804).

goranstupar, 2022-12-19:
You were right. It was a RAM issue.

I had 32GB of RAM and 2GB swap only. After increasing swap to 32GB, build is working properly.

btw, I haveno problem with building any other AOSP (10 of them) except this one on the same machine. Seems like Soong takes too many processes and almost 32GB of RAM for himself even with -j1. 

There might be some leak in build configuration.

KonstaT, 2022-12-19:
Something certainly changed with Android 13 as setting up Soong takes considerably longer than with previous Android versions and eats up a lot of RAM. I don't see how there could be anything in the Raspberry Pi specific configuration that would affect this.

closed 2022-12-19 by KonstaT