raspberry-vanilla/android_local_manifest #2: Compiler error

wianoski, 2022-12-26:
i tried to compile the source, when i ran `sudo lunch aosp_rpi4-userdebug` , the error showed like this, any idea how to solve this error ? thx
[image]

KonstaT, 2022-12-26:
You can solve this by following the instructions (https://github.com/raspberry-vanilla/android_local_manifest#how-to-build).

At no point it says you should run `lunch` with `sudo`. `lunch` is a macro of `build/envsetup.sh` which you need to source first.

closed 2022-12-26 by KonstaT

wianoski, 2022-12-26:
i've already following these steps, already in compile steps, i've run `. build/envsetup.sh` , and now i want to execute `lunch aosp_rpi4-userdebug`, but i got those errors, what should i do

KonstaT, 2022-12-26:
Have you tried uninstalling the `python-lunch` package you have installed for some reason?

wianoski, 2022-12-26:
i've tried it, i fix the libs instead, and now im facing new problem, that is when i tried execute `lunch aosp_rpi4-userdebut`, it showed that `No such file: aosp_rpi4-userdebug`, what did i miss ?

KonstaT, 2022-12-27:
Sorry, it's impossible for me to say what part of the instructions you have actually followed (and where you've improvised something). There isn't any "libs" that need any "fixing" if you've set up the build environment according to the instructions.

Everything in the build instructions (https://github.com/raspberry-vanilla/android_local_manifest#how-to-build) after initializing repo needs to executed from the root of Android source code. If you've synced the Raspberry Pi specific local manifest, there shouldn't be any thing missing.