raspberry-vanilla/android_local_manifest #5: "meson" is not allowed to be used.

candledragle, 2023-01-16:
I install meson command is 
pip install meson
not sudo snap install meson because is fail now

but I get the blow error

FAILED: build out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp Outputs: out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp Error: exited with code: 1 Command: /bin/bash -c "(rm -rf out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_LIBCAMERA//build ) && (cp -r external/libcamera//* out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (cd out/target/product/rpi4/obj/MESON_LIBCAMERA && PATH=/usr/bin:/usr/local/bin:\$PATH meson ./build --cross-file /mnt/pi4/out/target/product/rpi4/obj/MESON_LIBCAMERA_GEN/aosp_cross --buildtype=release -Dandroid=enabled -Dipas=raspberrypi -Dpipelines=raspberrypi -Dsysconfdir=/vendor/etc -Dtest=false ) && (PATH=/usr/bin:/bin:/sbin:\$PATH ninja -C out/target/product/rpi4/obj/MESON_LIBCAMERA/build ) && (touch out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp )" Output: "meson" is not allowed to be used. See https://android.googlesource.com/platform/build/+/master/Changes.md#PATH_Tools for more information.

candledragle, 2023-01-16:
my meson version is 1.0.0 is the latest

closed 2023-01-16 by candledragle

westman2000, 2023-01-25:
how you fixed it?