raspberry-vanilla/android_local_manifest #7: hwservice complaining about ffmpeg

dridri, 2023-02-11:
Hello, looking at logcat it seems that hwservice has some troubles using ffmpeg HAL : 
```
02-11 15:22:15.116   255   362 W HidlServiceManagement: Waited one second for android.hardware.media.c2@1.0::IComponentStore/ffmpeg
02-11 15:22:15.116   142   142 I hwservicemanager: Since android.hardware.media.c2@1.0::IComponentStore/ffmpeg is not registered, trying to start it as a lazy HAL.
02-11 15:22:15.117   255   362 I HidlServiceManagement: getService: Trying again for android.hardware.media.c2@1.0::IComponentStore/ffmpeg...
02-11 15:22:15.118   142  2538 W libc    : Unable to set property "ctl.interface_start" to "android.hardware.media.c2@1.0::IComponentStore/ffmpeg": error code: 0x20
02-11 15:22:15.118   142  2538 I hwservicemanager: Tried to start android.hardware.media.c2@1.0::IComponentStore/ffmpeg as a lazy service, but was unable to. Usually this happens when a service is not installed, but if the service is intended to be used as a lazy service, then it may be configured incorrectly.
01-01 00:06:18.004     0     0 E init    : Control message: Could not find 'android.hardware.media.c2@1.0::IComponentStore/ffmpeg' for ctl.interface_start from pid: 142 (/system/bin/hwservicemanager)
```

I can't check if video playback works as I didn't find a way to install Gapps yet to get Chrome, and Firefox just can't open any webpage

dridri, 2023-02-13:
Full logcat from boot to SystemUI :
https://gist.github.com/dridri/edcef23babffe90c3d620a955f81fe9c

KonstaT, 2023-02-13:
Have you synced the source code after ffmpeg projects were added to the local manifest? Are you using the latest local manifest in the first place? Did you make a clean build (i.e. make installclean) if you've built before? What does your /vendor/etc/vintf/manifest.xml looks like?

There's isn't/shouldn't be any android.hardware.media.c2@1.0::IComponentStore/ffmpeg to begin with, it's android.hardware.media.c2@1.2::IComponentStore/ffmpeg.

dridri, 2023-02-13:
Maybe I did or missed something in the wrong order, I'm going to retry with clean directory

dridri, 2023-02-13:
Problem still persists, these are the exact commands I used to build my image :
```
mkdir android_cm4
cd android_cm4
repo init -u https://android.googlesource.com/platform/manifest -b android-13.0.0_r30 --depth=1
curl --create-dirs -L -o .repo/local_manifests/manifest_brcm_rpi4.xml -O -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-13.0/manifest_brcm_rpi4.xml
curl --create-dirs -L -o .repo/local_manifests/remove_projects.xml -O -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-13.0/remove_projects.xml
repo sync -j32
. build/envsetup.sh
lunch aosp_rpi4-userdebug
make bootimage systemimage vendorimage -j32
./rpi4-mkimg.sh
sudo dd if= of=/dev/mmcblk0 bs=1M status=progress
```
Then add this to /vendor/etc/init/hw/init.rpi4.rc to enable adb on boot :
```
on property:dev.bootcomplete=1
    setprop ro.adb.security 0
    setprop service.adb.tcp.port 5555
    stop adbd
    start adbd
```

KonstaT, 2023-02-15:
Ok, build steps look fine if that was a fresh setup. If you've already synced and built the image, do `make installclean` before rebuilding. Unrelated, but you would probably like add those to https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-13.0/ramdisk/init.rpi4.rc before building the image.

I still don't know you can end up with this, though. What does your /vendor/etc/vintf/manifest.xml look like in the image you've built? How about /vendor/etc/vintf/manifest/android.hardware.media.c2@*.xml?

dridri, 2023-02-15:
Here is the manifest.xml
```xml
<!--
    Input:
        manifest.xml
-->
<manifest version="5.0" type="device" target-level="7">
    <hal format="hidl">
        <name>android.hardware.audio</name>
        <transport>hwbinder</transport>
        <version>7.1</version>
        <interface>
            <name>IDevicesFactory</name>
            <instance>default</instance>
        </interface>
        <fqname>@7.1::IDevicesFactory/default</fqname>
    </hal>
    <hal format="hidl">
        <name>android.hardware.audio.effect</name>
        <transport>hwbinder</transport>
        <version>7.0</version>
        <interface>
            <name>IEffectsFactory</name>
            <instance>default</instance>
        </interface>
        <fqname>@7.0::IEffectsFactory/default</fqname>
    </hal>
    <hal format="hidl">
        <name>android.hardware.bluetooth</name>
        <transport>hwbinder</transport>
        <version>1.0</version>
        <interface>
            <name>IBluetoothHci</name>
            <instance>default</instance>
        </interface>
        <fqname>@1.0::IBluetoothHci/default</fqname>
    </hal>
    <hal format="hidl">
        <name>android.hardware.camera.provider</name>
        <transport>hwbinder</transport>
        <version>2.5</version>
        <interface>
            <name>ICameraProvider</name>
            <instance>external/0</instance>
            <instance>legacy/0</instance>
        </interface>
        <fqname>@2.5::ICameraProvider/external/0</fqname>
        <fqname>@2.5::ICameraProvider/legacy/0</fqname>
    </hal>
    <hal format="hidl">
        <name>android.hardware.graphics.composer</name>
        <transport>hwbinder</transport>
        <version>2.4</version>
        <interface>
            <name>IComposer</name>
            <instance>default</instance>
        </interface>
        <fqname>@2.4::IComposer/default</fqname>
    </hal>
    <hal format="hidl">
        <name>android.hardware.media.omx</name>
        <transport>hwbinder</transport>
        <version>1.0</version>
        <interface>
            <name>IOmx</name>
            <instance>default</instance>
        </interface>
        <interface>
            <name>IOmxStore</name>
            <instance>default</instance>
        </interface>
        <fqname>@1.0::IOmx/default</fqname>
        <fqname>@1.0::IOmxStore/default</fqname>
    </hal>
    <sepolicy>
        <version>33.0</version>
    </sepolicy>
</manifest>
```

android.hardware.media.c2@1.0-service-v4l2.xml :
```xml
<manifest version="1.0" type="device">
    <hal format="hidl">
        <name>android.hardware.media.c2</name>
        <transport>hwbinder</transport>
        <fqname>@1.0::IComponentStore/v4l2</fqname>
    </hal>
</manifest>
```

android.hardware.media.c2@1.2-service-ffmpeg.xml :
```xml
<?xml version="1.0" encoding="utf-8"?>
<!-- Copyright [....] -->
<manifest version="1.0" type="device">
    <hal format="hidl">
        <name>android.hardware.media.c2</name>
        <transport>hwbinder</transport>
        <fqname>@1.2::IComponentStore/ffmpeg</fqname>
    </hal>
</manifest>
```

Everything seems good indeed..
Did just took a look at android.hardware.light@2.0-service.rpi.xml to compare and the version is explicitly set : 
```xml
<manifest version="1.0" type="device">
    <hal format="hidl">
        <name>android.hardware.light</name>
        <transport>hwbinder</transport>
        <version>2.0</version>
        <interface>
            <name>ILight</name>
            <instance>default</instance>
        </interface>
    </hal>
</manifest>
```
I'm going to try adding this to service-ffmpeg but don't have much hope

dpetrecki, 2023-02-20:
I have a similar issue. I've just cloned the repo as documented in README, changed the resolution and density as documented on wiki. Everything works, but the system complains in logs. In the previous build (before adding ffmpeg) the problem didn't occure.
[rpi4.log](https://github.com/raspberry-vanilla/android_local_manifest/files/10783813/rpi4.log)

KonstaT, 2023-02-21:
Fixed, please sync your sources.

closed 2023-02-21 by KonstaT