raspberry-vanilla/android_local_manifest #12: Wifi HAL and driver mismatch?

zhangyyun, 2023-03-13:
Don't know if this is the right place to ask, but I hope someone might give some light.

I have a raspberrypi 4b and I built aosp following the instructions here. Everything is going on well, until I watched the output of adb logcat. below is some fragment:
```
02-28 13:22:55.236   228   228 I WifiHAL : Creating message to get link statistics; iface = 8
02-28 13:22:55.237   228   228 D WifiHAL : error code -95 mapped to -3
02-28 13:22:55.239   411   537 E WifiVendorHal: getWifiLinkLayerStats_1_6_Internal(l.1199) failed {.code = ERROR_NOT_SUPPORTED, .description = }
02-28 13:22:58.250   228   228 I WifiHAL : Creating message to get link statistics; iface = 8
02-28 13:22:58.250   228   228 D WifiHAL : error code -95 mapped to -3
02-28 13:22:58.251   411   537 E WifiVendorHal: getWifiLinkLayerStats_1_6_Internal(l.1199) failed {.code = ERROR_NOT_SUPPORTED, .description = }
02-28 13:23:01.258   228   228 I WifiHAL : Creating message to get link statistics; iface = 8
02-28 13:23:01.259   228   228 D WifiHAL : error code -95 mapped to -3
```

It's so frequent that I can't just ignore it, and, after some digging around, it seems there is a mismatch. 
The source code of broadcom wifi hal is located at hardware/broadcom/wlan, a subdir bcmdhd exists there.
But the actual ko is brcmfmac in high version of android kernel, so netlink not supported reported.
[HAL](https://android.googlesource.com/platform/hardware/broadcom/wlan/+/refs/tags/android-13.0.0_r30/bcmdhd/wifi_hal/link_layer_stats.cpp#84)
[kernel bcmdhd-3.10 branch](https://android.googlesource.com/kernel/common.git/+/bcmdhd-3.10/drivers/net/wireless/bcmdhd/wl_cfgvendor.c#2406)

After some googling I still can't figure out why.
As brcmfmac is maintained in the upstream kernel, I think the right way is switching the Wifi HAL implement, but so many APIs, so I just wonder if google has a plan on it or not.
Help appreciated, thanks.

zhangyyun, 2023-03-13:
Thank you very much for the quick answer. But this is indeed a problem, no?

I know little about Android staffs, still I can't find the retrying logic of this log, would you mind giving more details if I can ignore them(or even remove them before build?).

KonstaT, 2023-03-14:
I don't know if this causes any actual issue. 

You can test if the previously linked AOSP change that implements link layer stats fixes this.

KonstaT, 2024-01-09:
Closing as this no longer seems to appear on AOSP14. Also in previous Android versions this likely didn't cause any actual issue other than log spam.

closed 2024-01-09 by KonstaT