raspberry-vanilla/android_local_manifest #16: Issue when compiling libdav1d for x86_64 Android

hmtheboy154, 2023-05-30:
Hi, I am currently using your dav1d fork and building it on Bliss OS 15.8 (based on Android 12L). I've cloned it into `external/dav1d` and on ffmpeg I have the same changes like this commit 
https://github.com/raspberry-vanilla/android_external_ffmpeg/commit/3a9de769738258c5667fd947e8580a3cdb0ed946#diff-0a1d48f2510d8d97f1a5d3e10ad77060a70080d247b280fec26bc4ec4735e2d8R31

But when I start to build, it gave me this error :  
```
FAILED: ninja: 'media/new-ej-drive/hmtheboy154/bliss15/out/target/product/x86_64/obj/MESON_LIBDAV1D/install/usr/local/lib/libdav1d.so', needed by '/media/new-ej-drive/hmtheboy154/bliss15/out/target/product/x86_64/obj/SHARED_LIBRARIES/libdav1d_intermediates/LINKED/libdav1d.so', missing and no known rule to make it
```

Is there anything wrong with my above setup that cause this error ?

KonstaT, 2023-06-12:
I haven't tested to build for x86_64 so I'm not sure.

Double check that dav1d is enabled properly on the FFmpeg side (check the generated configs). There could be something arm specific on the dav1d Android makefile as well (e.g. https://github.com/raspberry-vanilla/android_external_dav1d/blob/android-13.0/android/meson_cross.mk#L67).

KonstaT, 2023-07-12:
Closing as this is not directly related to AOSP on Raspberry Pi.

closed 2023-07-12 by KonstaT