raspberry-vanilla/android_local_manifest #17: failed to boot for "kvm: exiting hardware virtualization"

leelwh, 2023-06-06:
I sync the android and kernel code, build and use rpi4-mkimg.sh to generate ”RaspberryVanillaAOSP13-20230605-rpi4.img“,
but failed to boot for the reason "kvm: exiting hardware virtualization", and I can boot successfully with "AOSP13-20230412-KonstaKANG-rpi4.img" on my rpi4b device

reboot log:
[    4.981167] init: Wait for partitions returned after 916ms
[    4.988036] init: DSU not detected, proceeding with normal boot
[    4.995012] init: [libfs_mgr]superblock s_max_mnt_count:65535,/dev/block/sda2
[    5.006097] EXT4-fs (sda2): mounted filesystem without journal. Opts: barrier=1. Quota mode: none.
[    5.015352] init: [libfs_mgr]__mount(source=/dev/block/sda2,target=/system,type=ext4)=0: Success
[    5.048633] kvm: exiting hardware virtualization
[    5.068571] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[    5.090449] reboot: Restarting system with command 'bootloader'

full log:
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.104-v8-g5e9686786425 (build-user@build-host) (Android (8508608, based on r450784e) clang version 14.0.7 (https://android.googlesource.com/toolchain/llvm-project 4c603efb0cca074e9238af8b4106c30add4418f6), LLD 14.0.7) #1 SMP PREEMPT Fri Apr 28 09:04:50 UTC 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b2fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000007fffffff]
[    0.000000] On node 0, zone DMA32: 19712 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s80216 r8192 d30376 u118784
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496692
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:640x480M@60 smsc95xx.macaddr=DC:A6:32:C0:05:6C vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=ttyS0,115200 no_console_suspend root=/dev/ram0 rootwait androidboot.hardware=rpi4 androidboot.selinux=permissive androidboot.serialno=1000000042152e73 androidboot.btmacaddr=11:22:42:15:2e:73
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037300000-0x000000003b300000] (64MB)
[    0.000000] Memory: 1354412K/2018304K available (16768K kernel code, 2142K rwdata, 7204K rodata, 1536K init, 1365K bss, 139604K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 49224 entries in 193 pages
[    0.000000] ftrace: allocated 193 pages with 3 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] irq_brcmstb_l2: registered L2 intc (/soc/interrupt-controller@7ef00100, parent irq: 10)
[    0.000000] kfence: initialized - using 2097152 bytes for 255 objects at 0x(____ptrval____)-0x(____ptrval____)
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000321] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000358] pid_max: default: 32768 minimum: 301
[    0.000507] LSM: Security Framework initializing
[... 368 lines cut ...]
[    4.174144] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    4.248585] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[    4.367144] usb 1-1.4: New USB device found, idVendor=046d, idProduct=c52f, bcdDevice=30.00
[    4.375734] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.383196] usb 1-1.4: Product: USB Receiver
[    4.387548] usb 1-1.4: Manufacturer: Logitech
[    4.403992] logitech-djreceiver 0003:046D:C52F.0002: hidraw1: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:01:00.0-1.4/input0
[    4.483403] logitech-djreceiver 0003:046D:C52F.0003: hiddev97,hidraw2: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:01:00.0-1.4/input1
[    4.557811] logitech-djreceiver 0003:046D:C52F.0003: device of type eQUAD step 4 DJ (0x04) connected on slot 1
[    4.579174] input: Logitech B330/M330/M331 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.1/0003:046D:C52F.0003/0003:046D:4057.0004/input/input2
[    4.597553] logitech-hidpp-device 0003:046D:4057.0004: input,hidraw3: USB HID v1.11 Mouse [Logitech B330/M330/M331] on usb-0000:01:00.0-1.4/input1:1
[    4.933428] scsi 0:0:0:0: Direct-Access     SanDisk  Extreme Pro      0    PQ: 0 ANSI: 6
[    4.943645] sd 0:0:0:0: [sda] 250085376 512-byte logical blocks: (128 GB/119 GiB)
[    4.951968] sd 0:0:0:0: [sda] Write Protect is off
[    4.957601] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    4.970132]  sda: sda1 sda2 sda3 sda4
[    4.976691] sd 0:0:0:0: [sda] Attached SCSI removable disk
[    4.981167] init: Wait for partitions returned after 916ms
[    4.988036] init: DSU not detected, proceeding with normal boot
[    4.995012] init: [libfs_mgr]superblock s_max_mnt_count:65535,/dev/block/sda2
[    5.006097] EXT4-fs (sda2): mounted filesystem without journal. Opts: barrier=1. Quota mode: none.
[    5.015352] init: [libfs_mgr]__mount(source=/dev/block/sda2,target=/system,type=ext4)=0: Success
[    5.048633] kvm: exiting hardware virtualization
[    5.068571] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[    5.090449] reboot: Restarting system with command 'bootloader'

Karthi-Coder-18, 2023-06-08:
Hi leelwh, how did you collected the logs from the raspberry pi which is running on Android. 

I have also flashed the android 12 & 13 images, I am able to boot , but I was stuck at " Android is Starting "  screen. 
I am not sure how to collect logs as raspberry pi is running on Android & UI is not working 

Could you please elaborate on how you were able to collect the logs

[boot_issue_android12.webm](https://github.com/raspberry-vanilla/android_local_manifest/assets/119874162/3e43d2b7-b336-4f59-aacf-5467d5b32c4c)
 

Thank you!!

leelwh, 2023-06-09:
> Hi leelwh, how did you collected the logs from the raspberry pi which is running on Android.
>
> I have also flashed the android 12 & 13 images, I am able to boot , but I was stuck at " Android is [...]

there are many guides on how to use serial console to access raspberry pi, this may help: https://tp4348.medium.com/serial-to-raspberry-pi-da635122b4d0

KonstaT, 2023-06-12:
Looks like you're trying to boot from USB device (`[ 4.970132] sda: sda1 sda2 sda3 sda4`). Please see the relevant wiki section (https://github.com/raspberry-vanilla/android_local_manifest/wiki/USB-boot).

@Karthi-Coder-18
Is that Android Automotive? Raspberry Vanilla doesn't support automotive build variant so you would need ask elsewhere (I'm aware there's at least couple of Android Automotive variants forked from Raspberry Vanilla).