raspberry-vanilla/android_local_manifest #22: Android 13 -  boot loop: audioserver could not be started

imilk23, 2023-08-08:
I used Raspberry Pi 4 Model B Rev 1.2
My device can not boot and retry to start at audioserver.  Please help

```
[   11.567044] healthd: battery none chg=
[   11.791562] type=1400 audit(11.783:5): avc: denied { read } for comm="android.hardwar" name="u:object_r:default_prop:s0" dev="tmpfs" ino=99 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=1
[   11.813693] type=1400 audit(11.783:6): avc: denied { open } for comm="android.hardwar" path="/dev/__properties__/u:object_r:default_prop:s0" dev="tmpfs" ino=99 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=1
[   11.837280] type=1400 audit(11.783:7): avc: denied { getattr } for comm="android.hardwar" path="/dev/__properties__/u:object_r:default_prop:s0" dev="tmpfs" ino=99 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=1
[   11.861220] type=1400 audit(11.783:8): avc: denied { map } for comm="android.hardwar" path="/dev/__properties__/u:object_r:default_prop:s0" dev="tmpfs" ino=99 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:default_prop:s0 tclass=file permissive=1
[   11.884848] type=1400 audit(11.783:9): avc: denied { read } for comm="android.hardwar" name="u:object_r:hdmi_config_prop:s0" dev="tmpfs" ino=152 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:hdmi_config_prop:s0 tclass=file permissive=1
[   11.907364] type=1400 audit(11.783:10): avc: denied { open } for comm="android.hardwar" path="/dev/__properties__/u:object_r:hdmi_config_prop:s0" dev="tmpfs" ino=152 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:hdmi_config_prop:s0 tclass=file permissive=1
[   11.931442] type=1400 audit(11.783:11): avc: denied { getattr } for comm="android.hardwar" path="/dev/__properties__/u:object_r:hdmi_config_prop:s0" dev="tmpfs" ino=152 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:hdmi_config_prop:s0 tclass=file permissive=1
[   11.956058] type=1400 audit(11.783:12): avc: denied { map } for comm="android.hardwar" path="/dev/__properties__/u:object_r:hdmi_config_prop:s0" dev="tmpfs" ino=152 scontext=u:r:hal_tv_cec_default:s0 tcontext=u:object_r:hdmi_config_prop:s0 tclass=file permissive=1
[   12.882871] read descriptors
[   12.885878] read strings
[   13.329667] dwc2 fe980000.usb: bound driver configfs-gadget
[   16.534184] logd: logdr: UID=1000 GID=1003 PID=363 n tail=0 logMask=8 pid=241 start=0ns deadline=0ns
[   16.544987] logd: logdr: UID=1000 GID=1003 PID=363 n tail=0 logMask=1 pid=241 start=0ns deadline=0ns
[   16.593960] init: Untracked pid 363 exited with status 0
[   16.599487] init: Untracked pid 363 did not have an associated service entry and will not be reaped
[   16.600996] binder: release 241:241 transaction 1149 in, still active
[   16.609132] init: Service 'surfaceflinger' (pid 241) received signal 6
[   16.615281] binder: release 241:241 transaction 1067 out, still active
[   16.615296] binder: send failed reply for transaction 1149 to 229:229
[   16.622117] init: Sending signal 9 to service 'surfaceflinger' (pid 241) process group...
[   16.644655] libprocessgroup: Successfully killed process cgroup uid 1000 pid 241 in 0ms
[   16.653634] init: Sending signal 9 to service 'zygote' (pid 215) process group...
[   16.672270] libprocessgroup: Successfully killed process cgroup uid 0 pid 215 in 10ms
[   16.682978] init: Service 'zygote' (pid 215) received signal 9
[   16.700742] init: starting service 'exec 18 (/system/bin/vdc volume abort_fuse)'...
[   16.714952] init: Command 'write /sys/power/state on' action=onrestart (<Service 'zygote' onrestart>:2) took 0ms and failed: Unable to write to file '/sys/power/state': Unable to write file contents: Invalid argument
[   16.829061] binder: 229:229 transaction failed 29189/0, size 4-0 line 3122
[   16.836175] binder: send failed reply for transaction 1067, target dead
[   17.109151] type=1400 audit(17.103:35): avc: denied { read } for comm="composer@2.4-se" path="socket:[13245]" dev="sockfs" ino=13245 scontext=u:r:hal_graphics_composer_default:s0 tcontext=u:r:hal_graphics_composer_default:s0 tclass=netlink_kobject_uevent_socket permissive=1
[   17.194676] type=1400 audit(17.187:36): avc: denied { read } for comm="binder:379_2" name="u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   17.217821] type=1400 audit(17.211:37): avc: denied { open } for comm="binder:379_2" path="/dev/__properties__/u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   17.241221] type=1400 audit(17.211:38): avc: denied { getattr } for comm="binder:379_2" path="/dev/__properties__/u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   17.264450] type=1400 audit(17.211:39): avc: denied { map } for comm="binder:379_2" path="/dev/__properties__/u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   17.743004] logd: logdr: UID=1000 GID=1003 PID=437 n tail=0 logMask=8 pid=378 start=0ns deadline=0ns
[   17.753606] logd: logdr: UID=1000 GID=1003 PID=437 n tail=0 logMask=1 pid=378 start=0ns deadline=0ns
[   17.817756] binder: release 378:378 transaction 1845 in, still active
[   17.824369] binder: release 378:378 transaction 1568 out, still active
[   17.832197] binder: send failed reply for transaction 1845 to 229:229
[   18.040155] binder: 229:229 transaction failed 29189/0, size 4-0 line 3122
[   18.047266] binder: send failed reply for transaction 1568, target dead
[   21.837764] init: starting service 'zygote'...
[   21.844104] init: Created socket '/dev/socket/zygote', mode 660, user 0, group 1000
[   21.852492] init: Created socket '/dev/socket/usap_pool_primary', mode 660, user 0, group 1000
[   21.867716] init: starting service 'zygote_secondary'...
[   21.873838] init: Created socket '/dev/socket/zygote_secondary', mode 660, user 0, group 1000
[   21.883067] init: Created socket '/dev/socket/usap_pool_secondary', mode 660, user 0, group 1000
[   21.898242] init: starting service 'audioserver'...
[   21.909708] init: starting service 'netd'...
[   21.915457] init: Created socket '/dev/socket/dnsproxyd', mode 660, user 0, group 3003
[   21.924374] init: Created socket '/dev/socket/mdns', mode 660, user 0, group 1000
[   22.155601] type=1400 audit(22.147:40): avc: denied { read } for comm="composer@2.4-se" path="socket:[13245]" dev="sockfs" ino=13245 scontext=u:r:hal_graphics_composer_default:s0 tcontext=u:r:hal_graphics_composer_default:s0 tclass=netlink_kobject_uevent_socket permissive=1
[   22.270854] type=1400 audit(22.263:41): avc: denied { read } for comm="binder:457_2" name="u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   22.292545] type=1400 audit(22.263:42): avc: denied { open } for comm="binder:457_2" path="/dev/__properties__/u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   22.315880] type=1400 audit(22.263:43): avc: denied { getattr } for comm="binder:457_2" path="/dev/__properties__/u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   22.342081] type=1400 audit(22.263:44): avc: denied { map } for comm="binder:457_2" path="/dev/__properties__/u:object_r:audio_prop:s0" dev="tmpfs" ino=37 scontext=u:r:hal_audio_default:s0 tcontext=u:object_r:audio_prop:s0 tclass=file permissive=1
[   22.801334] logd: logdr: UID=1000 GID=1003 PID=517 n tail=0 logMask=8 pid=456 start=0ns deadline=0ns
[   22.811811] logd: logdr: UID=1000 GID=1003 PID=517 n tail=0 logMask=1 pid=456 start=0ns deadline=0ns
[   22.869905] binder: release 456:456 transaction 2555 in, still active
[   22.876528] binder: release 456:456 transaction 2328 out, still active
[   22.883735] binder: send failed reply for transaction 2555 to 229:229
[   23.090779] binder: 229:229 transaction failed 29189/0, size 4-0 line 3122
[   23.097885] binder: send failed reply for transaction 2328, target dead
[   26.869169] init: starting service 'zygote'...
[... cut ...]
[   27.897311] binder: release 531:531 transaction 2860 out, still active
[   27.904146] binder: send failed reply for transaction 3038 to 229:229
[   28.111227] binder: 229:229 transaction failed 29189/0, size 4-0 line 3122
[   28.118287] binder: send failed reply for transaction 2860, target dead

```

Full log:
[MobaXterm_COM16ProlificUSB-to-SerialCommPortCOM16_20230808_091622.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/12286619/MobaXterm_COM16ProlificUSB-to-SerialCommPortCOM16_20230808_091622.txt)

KonstaT, 2023-08-08:
Logcat?

imilk23, 2023-08-08:
> Logcat?

Thank @KonstaT

Please help me investigate  [logcat.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/12287919/logcat.txt).

KonstaT, 2023-08-08:
It's a display issue (`Abort message: 'HWC returned no active mode'`), nothing to do with audioserver.

As mentioned in https://github.com/raspberry-vanilla/android_local_manifest/wiki/HDMI-display you need to use a display that reports resolutions it supports via EDID. Or something else wrong with your display setup.

rsglobal, 2023-08-09:
It may happen when the display (HDMI) is connected but not powered up.

KonstaT, 2023-08-23:
Closing as assuming resolved with proper display setup.

closed 2023-08-23 by KonstaT