raspberry-vanilla/android_local_manifest #23: Camera/ffmpeg support in automotive build

dpetrecki, 2023-08-22:
Hello,
the standard image from https://konstakang.com/devices/rpi4/AOSP13/ contains ffmpeg and v4l2-ctl and basically the USB-connected camera works.
But the build aosp_rpi4_car-userdebug doesn't include those two and I can't read from the camera.
My camera is UVC-compatible Logitech c920.
I've tested it with EvsCameraPreview and OpenCamera (https://opencamera.org.uk/)
This is all I've added to the build (aosp_rpi4_car.mk):
```
PRODUCT_SYSTEM_DEFAULT_PROPERTIES += \
    config.override_forced_orient=true \
    persist.bluetooth.enablenewavrcp=false

DEVICE_PACKAGE_OVERLAYS += device/google/cuttlefish/shared/auto/overlay
ENABLE_EVS_SERVICE ?= true
ENABLE_MOCK_EVSHAL ?= true
ENABLE_CAREVSSERVICE_SAMPLE ?= true
ENABLE_SAMPLE_EVS_APP ?= true
ENABLE_CARTELEMETRY_SERVICE ?= true
ENABLE_EVS_SAMPLE  := false
TARGET_BOARD_INFO_FILE ?= device/google/cuttlefish/shared/auto/android-info.txt

CUSTOMIZE_EVS_SERVICE_PARAMETER := true
PRODUCT_COPY_FILES += \
    device/google/cuttlefish/shared/auto/evs/init.evs.rc:$(TARGET_COPY_OUT_VENDOR)/etc/init/init.evs.rc \
    device/google/cuttlefish/shared/auto/evs/evs_app_config.json:$(TARGET_COPY_OUT_SYSTEM)/etc/automotive/evs/config_override.json

BOARD_SEPOLICY_DIRS += device/google/cuttlefish/shared/auto/sepolicy/evs \
    packages/services/Car/cpp/evs/apps/sepolicy/private \
    packages/services/Car/cpp/evs/apps/sepolicy/cartelemetry

PRODUCT_PACKAGES += \
    CarEvsCameraPreviewAppRRO \
    com.google.android.car.evs \
    EvsHalDefaults \
    CarEvsCameraPreviewApp \
    android.hardware.automotive.evs@1.1-service \
    android.frameworks.automotive.display@1.0-service \
    android.hardware.automotive.evs-aidl-default-service \
    android.automotive.evs.manager@1.1 \
    android.hardware.automotive.evs@1.1-sample \
    librs_jni \
    cameraserver \
    evsmanagerd \
    android.hardware.automotive.evs@1.1-sample \
    evs_app \
    evs_app_support_lib \
    cardisplayproxyd 
    
PRODUCT_PACKAGE_OVERLAYS += packages/services/Car/tests/SampleRearViewCamera/overlay    

PRODUCT_COPY_FILES += \
    frameworks/base/data/sounds/effects/ogg/camera_click_48k.ogg:$(TARGET_COPY_OUT_PRODUCT)/media/audio/ui/camera_click.ogg \
    frameworks/native/data/etc/tablet_core_hardware.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/tablet_core_hardware.xml \
    device/google/cuttlefish/shared/auto/display_settings.xml:$(TARGET_COPY_OUT_VENDOR)/etc/display_settings.xml

PRODUCT_PACKAGE_OVERLAYS += \
    device/generic/car/common/overlay
```
and the overlay:
```
    <string translatable="false" name="config_evsRearviewCameraId">/dev/video0</string>
    <string name="config_evsCameraActivity" translatable="false">
        com.google.android.car.evs/com.google.android.car.evs.CarEvsCameraPreviewActivity
    </string>
```
The camera is connected as /dev/video0 and /dev/video1 (i don't know, why is it twice, but on the non-automotive build v4l2-ctl shows no stream on /dev/video1) :
```
# for I in /sys/class/video4linux/*; do cat $I/name; done
Logitech Webcam C930e
Logitech Webcam C930e
bcm2835-codec-decode
[... 216 lines cut ...]
2023-08-21 12:49:33.538   467-2784  CoreBackPreview         system_process                       D  Window{65e8de5 u0 com.google.android.car.evs/com.google.android.car.evs.CarEvsCameraPreviewActivity}: Setting back callback OnBackInvokedCallbackInfo{mCallback=android.window.IOnBackInvokedCallback$Stub$Proxy@6ea0b6b, mPriority=0}
2023-08-21 12:49:33.552  1649-1649  FocusParkingView        com.android.car.carlauncher          D  The window of Activity [com.android.car.carlauncher.AppGridActivity@1398f7d] gained focus
2023-08-21 12:49:33.553  1649-1649  FocusAreaHelper         com.android.car.carlauncher          D  The window of Activity [com.android.car.carlauncher.AppGridActivity@1398f7d] gained focus
2023-08-21 12:49:33.554  1649-1649  FocusAreaHelper         com.android.car.carlauncher          D  The window of Activity [com.android.car.carlauncher.AppGridActivity@1398f7d] gained focus
2023-08-21 12:49:33.555   747-747   CarSystemBarController  com.android.systemui                 D  refreshSystemBar: locked?: false homeDisabled: false notificationDisabled: false qcDisabled: false systemIconsDisabled: false
2023-08-21 12:49:33.559  3021-3037  OpenGLRenderer          com.google.android.car.evs           E  Unable to match the desired swap behavior.
2023-08-21 12:49:33.568   691-691   CarPackage...gerService com.android.car                      D  onWindowChange event received
2023-08-21 12:49:33.570   691-691   CarPackage...gerService com.android.car                      D  onWindowChange event received
2023-08-21 12:49:33.581   467-866   CoreBackPreview         system_process                       D  Window{a2cc880 u10 Splash Screen com.google.android.car.evs EXITING}: Setting back callback null
2023-08-21 12:49:33.585   467-866   InputManager-JNI        system_process                       W  Input channel object 'a2cc880 Splash Screen com.google.android.car.evs (client)' was disposed without first being removed with the input manager!
2023-08-21 12:49:33.599   292-382   TransactionTracing      surfaceflinger                       W  Could not find layer handle 0xb400007d40401db0
2023-08-21 12:49:33.600   292-382   TransactionTracing      surfaceflinger                       W  Could not find layer handle 0xb400007d403fcd10
2023-08-21 12:49:33.612  1649-1649  FocusParkingView        com.android.car.carlauncher          D  The window of Activity [com.android.car.carlauncher.AppGridActivity@1398f7d] lost focus
2023-08-21 12:49:33.613  1649-1649  FocusAreaHelper         com.android.car.carlauncher          D  The window of Activity [com.android.car.carlauncher.AppGridActivity@1398f7d] lost focus
2023-08-21 12:49:33.613  1649-1649  FocusAreaHelper         com.android.car.carlauncher          D  The window of Activity [com.android.car.carlauncher.AppGridActivity@1398f7d] lost focus
2023-08-21 12:49:33.635   467-484   ActivityTaskManager     system_process                       I  Displayed com.google.android.car.evs/.CarEvsCameraPreviewActivity: +189ms
2023-08-21 12:49:33.647   467-866   InputMetho...gerService system_process                       W  A background user is requesting window. Hiding IME.
2023-08-21 12:49:33.647   467-866   InputMetho...gerService system_process                       W  If you need to impersonate a foreground user/profile from a background user, use EditorInfo.targetInputMethodUser with INTERACT_ACROSS_USERS_FULL permission.
2023-08-21 12:49:33.775   292-382   TransactionTracing      surfaceflinger                       W  Could not find layer id -1
2023-08-21 12:49:33.776   292-382   TransactionTracing      surfaceflinger                       W  Could not find layer id -1
2023-08-21 12:49:35.181   292-382   TransactionTracing      surfaceflinger                       W  Could not find layer id -1
2023-08-21 12:49:35.181   292-382   TransactionTracing      surfaceflinger                       W  Could not find layer id -1
```

Any idea, how to use the camera in the automotive build?

KonstaT, 2023-08-22:
FFmpeg is included in Raspberry Vanilla Automotive builds and it does what it's intended (decoding H.264/H.265/AV1/etc videos).

v4l-utils are debugging tools and are not included in Raspberry Vanilla builds by default. v4l2-ctl is not needed for camera to function. You could easily add it to your build if you need it for debugging.
Add https://github.com/lineage-rpi/android_external_v4l-utils/tree/lineage-20.0 to the device manifest and add the tool you want to build in device.mk, e.g.:
```
PRODUCT_PACKAGES += \
    v4l2-ctl
```

Raspberry Vanilla includes two camera HALs.
libcamera for CSI camera modules:
https://github.com/raspberry-vanilla/android_external_libcamera/tree/android-13.0
https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-13.0/device.mk#L98-L118

Android external camera HAL for UVC webcams:
https://source.android.com/docs/core/camera/external-usb-cameras
https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-13.0/device.mk#L89-L96

Both work on Automotive builds as well given you've enabled the default camera service (`ENABLE_CAMERA_SERVICE := true`).
https://android.googlesource.com/platform/packages/services/Car/+/refs/tags/android-13.0.0_r67/car_product/build/car_base.mk#63

I haven't looked into the EVS HAL or if it's even supposed to work with UVC webcams. Again, reading https://android.googlesource.com/platform/packages/services/Car/+/refs/tags/android-13.0.0_r67/car_product/build/car_base.mk, this doesn't seem to be correct way to enable EVS. You would simply need to add:
```
ENABLE_EVS_SERVICE := true
CUSTOMIZE_EVS_SERVICE_PARAMETER := true #not sure if needed
ENABLE_EVS_SAMPLE := true
ENABLE_CAREVSSERVICE_SAMPLE := true
ENABLE_REAR_VIEW_CAMERA_SAMPLE := true
```

Error message is very clear at least. Could be e.g. permissions issue (https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-13.0/ramdisk/ueventd.rpi4.rc#L7) if it's present otherwise.
```
2023-08-21 12:49:33.517   691-748   CarServiceJNI           com.android.car                      E  /dev/video0 is not available
2023-08-21 12:49:33.517   691-748   CarServiceJNI           com.android.car                      E  Failed to open a camera device
2023-08-21 12:49:33.517   691-748   CAR.EVS                 com.android.car                      E  Failed to open a target camera device
```

KonstaT, 2023-08-22:
EVS HAL and the sample EVS camera preview app seem to work with USB UVC webcam. Image is a bit stretched but I'm not sure if that's intentional. CSI camera modules are not supported with EVS.

Please sync your sources.

[image]

closed 2023-08-22 by KonstaT

dpetrecki, 2023-08-23:
You're awesome! 😍 I've spent a week investigating this issue.