raspberry-vanilla/android_local_manifest #25: Compilation error

usatenko, 2023-09-18:
When compiling aosp_rpi4_car I get this error (the reduced size option).
```
repo init -u https://android.googlesource.com/platform/manifest -b android-13.0.0_r67 --depth=1
curl --create-dirs -L -o .repo/local_manifests/manifest_brcm_rpi4.xml -O -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-13.0/manifest_brcm_rpi4.xml
curl --create-dirs -L -o .repo/local_manifests/remove_projects.xml -O -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-13.0/remove_projects.xml

```

Can you please direct me to what the problem may be?

```
List 'out/target/product/rpi4/system/product/etc/vintf/': NAME_NOT_FOUND
The following HALs in device manifest are not declared in FCM <= level 7:
  android.hardware.media.c2@1.0::IComponentStore/v4l2
  android.hardware.media.c2@1.2::IComponentStore/ffmpeg
ERROR: files are incompatible: Device manifest and framework compatibility matrix are incompatible: HALs incompatible. Matrix level = 7. Manifest level = 7. The following requirements are not met:
android.hardware.thermal:
    required: @2.0::IThermal/default
    provided:

The following instances are in the device manifest but not specified in framework compatibility matrix:
    android.hardware.media.c2@1.0::IComponentStore/v4l2
    android.hardware.media.c2@1.2::IComponentStore/ffmpeg
Suggested fix:
1. Update deprecated HALs to the latest version.
2. Check for any typos in device manifest or framework compatibility matrices with FCM version >= 7.
3. For new platform HALs, add them to any framework compatibility matrix with FCM version >= 7 where applicable.
4. For device-specific HALs, add to DEVICE_FRAMEWORK_COMPATIBILITY_MATRIX_FILE or DEVICE_PRODUCT_COMPATIBILITY_MATRIX_FILE.: Success
INCOMPATIBLE
```

usatenko, 2023-09-18:
[check_vintf_compatible.log](https://github.com/raspberry-vanilla/android_local_manifest/files/12643997/check_vintf_compatible.log)

usatenko, 2023-09-18:
I suspect it fails after you removed this
[image]

KonstaT, 2023-09-18:
Please follow the steps in the instruction on how to compile i.e.
```
make bootimage systemimage vendorimage -j$(nproc)
```

Passing VINTF compatibility tests are currently not in the scope but you can of course work towards this if this is requirement for you for some reason (pull requests are welcome). You would at least need to add `DEVICE_FRAMEWORK_COMPATIBILITY_MATRIX_FILE` for
`android.hardware.media.c2` and implement thermal HAL (there's a mock HAL in https://android.googlesource.com/platform/hardware/interfaces/+/refs/tags/android-13.0.0_r67/thermal/2.0/default/) as suggested in the check_vintf_compatible.log. I know there's also some kernel options I've enabled that won't pass so you would need to work through those as well. You can use `make check-vintf-all` to check.

hardware/interfaces was previously forked because broadcom vendor bluetooth stack needed two external patches on Pi 4. These patches are no longer needed since I've switched to using btlinux bluetooth stack.

usatenko, 2023-09-18:
thank you very much, now I see what I was missing

closed 2023-09-18 by usatenko

usatenko, 2023-09-18:
@KonstaT BTW, is there any way to run the build in some emulation without writing to a flash card? This would speed up the dev.
I saw qemu has raspi4 support in some of dev commits, but they are not yet merged to master ...
Please advice.

KonstaT, 2023-09-19:
What dev commits are you referring to exactly? I could have a look.

I'm not aware of any way you could run the Pi 4 image in any emulator. You can of course build AOSP for the Android emulator (https://source.android.com/docs/setup/create/avd).

usatenko, 2023-09-19:
This one.
https://github.com/patchew-project/qemu/releases/tag/patchew%2F20230726132512.149618-1-sergey.kambalin%40auriga.com

I tried to apply the commits from this branch to qemu master (to some commit close to 20230726 by date) branch and after building qemu I was able to run raspbian in qemu as raspi4b8g machine type.
However, I was not able to run it in graphic mode.

```
./build/qemu-system-aarch64 -M raspi4b8g -serial stdio -kernel ../android_device_brcm_rpi4-kernel/Image -append "rw earlyprintk loglevel=8 dwc_otg.lpm_enable=0 root=/dev/mmcblk1p2 rootdelay=1" -dtb ../android_device_brcm_rpi4-kernel/bcm2711-rpi-4-b.dtb -device virtio-keyboard-pci -device virtio-mouse-pci -drive id=mydrive,if=none,format=raw,bus=0,index=0,file=/Users/ou/Downloads/2023-05-03-raspios-bullseye-arm64-lite.img -display cocoa -device sd-card,drive=mydrive
```