raspberry-vanilla/android_local_manifest #28: EvsCameraPreview is streaming black video always

SarbeenaS, 2023-10-13:
As per this ticket https://github.com/raspberry-vanilla/android_device_brcm_rpi4/commit/d2d71de3513e3e985538e834e4ebb42084d4921b  have done changes and I could see Camera icon  and EvsCameraPreview icon got added. But upon clicking EvsCameraPreview , app is opening, but black video is getting streamed as attached here
[image]

Only one USB camera device was attached to the Raspberry Board. Logcat is pointing out issue in GraphicBufferAllocator as:
 
MockEvsDriver: startVideoStream
10-13 07:01:02.100   170   170 I MockEvsDriver: Allocating 1 buffers for camera frames
10-13 07:01:02.101   282   303 E android.hardware.graphics.allocator@4.0-service.minigbm_gbm_mesa: Unsupported combination -- pixel format: RGBA_8888, drm format:DRM_FOURCC_AB24, usage: CPU_READ_NEVER | CPU_READ_RARELY | CPU_WRITE_NEVER | CPU_WRITE_RARELY | GPU_TEXTURE | CAMERA_OUTPUT (0x20122)
10-13 07:01:02.101   170   170 E GraphicBufferAllocator: Failed to allocate (1280 x 720) layerCount 1 format 1 usage 20122: 7
10-13 07:01:02.101   170   170 E MockEvsDriver: Error 7 allocating 1280 x 720 graphics buffer
10-13 07:01:02.101   170   170 E MockEvsDriver: Rolling back to previous frame queue size
10-13 07:01:02.101   170   170 E MockEvsDriver: Failed to start stream because we couldn't get a graphics buffer
10-13 07:01:02.101   694   744 E CarServiceJNI: StreamHandler failed to start a video stream: 5
10-13 07:01:02.101   694   744 E CAR.EVS : Failed to start a video stream
10-13 07:01:02.102   694   744 E CAR.EVS : Transition failed: error = -1
10-13 07:01:02.102  2613  2613 E CarEvsCameraPreviewActivity: Failed to start a video stream, error = -1 . 
Detailed log is attached here:
[log_evscaprevirew2.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/12891544/log_evscaprevirew2.txt)

Attached are the evs_config_override.json copied to /system/etc/automotive/evs/ path. 
evs_aidl_hal_configuration.xml copied to /vendor/etc/automotive/evs path 
evs_default_configuration.xml copied to /vendor/etc/automotive/evs path
[evs_config_override.json](https://github.com/raspberry-vanilla/android_local_manifest/files/12891662/evs_config_override.json)
[evs_default_configuration.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/12891830/evs_default_configuration.txt)
[evs_aidl_hal_configuration.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/12891831/evs_aidl_hal_configuration.txt)

Please provide inputs to proceed further.

KonstaT, 2023-10-13:
This issue tracker is meant for issues with the Raspberry Vanilla project. If you are building something else, sorry, this is not the right place to discuss it. Otherwise, I don't know what it is, I don't know what it does.

Raspberry Vanilla's Android Automotive build target on android-13.0 branch supports the EVS camera HAL. This doesn't require you to make any modifications, just simply following the build instructions in https://github.com/raspberry-vanilla/android_local_manifest/tree/android-13.0.

I still haven't gotten around getting everything sorted for Android Automotive build target on android-14.0 branch. EVS camera HAL also now works on android-14.0 branch after https://github.com/raspberry-vanilla/android_device_brcm_rpi4/commit/58de1fd5f1778d292a89d9cbebeefe9073acf055. So, also simply following the instructions in https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0 has this working (though there seems to be some issue at least with the CarLauncher).

As mentioned earlier EVS camera HAL only works with USB UVC webcams. CSI camera modules can be used with regular camera apps such as the built-in Camera application.

closed 2023-10-13 by KonstaT

SarbeenaS, 2023-10-13:
Hi Konsta, 
We are building Raspberry Vanilla project only taken from this https://github.com/grapeup/aaos_local_manifest .
Here we are working on EVS part, and the expectation is  EvsCameraPreview App should get added on Raspberry Vanilla Emulator and upon clicking video/camera feed should work fine. But the present progress have achieved black video coming upon clicking the EvsCameraPreview App. 
So please provide input to proceed further.

KonstaT, 2023-10-13:
It's not Raspberry Vanilla, it's an outdated fork. You need to ask whoever is maintaining that fork or switch to using the plain Raspberry Vanilla sources if you want my help.

SarbeenaS, 2023-10-26:
Hi Konsta, 
We switched to https://github.com/raspberry-vanilla/android_local_manifest/tree/android-13.0 branch and successfully created OS image and flashed to Raspberry Board. From there opened EvsCameraPreview App , and it launched with camera output from USB camera connected on the board as attached EvsCameraAppPreview.jpg - 
[image]

But while launching evs_app, facing segmentation fault. Below is the logs captured after connecting adb to the device.

|rpi4:/ # logcat | grep "EvsApp"
01-01 00:00:16.062   172   172 I EvsApp  : Acquiring EVS Display
01-01 00:00:16.065   172   172 I EvsApp  : Connecting to Vehicle HAL
01-01 00:00:16.070   172   172 I EvsApp  : Constructing state controller
01-01 00:00:16.072   172   172 I EvsApp  : Camera list callback received 6cameras.
01-01 00:00:16.072   172   172 W EvsApp  : No config information for hardware camera /dev/video22
01-01 00:00:16.072   172   172 W EvsApp  : No config information for hardware camera /dev/video21
01-01 00:00:16.072   172   172 W EvsApp  : No config information for hardware camera /dev/video15
01-01 00:00:16.072   172   172 W EvsApp  : No config information for hardware camera /dev/video14
01-01 00:00:16.072   172   172 W EvsApp  : No config information for hardware camera group0
01-01 00:00:16.072   172   172 I EvsApp  : Entering running state
01-01 00:00:16.542   172   442 I EvsApp  : Activated state 4
10-26 12:05:22.128  2599  2599 I EvsApp  : EVS app starting
10-26 12:05:22.130  2599  2599 I EvsApp  : Acquiring EVS Enumerator
10-26 12:05:22.132  2599  2599 I EvsApp  : Acquiring EVS Display
10-26 12:05:22.157  2599  2599 I EvsApp  : Connecting to Vehicle HAL
10-26 12:05:22.166  2599  2599 I EvsApp  : Constructing state controller
10-26 12:05:22.168  2599  2599 I EvsApp  : Camera list callback received 6cameras.
10-26 12:05:22.168  2599  2599 W EvsApp  : No config information for hardware camera /dev/video22
10-26 12:05:22.168  2599  2599 W EvsApp  : No config information for hardware camera /dev/video21
10-26 12:05:22.168  2599  2599 W EvsApp  : No config information for hardware camera /dev/video15
10-26 12:05:22.168  2599  2599 W EvsApp  : No config information for hardware camera group0
10-26 12:05:22.168  2599  2599 W EvsApp  : No config information for hardware camera /dev/video14
10-26 12:05:22.169  2599  2599 I EvsApp  : Entering running state
10-26 12:05:22.224  2599  2603 I EvsApp  : Activated state 4
10-26 12:08:15.425  2633  2633 I EvsApp  : EVS app starting
10-26 12:08:15.426  2633  2633 I EvsApp  : Acquiring EVS Enumerator
10-26 12:08:15.428  2633  2633 I EvsApp  : Acquiring EVS Display
10-26 12:08:15.429  2633  2633 I EvsApp  : Connecting to Vehicle HAL
10-26 12:08:15.433  2633  2633 I EvsApp  : Constructing state controller
10-26 12:08:15.454  2633  2633 I EvsApp  : Camera list callback received 6cameras.
10-26 12:08:15.454  2633  2633 W EvsApp  : No config information for hardware camera /dev/video22
10-26 12:08:15.454  2633  2633 W EvsApp  : No config information for hardware camera /dev/video21
10-26 12:08:15.454  2633  2633 W EvsApp  : No config information for hardware camera /dev/video15
10-26 12:08:15.454  2633  2633 W EvsApp  : No config information for hardware camera group0
10-26 12:08:15.454  2633  2633 W EvsApp  : No config information for hardware camera /dev/video14
10-26 12:08:15.457  2633  2633 I EvsApp  : Entering running state
10-26 12:08:15.552  2633  2637 I EvsApp  : Activated state 4
10-26 12:09:03.553  2633  2633 W EvsApp  : evs_app is being terminated on receiving a signal 2
10-26 12:09:03.555  2633  2637 W EvsApp  : EvsStateControl update loop ending
10-26 12:09:03.555  2633  2637 E EvsApp  : Shutting down app due to state control loop ending

Is these camera details should be filled to configuration files and to be uploaded to /system/etc/automotive/evs/ path ? And do  config_override.json do need any other changes ? 
Please provide your valuable inputs like how to resolve this issue.

Thanks in advance,
Sarbeena