raspberry-vanilla/android_local_manifest #30: Support for reTerminal display

aanrudolph2, 2023-10-21:
I'm not sure how to add in additional driver sources here, in this case for a reTerminal display and peripherals (https://wiki.seeedstudio.com/reTerminal/). SeeedStudio provides a DKMS module for Raspberry Pi OS, which suggests this can be added  -- see https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/master/overlays/rpi.

The benefit is that this is a fully feature device with an accelerometer, light sensor, and buttons that can be utilized in Android OS.

KonstaT, 2023-10-25:
I'm not aware of any easy solution for building out-of-tree kernel modules for Android.

Since you can't compile the kernel on the Android device, you need to cross-compile the Android kernel on some other host machine. You could add the specific drivers to the kernel source but you'd probably also need to add relevant Kconfig and Makefile entries. There's kernel build instructions available here.
https://github.com/raspberry-vanilla/android_kernel_manifest

Just by adding a kernel driver for e.g. accelerometer or light sensor doesn't utilize it in Android. You would need to add Android sensor HAL for that purpose. Raspberry Vanilla doesn't include support for such external hardware but you find the source code for IIO sensor HAL used in LineageOS/personal AOSP builds here.
https://github.com/lineage-rpi/android_hardware_intel_sensors-iio

Here is some discussion related to adding additional sensor support.
https://github.com/lineage-rpi/android_kernel_brcm_rpi/issues/28
http://disq.us/p/2w873fx

KonstaT, 2023-11-24:
Closing as there has been no activity on this issue for a month.

Goal of the Raspberry Vanilla project is to provide support for the built-in hardware on Raspberry Pi.

closed 2023-11-24 by KonstaT