raspberry-vanilla/android_local_manifest #32: Cannot checkout raspberry-vanilla/android_build

sandeepcg, 2023-10-27:
I'm facing below error while repo sync .
`Cannot checkout raspberry-vanilla/android_build: GitCommandError: git command failure`

How to fix above error

KonstaT, 2023-10-27:
By reading/posting the complete output that shows what the actual error is.

Have you followed the instructions on https://github.com/raspberry-vanilla/android_local_manifest? What branch are you syncing?

sandeepcg, 2023-10-27:
I'm syncing "android-14.0.0_r1 ".
Yeah.Same instructions are following

KonstaT, 2023-10-27:
Yet it will work just fine if you've followed those instructions in the right order.

I can't possibly know if you've done something else. And I also can't know the exact issue because you're not posting the complete output.

closed 2023-10-28 by sandeepcg