raspberry-vanilla/android_local_manifest #33: Bluetooth issues on aosp 14

SuperPou1, 2023-10-27:
When I pair my phone, it starts rapidly connecting and disconnecting. On the pi bluetooth actually turns off and on. Sometimes when I re-pair my phone, it doesn't do that, but aosp doesn't say my phone is connected, and on my phone when I try to play music for example, it instantly stops. I also tried other phones with same results.

My phone: Samsung galaxy a52s 5g, android 13 oneui 5.1
Rpi4: 2gb

Did anyone also encounter this issue?

SuperPou1, 2023-10-28:
okay, so i think i found the cause of the issue. disabling text messaging access in bluetooth device settings fixes the problem, and i can play music as normal.

KonstaT, 2023-10-30:
This issue report is missing all kinds of important information. Please edit the issue to include appropriate logs (`logcat`) and information about the build target.

Bluetooth has several issues on Android Automotive builds, which I have to assume this issue report is about. I'm actually very surprised if A2DP sink works and you can play music from another device (which you naturally can't between e.g. two Android phones).

Nols1000, 2024-01-06:
Hi I can confirm, that after denying text-message access, audio-playback via bluetooth works on my build of the `aosp_rpi4_car-userdebug` target. Currently its only working with the 3.5mm audio jack. I'll look into getting the HiFiBerry Amp2 working. 

I'll look for the logs of the disconnecting and reconnecting issue later.

[image]
[image]

adufftpc, 2024-03-25:
I can propose disabling MAP as a workaround. It allows to avoid cyclic reconnections. https://github.com/adufftpc/android_device_brcm_rpi4/pull/1

adufftpc, 2024-03-25:
I can propose to disable BT MAP client to avoid cyclic reconnections. https://github.com/raspberry-vanilla/android_device_brcm_rpi4/pull/2

KonstaT, 2024-03-25:
> I can propose to disable BT MAP client to avoid cyclic reconnections. [raspberry-vanilla/android_device_brcm_rpi4#2](https://github.com/raspberry-vanilla/android_device_brcm_rpi4/pull/2)

Thank you for your contribution. This seems reasonable but it's been a while since I've bothered to make any Android Automotive builds so I'm not able to test at the moment.

Does this resolve some specific issue with bluetooth on Android Automotive since IIRC there are quite a few?

adufftpc, 2024-03-27:
> Does this resolve some specific issue with bluetooth on Android Automotive since IIRC there are quite a few?

It removes rapid connections and disconnections from @SuperPou1 initial issue description. Also there was an issue he described as "on my phone when I try to play music for example, it instantly stops". It is resolved by adding the AVRCP profile. I can add a commit with the fix to the same pr if needed.

KonstaT, 2024-03-27:
> I can add a commit with the fix to the same pr if needed.

Please do if that fixes something. I'm not exactly sure how this should be configured for Automotive target. By Google's plan these  profiles (https://android.googlesource.com/platform/packages/services/Car/+/refs/tags/android-14.0.0_r22/car_product/properties/bluetooth.prop) should be enabled by default on car configuration but I guess that doesn't work on this hardware.

Also please prepend the commit messages with `car:` tag as this only affect Android Automotive build target.

KonstaT, 2024-03-28:
Closing as the issues reported in the original post should be be fixed with https://github.com/raspberry-vanilla/android_device_brcm_rpi4/pull/2.

closed 2024-03-28 by KonstaT

ggdover, 2024-10-23:
I had the exact same symptoms as the OP described, with my phone constantly disconnecting and reconnecting
and even with disabling text messaging access like this: https://github.com/raspberry-vanilla/android_device_brcm_rpi4/pull/2
I still got the same issue.

I checked "adb logs" and found this crash happening:
```
10-23 09:10:46.955  2819  2874 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 2874 (bt_a2dp_sink_wo), pid 2819 (droid.bluetooth)
10-23 09:10:47.718     0     0 W audit   : audit_lost=252 audit_rate_limit=5 audit_backlog_limit=64
10-23 09:10:47.729     0     0 E audit   : rate limit exceeded
10-23 09:10:47.871  2931  2931 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
10-23 09:10:47.871  2931  2931 F DEBUG   : Build fingerprint: 'Raspberry/aosp_rpi4_car/rpi4:14/AP2A.240905.003/eng.filip.20241018.102801:userdebug/test-keys'
10-23 09:10:47.871  2931  2931 F DEBUG   : Revision: '0'
10-23 09:10:47.871  2931  2931 F DEBUG   : ABI: 'arm64'
10-23 09:10:47.871  2931  2931 F DEBUG   : Timestamp: 2024-10-23 09:10:47.089046018+0000
10-23 09:10:47.871  2931  2931 F DEBUG   : Process uptime: 0s
10-23 09:10:47.871  2931  2931 F DEBUG   : Cmdline: com.android.bluetooth
10-23 09:10:47.871  2931  2931 F DEBUG   : pid: 2819, tid: 2874, name: bt_a2dp_sink_wo  >>> com.android.bluetooth <<<
10-23 09:10:47.871  2931  2931 F DEBUG   : uid: 1001002
10-23 09:10:47.872  2931  2931 F DEBUG   : tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
10-23 09:10:47.872  2931  2931 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
10-23 09:10:47.872  2931  2931 F DEBUG   : Abort message: '[FATAL:btif_avrcp_audio_track.cc(120)] Check failed: result == AAUDIO_OK. 
10-23 09:10:47.872  2931  2931 F DEBUG   : #00 0x00000070d9b5397f /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x0000000000aa197f
10-23 09:10:47.872  2931  2931 F DEBUG   : #01 0x00000070d985623b /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x00000000007a423b
10-23 09:10:47.872  2931  2931 F DEBUG   : #02 0x00000070d97ed0ab /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x000000000073b0ab
10-23 09:10:47.872  2931  2931 F DEBUG   : #03 0x00000070d9b5687b /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x0000000000aa487b
10-23 09:10:47.872  2931  2931 F DEBUG   : #04 0x00000070d9b55dfb /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x0000000000aa3dfb
10-23 09:10:47.872  2931  2931 F DEBUG   : #05 0x00000070d9b5610f /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x0000000000aa410f
10-23 09:10:47.872  2931  2931 F DEBUG   : #06 0x00000070d9b584ef /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x0000000000aa64ef
10-23 09:10:47.872  2931  2931 F DEBUG   : #07 0x00000070d9b64583 /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x0000000000ab2583
10-23 09:10:47.872  2931  2931 F DEBUG   : #08 0x00000070d9a7d247 /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x00000000009cb247
10-23 09:10:47.872  2931  2931 F DEBUG   : #09 0x00000070d9a7cd27 /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x00000000009cad27
10-23 09:10:47.872  2931  2931 F DEBUG   : #10 0x00000070d9a7d837 /apex/com.android.btservices/lib64/libbluetooth_jni.so+0x00000000009cb837
10-23 09:10:47.872  2931  2931 F DEBUG   : #11 0x000000742cb32c57 /apex/com.android.runtime/lib64/bionic/libc.so+0x0000000000076c57
10-23 09:10:47.872  2931  2931 F DEBUG   : #12 0x000000742cb25e6b /apex/com.android.runtime/lib64/bionic/libc.so+0x0000000000069e6b'
10-23 09:10:47.872  2931  2931 F DEBUG   :     x0  0000000000000000  x1  0000000000000b3a  x2  0000000000000006  x3  00000070c06dd9b0
10-23 09:10:47.872  2931  2931 F DEBUG   :     x4  62622d6a62607173  x5  62622d6a62607173  x6  62622d6a62607173  x7  7f7f7f7f7f7f7f7f
10-23 09:10:47.872  2931  2931 F DEBUG   :     x8  00000000000000f0  x9  000000742cadf390  x10 ffffff80fffffb9f  x11 0000000000000000
10-23 09:10:47.872  2931  2931 F DEBUG   :     x12 00000070c06dda28  x13 00000070c06dda1e  x14 00000070d910d826  x15 0000000000000054
10-23 09:10:47.872  2931  2931 F DEBUG   :     x16 000000742cb9efe0  x17 000000742cb82860  x18 0000007039eb8060  x19 0000000000000b03
10-23 09:10:47.872  2931  2931 F DEBUG   :     x20 0000000000000b3a  x21 00000000ffffffff  x22 00000070c06ddf70  x23 0000000000000007
10-23 09:10:47.872  2931  2931 F DEBUG   :     x24 00000070d9c7b000  x25 00000070c06ddac0  x26 00000070c06deac0  x27 00000070c06dda99
10-23 09:10:47.872  2931  2931 F DEBUG   :     x28 00000000000fe000  x29 00000070c06dda30
10-23 09:10:47.872  2931  2931 F DEBUG   :     lr  000000742cb22ad8  sp  00000070c06dd990  pc  000000742cb22b08  pst 0000000000000000
10-23 09:10:47.872  2931  2931 F DEBUG   : 15 total frames
10-23 09:10:47.872  2931  2931 F DEBUG   : backtrace:
10-23 09:10:47.872  2931  2931 F DEBUG   :       #00 pc 0000000000066b08  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: 995c9ec5a72bec30d188a89027728cd8)
10-23 09:10:47.872  2931  2931 F DEBUG   :       #01 pc 0000000000aa2240  /apex/com.android.btservices/lib64/libbluetooth_jni.so (base::debug::BreakDebugger()+48) (BuildId: 000e1b69aef48dc214b686f437ccb867)
10-23 09:10:47.872  2931  2931 F DEBUG   :       #02 pc 0000000000aa1c8c  /apex/com.android.btservices/lib64/libbluetooth_jni.so (logging::LogMessage::~LogMessage()+1008) (BuildId: 000e1b69aef48dc214b686f437ccb867)
[... cut ...]
**10-23 09:10:47.872  2931  2931 F DEBUG   : Abort message: '[FATAL:btif_avrcp_audio_track.cc(120)] Check failed: result == AAUDIO_OK.** 

If you check the surrounding code it looks like it fails to open a stream.
What I tested as a result was to switch which hdmi port I was using for the display output (since it also carries audio).
I'm running on a rpi 4 model b and when hdmi was plugged into "HDMI 1" (the hdmi furthest from the usb-c port)
I got the issue but when I plugged it into "HDMI 0" instead suddenly the issue was gone.
(The attached image shows the hdmi port I connected to for issue to disappear
[image]
)