raspberry-vanilla/android_local_manifest #34: Build failure in AOSP 14 r1

sandeepcg, 2023-10-28:
Facing issue with building with android 14 r1.

`ERROR: files are incompatible: android.hardware.automotive.vehicle@2.0::IVehicle/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/external/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/legacy/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.gatekeeper@1.0::IGatekeeper/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.graphics.allocator@4.0::IAllocator/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.graphics.composer@2.4::IComposer/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.keymaster@4.1::IKeymasterDevice/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.tv.cec@1.0::IHdmiCec/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.usb.gadget@1.2::IUsbGadget/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.automotive.vehicle@2.0::IVehicle/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/external/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/legacy/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.gatekeeper@1.0::IGatekeeper/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.graphics.allocator@4.0::IAllocator/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.graphics.composer@2.4::IComposer/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.keymaster@4.1::IKeymasterDevice/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.tv.cec@1.0::IHdmiCec/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.usb.gadget@1.2::IUsbGadget/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.automotive.vehicle@2.0::IVehicle/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/external/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/legacy/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.gatekeeper@1.0::IGatekeeper/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.graphics.composer@2.4::IComposer/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.keymaster@4.1::IKeymasterDevice/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.tv.cec@1.0::IHdmiCec/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.usb.gadget@1.2::IUsbGadget/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.automotive.vehicle@2.0::IVehicle/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/external/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.camera.provider@2.5::ICameraProvider/legacy/0 is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.gatekeeper@1.0::IGatekeeper/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.graphics.allocator@4.0::IAllocator/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.graphics.composer@2.4::IComposer/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.keymaster@4.1::IKeymasterDevice/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.tv.cec@1.0::IHdmiCec/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
android.hardware.usb.gadget@1.2::IUsbGadget/default is deprecated in compatibility matrix at FCM Version 8; it should not be served.
The following instances are in the device manifest but not specified in framework compatibility matrix: 
    android.hardware.automotive.vehicle@2.0::IVehicle/default
    android.hardware.camera.provider@2.5::ICameraProvider/external/0
    android.hardware.camera.provider@2.5::ICameraProvider/legacy/0
    android.hardware.gatekeeper@1.0::IGatekeeper/default
    android.hardware.graphics.allocator@4.0::IAllocator/default
    android.hardware.graphics.composer@2.4::IComposer/default
    android.hardware.keymaster@4.1::IKeymasterDevice/default
    android.hardware.tv.cec@1.0::IHdmiCec/default
    android.hardware.usb.gadget@1.2::IUsbGadget/default
Suggested fix:
1. Update deprecated HALs to the latest version.
2. Check for any typos in device manifest or framework compatibility matrices with FCM version >= 8.
3. For new platform HALs, add them to any framework compatibility matrix with FCM version >= 8 where applicable.
4. For device-specific HALs, add to DEVICE_FRAMEWORK_COMPATIBILITY_MATRIX_FILE or DEVICE_PRODUCT_COMPATIBILITY_MATRIX_FILE.: Success
INCOMPATIBLE`

ftab, 2023-10-28:
See here: https://github.com/raspberry-vanilla/android_local_manifest/issues/25

sandeepcg, 2023-10-29:
@ftab I checked link. Looks similar issue. But could you please provide exact steps to fix . 
Should I re-sync the repo ? or any configuration should fix this issue ?

ftab, 2023-10-29:
It is the same issue. I ran into the same one myself before I realized what I did wrong (just typing `make` and hoping for the best).

The fix is compiling only these things, bypassing the compatibility matrix check:

`make bootimage systemimage vendorimage -j$(nproc)`

KonstaT, 2023-10-30:
> Could you please provide exact steps to fix ?

Exact steps are in the build instructions.
https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0

Step 7. Compile:
`make bootimage systemimage vendorimage -j$(nproc)`

As commented in https://github.com/raspberry-vanilla/android_device_brcm_rpi4/commit/eeeab1fb7e41fa185dd94a7bb173bfec72b99c3a android-14.0 branch doesn't currently pass `make check-vintf-all`. android-13.0 branch does if this a requirement for you for some reason.

closed 2023-10-30 by KonstaT

sandeepcg, 2023-10-30:
@KonstaT  & @ftab  Thanks for your input. I will try this. :)