raspberry-vanilla/android_local_manifest #35: no raspberry pi settings in android automotive 14?

SuperPou1, 2023-10-28:
I built aaos 14 and in settings > system there are no raspberry pi settings like there are on your prebuilt images. why is it?

KonstaT, 2023-10-30:
Because it's not part of Raspberry Vanilla project.

You can achieve the same result for most parts by following the documentation (https://github.com/raspberry-vanilla/android_local_manifest/wiki) or editing config.txt (GPIO buttons, CPU overclocking, etc).

closed 2023-10-30 by KonstaT

abhilashdeshmukh, 2023-12-05:
I am quite new to android builds, so what do you mean it is not part of Raspberry Vanilla project?  
I am guessing raspberry pi setting is not part of it, but does that means we can add it for AAOS build also? 
I actually want to reboot into recovery so that I can install Google Play, it will be great if you can help me for that.

ssrakesh, 2025-09-14:
thanks for the guide I was able to integrate and use