raspberry-vanilla/android_local_manifest #37: DSI display no longer working on 14?

ftab, 2023-11-02:
Tried my own image (followed the instructions in the wiki to modify those files) and yours (set for touchscreen in Raspberry Pi settings).

I just get a blank screen, or the display briefly fades to white and then back to grey...

KonstaT, 2023-11-03:
I don't know. I don't have any DSI displays to test and this is the first report about DSI displays on Android 14 on one way or another.

I would assume it's not broken on kernel side at least (I switched to Linux 6.1 kernel on Android 14). It was specifically required to install 6.1 kernel on Android 13 to get some DSI displays working.

I noticed that headless mode (i.e. booting without HDMI display being connected) is broken on Android 14. Probably something released to HDMI-CEC. Could be the same issue here as you're not having a HDMI display connected. Logs (`logcat`) would tell for sure.

ftab, 2023-11-03:
Seems so...
```
01-01 00:00:32.149   376   422 I hwc-drm-two: No pipelines available. Creating null-display for headless mode
01-01 00:00:32.149   376   422 I hwc-display-configs: set mode 800x480@60Hz for HEADLESS-MODE
...
01-01 00:00:32.543   375   436 I [minigbm:gbm_mesa_internals.cpp(233)]: Found GPU v3d
01-01 00:00:32.543   375   436 I [minigbm:gbm_mesa_internals.cpp(249)]: GPU require KMSRO entry, searching for separate KMS driver...
01-01 00:00:32.544   375   436 E [minigbm:gbm_mesa_internals.cpp(264)]: Unable to find/open /dev/card node with KMS capabilities.
01-01 00:00:32.544   375   436 E [minigbm:gbm_mesa_internals.cpp(270)]: Unable to find or open DRM node
01-01 00:00:32.544   375   436 E [minigbm:gbm_mesa_internals.cpp(359)]: Failed to init gbm driver
01-01 00:00:32.545   375   436 E android.hardware.graphics.allocator@4.0-service.minigbm_gbm_mesa: Failed to create bo.
01-01 00:00:32.547   395   395 E GraphicBufferAllocator: Failed to allocate (800 x 480) layerCount 1 format 1 usage 1b00: 5
01-01 00:00:32.547   395   395 E BufferQueueProducer: [FramebufferSurface](id:18b00000000,api:1,p:395,c:395) allocateBuffers: failed to allocate buffer (0 x 0, format 1, usage 0x300)
```

Full log:
[logcat-no-dsi.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/13251692/logcat-no-dsi.txt)

KonstaT, 2023-11-05:
That doesn't look like the same issue.

Are you sure you have `dtoverlay=vc4-kms-dsi-7inch` in boot/config.txt?

Also this log looks like it's from my personal build rather than Raspberry Vanilla build. On my releases you would follow the steps in the FAQ on my site to enable the official 7" DSI display. On your own Raspberry Vanilla builds you follow the steps in the wiki.

ftab, 2023-11-05:
Yeah, I double checked both images after following the steps, by opening the config.txt on my computer to see that the overlays are set up. The logcat is from your flavor. Would it be better to get a logcat from my build? (my builds are automotive)

sandeepcg, 2023-11-08:
@ftab Did you find any solution to fix this issue ? I'm facing the same issue.

KonstaT, 2023-11-08:
Have you tried with a recent build from `android-13.0` branch? Just to rule out any issue with latest drm_hwcomposer/minigbm updates. You could also try AOSP 13 build with 6.1 kernel to rule out any issue with the kernel (sync https://github.com/raspberry-vanilla/android_local_manifest/blob/android-13.0/manifest_brcm_rpi4.xml#L9 and https://github.com/raspberry-vanilla/android_local_manifest/blob/android-13.0/manifest_brcm_rpi4.xml#L10 using `android-14.0` branch).

sandeepcg, 2023-11-08:
@KonstaT The above fix is working. But stuck at boot animation.

KonstaT, 2023-11-10:
@sandeepcg Using 6.1 kernel with `android-13.0` branch should work just fine. Likely some other reason if you're stuck in bootloop and logcat would tell exactly why.

This wasn't meant as a "fix" but to give something to bisect where the issue might be. Since kernel and drm_hwcomposer are not the cause, how about AOSP 14 build using minigbm from android-13.0 branch (https://github.com/raspberry-vanilla/android_local_manifest/blob/android-14.0/manifest_brcm_rpi4.xml#L51)?

ftab, 2023-11-10:
DSI works fine on `android-13.0`. I'll see what happens if I try the newer kernel & vendor/brcm.

KonstaT, 2023-11-11:
> DSI works fine on `android-13.0`. I'll see what happens if I try the newer kernel & vendor/brcm.

Rather try the older minigbm on `android-14.0` as mentioned above.

ftab, 2023-11-13:
Sorry, my bad. Synced android-13.0 branch of minigbm but it didn't work. The display flashes light grey briefly and then bleeds out to a dark grey/black pattern.

sandeepcg, 2023-11-14:
> @sandeepcg Using 6.1 kernel with `android-13.0` branch should work just fine. Likely some other reason if you're stuck in bootloop and logcat would tell exactly why.
>
> This wasn't meant as a "fix" [...]

@KonstaT facing same issue. Stuck in bootanimation. I will try to get logcat output.

ftab, 2023-11-14:
@sandeepcg that sounds different than what I'm getting. I don't even get the boot animation on DSI display.

sandeepcg, 2023-11-21:
@ftab How do you taking adb logs ? Serial console are you using ?

ftab, 2023-11-21:
> @ftab How do you taking adb logs ? Serial console are you using ?

Powering the Pi with the 5v header from the DSI display, then plugging a USB-A to C cable into the C port on the Pi. I don't have any suitable 3.3v serial adapters available right now, so this is the best I can get

sandeepcg, 2023-11-21:
[bootlogo_stuck_log.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/13428216/bootlogo_stuck_log.txt)
@KonstaT Attaching the logs for boot animation stuck issue.

`609 E Zygote  : System zygote died with fatal exception
11-08 17:57:36.029   609   609 E Zygote  : java.lang.RuntimeException: Failed to start service com.android.server.hdmi.HdmiControlService: onStart threw an exception
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.SystemServiceManager.startService(SystemServiceManager.java:257)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.SystemServiceManager.startService(SystemServiceManager.java:233)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.SystemServer.startOtherServices(SystemServer.java:2441)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.SystemServer.run(SystemServer.java:942)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.SystemServer.main(SystemServer.java:664)
11-08 17:57:36.029   609   609 E Zygote  : 	at java.lang.reflect.Method.invoke(Native Method)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:949)
11-08 17:57:36.029   609   609 E Zygote  : Caused by: java.lang.IllegalArgumentException: address should be positive.
11-08 17:57:36.029   609   609 E Zygote  : 	at android.hardware.hdmi.HdmiPortInfo$Builder.<init>(HdmiPortInfo.java:276)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.hdmi.HdmiCecController$NativeWrapperImpl.nativeGetPortInfos(HdmiCecController.java:1496)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.hdmi.HdmiCecController.getPortInfos(HdmiCecController.java:309)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.hdmi.HdmiCecNetwork.initPortInfo(HdmiCecNetwork.java:433)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.hdmi.HdmiControlService.initService(HdmiControlService.java:737)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.hdmi.HdmiControlService.onStart(HdmiControlService.java:651)
11-08 17:57:36.029   609   609 E Zygote  : 	at com.android.server.SystemServiceManager.startService(SystemServiceManager.java:255)`

There is something issue with HDMI

KonstaT, 2023-11-21:
@sandeepcg Ok, now we're at the same issue I mentioned earlier.

> I noticed that headless mode (i.e. booting without HDMI display being connected) is broken on Android 14. It looked like something probably related to HDMI-CEC. Could be the same issue here as you're not having any HDMI display connected. Logs (logcat) would tell for sure.

From what is this now exactly? AOSP14 with minigbm from `android-13.0` branch?

So, there's two separate issues with DSI display on AOSP14. One with minigbm on `android-14.0` and something changed in AOSP on how the HDMI-CEC is handled.

You can remove HDMI-CEC from your build by reverting https://github.com/raspberry-vanilla/android_device_brcm_rpi4/commit/c0a8378d8b8b25611c086d0745b44d2347bb923b.

sandeepcg, 2023-11-22:
> @sandeepcg Ok, now we're at the same issue I mentioned earlier.
> 
> > I noticed that headless mode (i.e. booting without HDMI display being connected) is broken on Android 14. It looked like something probably related to HDMI-CEC. Could be the same issue here as you're not having any HDMI display connected. Logs (logcat) would tell for sure.
> 
> From what is this now exactly? AOSP14 with minigbm from `android-13.0` branch?
> 
> If so, there's two separate issues with DSI display on AOSP14. One with minigbm on `android-14.0` branch and something changed in AOSP14 on how the HDMI-CEC is handled.
> 
> You can remove HDMI-CEC from your build by reverting [raspberry-vanilla/android_device_brcm_rpi4@c0a8378](https://github.com/raspberry-vanilla/android_device_brcm_rpi4/commit/c0a8378d8b8b25611c086d0745b44d2347bb923b).

I'm using AOSP14 with minigbm from `android-13.0` branch
I tried removing HDMI-CEC changes, stuck at boot animation issue is resolved. Able to boot the OS successfully.

KonstaT, 2023-11-22:
> I'm using AOSP14 with minigbm from android-13.0 branch
I tried removing HDMI-CEC changes, stuck at boot animation issue is resolved. Able to boot the OS successfully.

Great, good to know it's working this way.

I just need to figure out to do with the HDMI-CEC. `android.hardware.tv.cec@1.0` that I'm still using was deprecated in Android 14 which is likely why. I guess someone needs to write a new CEC HAL or drop the feature for now.

KonstaT, 2023-11-24:
I've pushed the minigmb on `android-14.0` to match `android-13.0` branch. I'd probably need to have some DSI display myself to bisect and test what exactly broke it if no one else will.

You're also welcome to look into how to workaround the issue with HDMI-CEC as removing the support entirely is likely not something that we'd want.

KonstaT, 2023-11-26:
Please sync your sources. This should be now fixed in `android-14.0` branch.

closed 2023-11-26 by KonstaT

znlsl, 2023-11-28:
Hi KonstaT
I synchronized the latest code according to the README of AOSP14 and compiled the version. Using a 7-inch DSI screen, I removed the comment in front of dtoverlay=vc4-kms-dsi-7inch, but the boot screen still does not work. After removing the comment from dtoverlay=vc4-kms-dsi-7inch, HDMI works normally.

I would like to ask if it's necessary to change the entries in .repo/local_manifests/manifest_brcm_rpi.xml from 
<project path="device/brcm/rpi4-kernel" name="raspberry-vanilla/android_device_brcm_rpi4-kernel" remote="github" revision="android-14.0" /> 
<project path="vendor/brcm" name="raspberry-vanilla/proprietary_vendor_brcm" remote="github" revision="android-14.0" /> 
to the android-13.0 branch?

KonstaT, 2023-11-28:
No, it shouldn't necessary to anything other than follow the relevant wiki page. It's more than just one line in config.txt.
https://github.com/raspberry-vanilla/android_local_manifest/wiki/DSI-display

znlsl, 2023-11-28:
> No, it shouldn't be necessary to do anything other than follow the relevant wiki page. It's more than just one line in config.txt. https://github.com/raspberry-vanilla/android_local_manifest/wiki/DSI-display

I recompiled and works normally, thank you very much!

KonstaT, 2023-11-28:
> I recompiled and works normally, thank you very much!

Thank you for testing and confirming.

KMiika, 2023-12-12:
> No, it shouldn't be necessary to do anything other than follow the relevant wiki page. It's more than just one line in config.txt. https://github.com/raspberry-vanilla/android_local_manifest/wiki/DSI-display

Hello KonstaT!
The android automotive version doesn't seem to work with these instructions.  Any idea why?

KonstaT, 2023-12-20:
> The Android Automotive version doesn't seem to work with these instructions. Any idea why?

No, that's what logs are for (`logcat` in this case). :)