raspberry-vanilla/android_local_manifest #38: Enable Android to act as a bluetooth speaker

usatenko, 2023-11-03:
I am trying to install RPI with Android 12 to the car and I am looking for a way RPI to act as a BT speaker, so, when I connect a phone via BT, music from the phone routes to the RPI.
Can you please suggest where to find the configuration files for this? For now, even if connected the phone just playing music with its speaker.
Thank you.

KonstaT, 2023-11-03:
`android-12.1` branch on Raspberry Vanilla is no longer maintained. Please use `android-13.0` or `android-14.0` branches instead.

According to https://github.com/raspberry-vanilla/android_local_manifest/issues/33 this somehow works on Android Automotive build target. You can look into the differences in bluetooth configuration between the targets (https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-14.0/vendor.prop#L10-L33 vs. https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-14.0/aosp_rpi4_car.mk#L18-L36) or if there's something else included in Android Automotive to enable A2DP sink.

usatenko, 2023-11-03:
thank you, I will try to research this question for 12.1 as my system is built already, I will make a pull request if I found a solution

KonstaT, 2023-11-05:
Pull requests are of course welcome and this could help other people looking for similar feature in their personal projects.

A2DP can be configured as either a host or a sink - not both at the same time. A2DP sink is not something that will be accepted for regular build targets as we'd generally want to stream bluetooth audio from Pi to an external speaker/headphones.

usatenko, 2023-11-06:
If at some point it will be possible to find the correct settings for sink and host it would be great to put to rpi settings all

KonstaT, 2024-01-09:
Closing as there hasn't been any activity in couple of months. This was more of an feature request as A2DP sink is not something intended for regular build targets.

closed 2024-01-09 by KonstaT