raspberry-vanilla/android_local_manifest #41: Boot failure with automotive build

tanisou, 2023-11-18:
I built "aosp_rpi4_car-userdebug" .img file following the instruction
Build process was success, but booting from it image was failed like below image.
I take below actions to resolve this.
1.  burn .img using two tools  "Raspberry Pi Imager" and "balena Etcher".
2. remove repository directory and rebuild it.
3. boot from two different kernel, one is original kernel build from this repository, other is manually built from [here]

(https://github.com/raspberry-vanilla/android_kernel_manifest/tree/android-14.0) following your description.
[image]

KonstaT, 2023-11-20:
That doesn't look like anything I've seen before. Have you tested using a different sdcard? There could be issue with some storage devices if they have GPT partition table. You can use e.g. GParted to create MBR partition table. IIRC Raspberry Pi imager's 'wipe sdcard' option also does this.

If builds from my site (https://konstakang.com/devices/rpi4/) boot, so should your own build.

KonstaT, 2023-12-02:
Closing as there's been no activity in two weeks. Assuming faulty sdcard and fixed by using a different one.

closed 2023-12-02 by KonstaT