raspberry-vanilla/android_local_manifest #42: External USB GPS support

dat-tran-halo, 2023-11-19:
Does any of the Android versions  for RP-14 support a USB GPS module?

dat-tran-halo, 2023-11-20:
I was able to compile and load Android 13 (automotive). I installed Maps but no GPS location with the USB module. I am able to read from /dev/ttyACM0. How can I configure the system to read the location from this device? Thank you.

KonstaT, 2023-11-20:
Raspberry Vanilla project doesn't provide support for external hardware such as USB GPS dongles.

You would need to implement Android serial GPS HAL for this to work. Some options to look into:
https://github.com/lineage-rpi/android_hardware_gps
https://github.com/zxcwhale/aosp13_gnss_hal

dat-tran-halo, 2023-11-20:
I was able to verify that your personal build for AOSP-13 works with the USB GPS module I have. 

However, when I try to integrate https://github.com/lineage-rpi/android_hardware_gps with the build from source provided by this repo, it doesn't give me a location.  Can you provide your gnss/gps changes to get it working with AOSP? Thanks

**My changes:**

```
project device/brcm/rpi4/
diff --git a/device.mk b/device.mk
index de2ed12..f961500 100644
--- a/device.mk
+++ b/device.mk
@@ -273,6 +273,12 @@ PRODUCT_PACKAGES += \
     wpa_supplicant \
     wpa_supplicant.conf
 
+# Gnss HAL
+PRODUCT_PACKAGES += \
+       android.hardware.gnss@1.0 \
+       android.hardware.gnss@1.0-impl \
+       android.hardware.gnss@1.0-service
+
 PRODUCT_COPY_FILES += \
     hardware/broadcom/wlan/bcmdhd/config/wpa_supplicant_overlay.conf:$(TARGET_COPY_OUT_VENDOR)/etc/wifi/wpa_supplicant_overlay.conf

```

**Extra build step:**

Default gps path to ttyACM0

mmm vendor/android_hardware_gps/

**I get these runtime errors in logcat:**

```
130|rpi4:/ $ logcat | grep gnss                                                                                                                                                                                                                                                      
01-01 00:00:17.036     0     0 I init    : starting service 'vendor.gnss_service'...
01-01 00:00:22.050     0     0 I init    : starting service 'vendor.gnss_service'...
01-01 00:00:22.135     0     0 I init    : Service 'vendor.gnss_service' (pid 471) exited with status 0
01-01 00:00:22.143     0     0 I init    : Sending signal 9 to service 'vendor.gnss_service' (pid 471) process group...
01-01 00:00:27.062     0     0 I init    : starting service 'vendor.gnss_service'...
01-01 00:00:27.119     0     0 I init    : Service 'vendor.gnss_service' (pid 493) exited with status 0
01-01 00:00:27.137     0     0 I init    : Sending signal 9 to service 'vendor.gnss_service' (pid 493) process group...
01-01 00:00:32.074     0     0 I init    : starting service 'vendor.gnss_service'...
01-01 00:00:32.136     0     0 I init    : Service 'vendor.gnss_service' (pid 530) exited with status 0
01-01 00:00:32.143     0     0 I init    : Sending signal 9 to service 'vendor.gnss_service' (pid 530) process group...
11-19 15:36:53.437   154   154 I servicemanager: Could not find android.hardware.gnss.IGnss/default in the VINTF manifest.
11-19 15:36:53.439   155   155 I hwservicemanager: getTransport: Cannot find entry android.hardware.gnss@2.1::IGnss/default in either framework or device VINTF manifest.
11-19 15:36:53.439   473   473 D GnssJni : gnssHal 2.1 was null, trying 2.0
11-19 15:36:53.440   155   155 I hwservicemanager: getTransport: Cannot find entry android.hardware.gnss@2.0::IGnss/default in either framework or device VINTF manifest.
11-19 15:36:53.440   473   473 D GnssJni : gnssHal 2.0 was null, trying 1.1
11-19 15:36:53.440   155   155 I hwservicemanager: getTransport: Cannot find entry android.hardware.gnss@1.1::IGnss/default in either framework or device VINTF manifest.
11-19 15:36:53.441   473   473 D GnssJni : gnssHal 1.1 was null, trying 1.0
11-19 15:36:53.444   155   155 I hwservicemanager: getTransport: Cannot find entry android.hardware.gnss@1.0::IGnss/default in either framework or device VINTF manifest.
01-01 00:00:37.083     0     0 I init    : starting service 'vendor.gnss_service'...
11-19 15:36:53.634   155   155 I hwservicemanager: getTransport: Cannot find entry android.hardware.gnss@1.0::IGnss/default in either framework or device VINTF manifest.
11-19 15:36:53.642   155   155 I hwservicemanager: getTransport: Cannot find entry android.hardware.gnss@1.0::IGnss/default in either framework or device VINTF manifest.
11-19 15:36:53.642   673   673 E HidlServiceManagement: Service android.hardware.gnss@1.0::IGnss/default must be in VINTF manifest in order to register/get.
11-19 15:36:53.642   673   673 E LegacySupport: Could not register service android.hardware.gnss@1.0::IGnss/default (-2147483648).
01-01 00:00:37.187     0     0 I init    : Service 'vendor.gnss_service' (pid 673) exited with status 0
01-01 00:00:37.201     0     0 I init    : Sending signal 9 to service 'vendor.gnss_service' (pid 673) process group...
11-19 15:36:53.627   673   673 I android.hardwar: type=1400 audit(0.0:86): avc: denied { read } for name="vndbinder" dev="binder" ino=6 scontext=u:r:hal_gnss_default:s0 tcontext=u:object_r:vndbinder_device:s0 tclass=chr_file permissive=1
11-19 15:36:53.627   673   673 I android.hardwar: type=1400 audit(0.0:87): avc: denied { write } for name="vndbinder" dev="binder" ino=6 scontext=u:r:hal_gnss_default:s0 tcontext=u:object_r:vndbinder_device:s0 tclass=chr_file permissive=1
11-19 15:36:53.627   673   673 I android.hardwar: type=1400 audit(0.0:88): avc: denied { open } for path="/dev/binderfs/vndbinder" dev="binder" ino=6 scontext=u:r:hal_gnss_default:s0 tcontext=u:object_r:vndbinder_device:s0 tclass=chr_file permissive=1
11-19 15:36:53.627   673   673 I android.hardwar: type=1400 audit(0.0:89): avc: denied { ioctl } for path="/dev/binderfs/vndbinder" dev="binder" ino=6 ioctlcmd=0x6209 scontext=u:r:hal_gnss_default:s0 tcontext=u:object_r:vndbinder_device:s0 tclass=chr_file permissive=1
11-19 15:36:53.627   673   673 I android.hardwar: type=1400 audit(0.0:90): avc: denied { map } for path="/dev/binderfs/vndbinder" dev="binder" ino=6 scontext=u:r:hal_gnss_default:s0 tcontext=u:object_r:vndbinder_dev [... cut ...]
11-20 15:43:23.612  1452  1452 I android.hardwar: type=1400 audit(0.0:102): avc: denied { ioctl } for path="/dev/binderfs/vndbinder" dev="binder" ino=6 ioctlcmd=0x6209 scontext=u:r:hal_gnss_default:s0 tcontext=u:object_r:vndbinder_device:s0 tclass=chr_file permissive=1
11-20 15:43:23.612  1452  1452 I android.hardwar: type=1400 audit(0.0:103): avc: denied { map } for path="/dev/binderfs/vndbinder" dev="binder" ino=6 scontext=u:r:hal_gnss_default:s0 tcontext=u:object_r:vndbinder_device:s0 tclass=chr_file permissive=1

```

dat-tran-halo, 2023-11-20:
I was able to fix the manifest error by adding to device/brcm/rpi4/manifest.xml

```
diff --git a/manifest.xml b/manifest.xml
old mode 100644
new mode 100755
index 20e91b0..a574b92
--- a/manifest.xml
+++ b/manifest.xml
@@ -67,4 +67,14 @@
             <instance>default</instance>
         </interface>
     </hal>
+    <hal format="hidl">
+        <name>android.hardware.gnss</name>
+        <transport>hwbinder</transport>
+        <version>1.0</version>
+        <interface>
+            <name>IGnss</name>
+            <instance>default</instance>
+        </interface>
+        <fqname>@1.0::IGnss/default</fqname>
+    </hal>
 </manifest>
diff --git a/ramdisk/ueventd.rpi4.rc b/ramdisk/ueventd.rpi4.rc
index bcdd85f..5b4e060 100644
--- a/ramdisk/ueventd.rpi4.rc
+++ b/ramdisk/ueventd.rpi4.rc
@@ -25,3 +25,5 @@
 /dev/video18                                                                 0660   media      media
 /dev/video19                                                                 0660   media      media
 /dev/video31                                                                 0660   media      media
+
+/dev/ttyACM0                                                                 0660   gps        gps

```

KonstaT, 2023-11-21:
I don't see anything missing but the permission (copy from `frameworks/native/data/etc/android.hardware.location.gps.xml`).

Also note that the GPS HAL might involve some hacking around https://android.googlesource.com/platform/frameworks/base/+/refs/tags/android-14.0.0_r14/location/java/android/location/LocationResult.java#109 as not all serial GPS devices report complete location. This might be also fixed in some serial GPS HALs you might find.

dat-tran-halo, 2023-11-21:
I'll give that a try. Thanks for all your help.

Here is my latest patch for GPS if it helps anyone else:

```
project device/brcm/rpi4/
diff --git a/device.mk b/device.mk
index de2ed12..3355c17 100644
--- a/device.mk
+++ b/device.mk
@@ -273,6 +273,15 @@ PRODUCT_PACKAGES += \
     wpa_supplicant \
     wpa_supplicant.conf
 
+# Gnss HAL
+PRODUCT_PACKAGES += \
+       android.hardware.gnss@1.0 \
+       android.hardware.gnss@1.0-impl \
+       android.hardware.gnss@1.0-service
+
+PRODUCT_COPY_FILES += \
+    frameworks/native/data/etc/android.hardware.location.gps.xml:$(TARGET_COPY_OUT_VENDOR)/etc/permissions/android.hardware.location.gps.xml
+
 PRODUCT_COPY_FILES += \
     hardware/broadcom/wlan/bcmdhd/config/wpa_supplicant_overlay.conf:$(TARGET_COPY_OUT_VENDOR)/etc/wifi/wpa_supplicant_overlay.conf
 
diff --git a/manifest.xml b/manifest.xml
old mode 100644
new mode 100755
index 20e91b0..a574b92
--- a/manifest.xml
+++ b/manifest.xml
@@ -67,4 +67,14 @@
             <instance>default</instance>
         </interface>
     </hal>
+    <hal format="hidl">
+        <name>android.hardware.gnss</name>
+        <transport>hwbinder</transport>
+        <version>1.0</version>
+        <interface>
+            <name>IGnss</name>
+            <instance>default</instance>
+        </interface>
+        <fqname>@1.0::IGnss/default</fqname>
+    </hal>
 </manifest>
diff --git a/ramdisk/ueventd.rpi4.rc b/ramdisk/ueventd.rpi4.rc
index bcdd85f..5b4e060 100644
--- a/ramdisk/ueventd.rpi4.rc
+++ b/ramdisk/ueventd.rpi4.rc
@@ -25,3 +25,5 @@
 /dev/video18                                                                 0660   media      media
 /dev/video19                                                                 0660   media      media
 /dev/video31                                                                 0660   media      media
+
+/dev/ttyACM0                                                                 0660   gps        gps
diff --git a/vendor.prop b/vendor.prop
index c5fd75c..9b05a96 100644
--- a/vendor.prop
+++ b/vendor.prop
@@ -79,3 +79,5 @@ ro.vendor.v4l2_codec2.encode_concurrent_instances=8
 # Wifi
 wifi.interface=wlan0
 
+# GPS
+ro.kernel.android.gps=ttyACM0

```

dat-tran-halo, 2023-11-21:
@KonstaT, your suggestion for the permission file worked perfectly. Thanks again for your help.

closed 2023-11-21 by dat-tran-halo