raspberry-vanilla/android_local_manifest #43: Ask About Instrument Cluster Android Automotive OS 13 on RaspberryPi 4

fakry32, 2023-11-21:
Hi Konsta

So I have a problem where I am confused about how to display the Android Automotive OS 13 Instrument Cluster on RaspberryPi 4. Can you help me?

for the details of the problem I have written in the document that I attached
[Instrument Cluster Problem.pdf](https://github.com/raspberry-vanilla/android_local_manifest/files/13421392/Instrument.Cluster.Problem.pdf)

KonstaT, 2023-12-02:
Android code search would probably come in handy looking into e.g. what BUILD_EMULATOR_CLUSTER_DISPLAY does (https://cs.android.com/search?q=BUILD_EMULATOR_CLUSTER_DISPLAY -> https://cs.android.com/android/platform/superproject/main/+/main:device/generic/car/emulator/cluster/cluster-hwservicemanager.mk).

You would likely need to implement something similar in the Raspberry Pi device configuration to show the instrument cluster on secondary HDMI display.

RobinHeart317, 2023-12-28:
Hi guys,

Is there anyway to display Instrument Cluster on RaspberryPi 4 yet? I am also doing this project but still dont know how.

closed 2024-09-07 by KonstaT