raspberry-vanilla/android_local_manifest #47: "adb remount" cannot be used on AOSP14 for RPI4.

znlsl, 2023-11-28:
Hi Konsta

I synced to the latest source code and then locally compiled the complete AOSP14 version for RPI4. After the system started, I found that adb remount failed. Is there any other operation required to unlock the BootLoader?

$ adb root
$ adb remount
Device must be bootloader unlocked

Thank you!

znlsl, 2023-11-28:
I followed the above steps, and the remount worked perfectly! Thank you so much!!!

closed 2023-11-28 by znlsl

mrarjun, 2024-01-17:
Same issue observed in RPI-5 with 14_r20 code base.
Tried to update in device/brcm/rpi5/BoardConfig.mk file with below line. But still not able to remount
```BOARD_KERNEL_CMDLINE += ro.boot.verifiedbootstate=orange```
How can i fix ```Device must be bootloader unlocked``` issue using ```ro.boot.verifiedbootstate=orange``` solution?

Not tried to update system/core file.

KonstaT, 2024-01-22:
@mrarjun Have you checked that it actually made it to /boot/cmdline.txt?

Adding something to `BOARD_KERNEL_CMDLINE` with `+=` would only work after it's being defined (https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-14.0/BoardConfig.mk#L48). It gets otherwise overriden with the `:=` definition.

mrarjun, 2024-01-23:
In `out/target/product/rpi5/rpiboot/cmdline.txt` file displaying as below

`console=ttyAMA0,115200 no_console_suspend root=/dev/ram0 rootwait androidboot.hardware=rpi5 ro.boot.verifiedbootstate=orange androidboot.selinux=permissive`

Is there any other change expected?

KonstaT, 2024-01-24:
> In `out/target/product/rpi5/rpiboot/cmdline.txt` file displaying as below

Yeah, but that's still not `/boot/cmdline.txt` on the image that you've flashed to the storage device.

Everything you need to know is already explained in https://github.com/raspberry-vanilla/android_local_manifest/issues/47#issuecomment-1829888921.