raspberry-vanilla/android_local_manifest #51: Are there any plans to support TYPE-C related functions on rpi5?

xulinkun, 2023-12-13:
Hi, guy. Thank you very much for your contribution to AOSP, but it's not quite perfect yet and I'd like to know if you have any plans to support TYPE-C(adb) on rpi5, which is important to me.

KonstaT, 2023-12-20:
USB-C (ADB, MTP, PTP, USB tethering) - dwc2 is not yet supported on Pi 5 firmware.
https://github.com/raspberrypi/bookworm-feedback/issues/77

Not really something I can do anything about but will leave this open until this is supported on the firmware.

josinact, 2024-01-04:
Hello, issue 77 seems to be fixed so will AOSP rpi5 be able to support Type-C once I update the firmware?

closed 2024-01-04 by KonstaT

xulinkun, 2024-01-16:
> Yes, this is now resolved on Raspberry Vanilla sources and my latest AOSP14 release (20240104).
> 
> This requires three things:
> 
>     1. EEPROM update to 2023-12-06 or newer
> 
>     2. kernel with [raspberry-vanilla/android_kernel_brcm_rpi@82069a7](https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/commit/82069a7a02632aa60fa5c69415bf891ede7d6fd4)
> 
>     3. Android USB gadget HAL with fixed UDC path that closed this issue ([raspberry-vanilla/android_device_brcm_rpi5@50ef80a](https://github.com/raspberry-vanilla/android_device_brcm_rpi5/commit/50ef80ac75730e7a9e0b26073ed7b4e2e2810324))

So, the AOSP14 release (20240104) has included 2,3 requires. And then, how to update EEPROM?

josinact, 2024-01-16:
> So, the AOSP14 release (20240104) has included 2,3 requires. And then, how to update EEPROM?

Load a 
https://raspberrypi.stackexchange.com/questions/140846/is-it-safe-to-update-eeprom-to-latest-using-rpi-eeprom-update

xulinkun, 2024-01-16:
> > > Yes, this is now resolved on Raspberry Vanilla sources and my latest AOSP14 release (20240104).
> > > This requires three things:
> > > ```
> > > 1. EEPROM update to 2023-12-06 or newer
> > > 
> > > 2. kernel with [raspberry-vanilla/android_kernel_brcm_rpi@82069a7](https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/commit/82069a7a02632aa60fa5c69415bf891ede7d6fd4)
> > > 
> > > 3. Android USB gadget HAL with fixed UDC path that closed this issue ([raspberry-vanilla/android_device_brcm_rpi5@50ef80a](https://github.com/raspberry-vanilla/android_device_brcm_rpi5/commit/50ef80ac75730e7a9e0b26073ed7b4e2e2810324))
> > > ```
> > 
> > 
> > So, the AOSP14 release (20240104) has included 2,3 requires. And then, how to update EEPROM?
> 
> Load a normal Bookworm OS onto your pi and update the EEPROM by downloading the latest pi updates (sudo apt-get install sudo apt-get full-upgrade) and then use sudo raspi-config advanced options to select the latest bootloader https://raspberrypi.stackexchange.com/questions/140846/is-it-safe-to-update-eeprom-to-latest-using-rpi-eeprom-update

Awesome! I updated the latest bootloader and then, ADB is working normally. Thanks a lot.