raspberry-vanilla/android_local_manifest #52: build error on ubuntu 18.04

CharlesHe023, 2023-12-21:
FAILED: out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp
/bin/bash -c "(rm -rf out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_LIBCAMERA//build ) && (cp -r external/libcamera//* out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (cd out/target/product/rpi4/obj/MESON_LIBCAMERA && PATH=/usr/bin:/usr/local/bin:\$PATH meson ./build --cross-file /disk/charles/opensource/android/rpi/rpi_android_13/out/target/product/rpi4/obj/MESON_LIBCAMERA_GEN/aosp_cross --buildtype=release -Dwerror=false -Dandroid=enabled -Dipas=raspberrypi -Dpipelines=raspberrypi -Dsysconfdir=/vendor/etc -Dtest=false ) && (PATH=/usr/bin:/bin:/sbin:\$PATH ninja -C out/target/product/rpi4/obj/MESON_LIBCAMERA/build ) && (touch out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp )"
"meson" is not allowed to be used. See https://android.googlesource.com/platform/build/+/master/Changes.md#PATH_Tools for more information.

CharlesHe023, 2023-12-21:
> FAILED: out/target/product/rpi4/obj/MESON_LIBCAMERA/.build.timestamp /bin/bash -c "(rm -rf out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) && (mkdir -p out/target/product/rpi4/obj/MESON_LIBCAMERA/ ) [...]

fixed after install python modules(ply jinja2 pyyaml) and upgrade ninja.

KonstaT, 2023-12-22:
Please follow the instructions on how to establish build environment including installing additional packages.

closed 2023-12-22 by KonstaT

CharlesHe023, 2023-12-24:
> Please follow the instructions on how to establish build environment including installing additional packages.

fixed after establish build environment, thanks.