raspberry-vanilla/android_local_manifest #53: Add support (new pid) for : SiS touch controller - Hanspree 231 - 23" Touch monitor

Bingo600, 2023-12-24:
Add support for :
SiS touch controller - Hanspree 231 - 23" Touch monitor
VID/PID - 0457:1057

***********

I have made two 23-inch "kitchen tablets",for the wife.
They both use a : Hanspree 231 23" 10-point multitouch screen, and right now they use a Remix Android device (EOL/EOS).
I found out the "hard way", that the VID:PID of the Hanspree 231. Is not default supported in Android.
And after buying several "TV-Boxes" where touch didn't work, i finnaly found the Remix mini that worked.
https://www.techradar.com/reviews/pc-mac/pc-mac-desktops/jide-remix-mini-1327646/review 

But that device is now EOL/EOS, and i'm a bit worried about the device, still trying to update from the Jide repos.
Ie. Chrome isn't from Google Appstore, but from Jide's repos, and is still on v.89.x.x.

I would love to get two .. either Raspi4 or Raspi5 as replacement for the Remix boxes.
But i need to make sire the USB Touchcontroller is supported (Known).

In fact i already have one 4GB Raspi4 that i can test with.

***********

As seen here (older version), it seems like it's just the PID that has to be known, the SIS support is already in the code.
HANNS-G HT231
https://forum.odroid.com/viewtopic.php?t=3707&start=80

***********

I have searched your AOSP14 kernel, and i was wondering if it would be possible for you to add VID 1057

SIS
https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-14.0/drivers/hid/hid-quirks.c

https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/blob/android-14.0/drivers/hid/hid-ids.h

https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/tree/android-14.0/drivers/hid

I assume the new PID has to be added to hid-ids.h
And a new entry has to be added to hid-quirks.c  - Where the other SIS Touch devices are listed

I have made a suggestion, and attached a hanspree-231.patch

TIA
Bingo

***********

Device info:

https://www.hannspree.eu/ht-231-hpb%20touch%20monitor.html

SiS touch controller - Hanspree 231
VID/PID - 0457:1057

Bus 001 Device 040: ID 0457:1057 Silicon Integrated Systems Corp.

***********

Bingo600, 2023-12-24:
Hmm the file (diff/patch) 
[hanspree-231.patch.tar.gz](https://github.com/raspberry-vanilla/android_local_manifest/files/13761515/hanspree-231.patch.tar.gz)
were lost

KonstaT, 2023-12-28:
Yes, this is something that could be included if you've actually confirmed that this patch does what it's expected.

There's kernel build instructions available at https://github.com/raspberry-vanilla/android_kernel_manifest/tree/android-14.0 and you can send a pull request to the kernel repo (https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/tree/android-14.0) once you've tested this (since I don't have the hardware to test myself).

Bingo600, 2023-12-28:
Thanx i'll give it a spin in the new year, when i'm home again.
I'm not experienced w. github , have no idea how to make a pull request.
But i can attach the git diff, once i have confirmed it to work.

Bingo600, 2024-01-02:
I followed the build instructons , and downloaded 96GB of repo ....
Then i built for Raspi4.
One "original" "Boot dir" , and one "hanspree patched Boot dir".

I'ts my first Android fiddeling , so how do i utilize/test the kernelimage + dts dir ?
Do i download your AOSP14 image , extract to a SD card , and replace the files there ?

Bingo600, 2024-01-03:
How do i build the full installable image 

Should i have used this recipe :
https://github.com/raspberry-vanilla/android_local_manifest
Instead of the kernel recipe ?

Any idea about how much data the above recipe will pull from repos ?
Sorry for being "green" :-)

KonstaT, 2024-01-03:
> I'ts my first Android fiddeling , so how do i utilize/test the kernelimage + dts dir ?
> Do i download your AOSP14 image , extract to a SD card , and replace the files there ?

You would write the AOSP14 image on your sdcard and mount the sdcard on your computer after this. You would copy `out/common/arch/arm64/boot/Image` (i.e. the kernel you built) as `/boot/Image` on the sdcard to replace the existing kernel.

> How do i build the full installable image
> ...

Yes, by following the instruction but it's not necessary for this.

Bingo600, 2024-01-06:
Hi Konsta
I can confirm that the above Raspi4 "kernel patch"- Aka. Add new VID for the touchcontroller used in the HT231.
It enables the touch interface on my Hanspree / HANNS.G HT231 - 23" Touch monitor.
I have no idea how to do a "pull request", but it's just 1 line in two separate files.

As described i just replaced the /boot/Image of the SD Card , with the one built after i added the new PID in the two files.

[hanspree-231-patch.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/13849184/hanspree-231-patch.txt)
[hanspree-231-patch.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/13849186/hanspree-231-patch.txt)

I can also confirm that WiFi is working.

I changed the sound to HDMI0, and i heard "Touch keyboard clicks".
Strange that sound on : dr.dk  (DK National TV) "inline video" or web streaming tv- Did NOT work in browser.

But  i heard the keyboard clicks, so i was quite sure the sound was working 
And sound via HDMI0 worked on Youtube in browser.

If you would apply the patch to your future kernel build(s), it will be much appreciated.

closed 2024-01-08 by KonstaT