raspberry-vanilla/android_local_manifest #54: AOSP on rpi5 - first report and issue with 800x480 hdmi monitor

samm-git, 2023-12-26:
First of all - thank you for your work, The Android port works surprisingly well. My target is to build a device to play spotify/apple music/some streaming for home use. On the latest image Android boots fine and works well (tested wifi, Bluetooth, ethernet, USB) but only if connected to the primary monitor with 1920x1080@60 resolution. I also do have a small 5 inch HDMI LCD monitor which works at 800x480 resolution and supports touchscreen (with USB connection).  When i am trying to connect it - boot fails. I also tried to put 800x480 to the `resolution.txt` on the boot partition but it does not help. I was able to use this monitor with raspberry os on the same board, so the problem is not hardware. 

So few questions:

1. Is it a known issue? Any workarounds?
2. What is the best way to collect boot logs? I do have various usb2serial converters, please let me know if serial output is active or i need to modify something. 
3. I am thinking to get another monitor from Ali, also HDMI, but a bit curious about support of non-standard resolutions.

Thank you again, hope to get things working :)

samm-git, 2023-12-26:
I was able to get serial port connection to rpi5, here is a connect log:

```

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.1.61-v8-g2cf4312bc060 (build-user@build-host) (Android (8508608, based on r450784e) clang version 14.0.7 (https://android.googlesource.com/toolchain/llvm-project 4c603efb0cca074e9238af8b4106c30add4418f6), LLD 14.0.7) #1 SMP PREEMPT Mon Nov 20 14:01:42 UTC 2023
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ac00000, size 320 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000001ac00000..0x000000002ebfffff ( 327680 KB ) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff ( 512 KB ) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd167a0..0x000000003fd16 d Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fafffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 1280 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 29 pages/cpu s81384 r8192 d29208 u118784
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Hardware dirty bit management
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2063104
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 pci=pcie_bus_safe snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=D8:3A:DD:BB:1C:B3 vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 no_console_suspend root=/dev/ram0 rootwait androidboot.hardware=rpi5 androidboot.selinux=permissive androidboot.serialno=04360e47f305cea3 androidboot.btmacaddr=11:22:f3:05:ce:a3
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000003bb00000-0x000000003fb00000] (64MB)
[    0.000000] Memory: 7793412K/8383488K available (18560K kernel code, 2330K rwdata, 6876K rodata, 1920K init, 1479K bss, 262396K reserved, 327680K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 54701 entries in 214 pages
[    0.000000] ftrace: allocated 214 pages with 5 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
 [NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0bled.
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000206] kfence: initialized - using 2097152 bytes for 255 objects at 0x(____ptrval____)-0x(____ptrval____)
[    0.000266] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000275] pid_max: default: 32768 minimum: 301
[    0.000318] LSM: Security Framework initializing
[    0.000332] SELinux:  Initializing.
[    0.000422] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000448] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001076] cgroup: Disabling memory control group subsystem
[    0.001900] cblist_init_generic: Setting adjustable number of callback queues.
[    0.001904] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.001964] cblist_init_generic: Setting adjustable number of callback queues.
[    0.001967] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.002025] cblist_init_generic: Setting adjustable number of callback queues.
[    0.002028] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.002162] rcu: Hierarchical SRCU implementation.
[... 1499 lines cut ...]
[  112.284502] EXT4-fs (loop11): unmounting filesystem.
[... cut ...]
[  112.888708] EXT4-fs (loop4): unmounting filesystem.
[  112.894174] EXT4-fs (dm-21): unmounting filesystem.
[  112.940719] EXT4-fs (loop14): unmounting filesystem.
[  112.947657] EXT4-fs (dm-16): unmounting filesystem.
[  113.013795] printk: apexd: 59 output lines suppressed due to ratelimiting
[  113.020755] apexd: apexd terminated by exit(1)
[  113.020755]
[  113.039860] EXT4-fs (mmcblk0p4): unmounting filesystem.
[  113.254788] kvm: exiting hardware virtualization
[  113.265254] reboot: Restarting system with command 'RescueParty'
```

samm-git, 2023-12-26:
Looks like this display does not have correct edid, this is from serial console, before it rebooted:

```
dumpsys SurfaceFlinger --display-id
Display 0 (HWC display 0): invalid EDID
```
And this is on normal (working) monitor:
```
dumpsys SurfaceFlinger --display-id
Display 0 (HWC display 0): port=0 pnpId=HWP displayName="HP ZR2240w"
```

Do you know if there is a way to provide fake edid for the screen not supporting it?

samm-git, 2023-12-26:
I also found that if i am setting resolution to "preferred", booting with normal HDMI display and later connecting to small 800x480 it is actually working! touchscreen seems to be uncalibrated, but its a different issue.

samm-git, 2023-12-26:
Okay, spent an evening but was able to create working solution with overriden EDID.

Problems are:

1. EDID in the screen is very naive and contain only very few fields and does not pass android validation
2. I was able to fake it by putting  `drm.edid_firmware="edid-fixed.bin" firmware_class.path="/boot"` to the `/boot/cmdline.txt` and creating own EDID which contains correct settings.

It now finally boots and displays correctly. I think this way we can fix other non-edid compatible devices.

samm-git, 2023-12-26:
Just in case if anyone will want patched EDID for this device i will attach it here: 
[edid-fixed.bin.zip](https://github.com/raspberry-vanilla/android_local_manifest/files/13773843/edid-fixed.bin.zip)

Changes are - added some CEA optional blocks, added detailed timing description, fixed Detailed Descriptor fields, all done with AW EDID Editor

KonstaT, 2023-12-28:
Closing as assuming resolved.

> I am using AOSP14-20231123-KonstaKANG-rpi5.zip image.

In which case where you got the build would be the appropriate place to discuss this. In the relevant FAQ section:
Q: My display is not working. I can only see the rainbow screen but no Android boot animation. What should I do?
A: This build only supports HDMI displays that report supported resolutions using EDID. 1920x1080 resolution is used by default with this build. You can change value in /boot/resolution.txt to use a different resolution that your display supports. Removing /boot/resolution.txt will use the preferred resolution of your display.

If you are building Raspberry Vanilla yourself, you would follow instructions on the relevant wiki page:
https://github.com/raspberry-vanilla/android_local_manifest/wiki/HDMI-display

closed 2023-12-28 by KonstaT