raspberry-vanilla/android_local_manifest #55: Cannot open "Developer options"

RobinHeart317, 2023-12-28:
Hi KonstaT,

Thank you for your work. It is wonderful. I have followed the file "Readme" in branch android-13.0 and built successfully. But I met an issue that I could not open "Developer Options" in Setting > System > Developer Options. Did I do anything wrong or did I miss something or even forgot to config something before build?

KonstaT, 2023-12-28:
On `userdebug`/`user` builds you need to click Settings -> About tablet -> Build number several times (vs. `eng` builds where developer options are enabled by default).

RobinHeart317, 2023-12-29:
I have enabled developer options but the problem is when I click on it it takes me back to settings. Please see the attached video below. 

https://drive.google.com/file/d/1wqw_5vyVIypDQcS_M0O8tSnDaY7yXgJG/view?usp=sharing

KonstaT, 2023-12-29:
`logcat`?

KonstaT, 2024-01-22:
Something like https://github.com/LineageOS/android_packages_modules_adb/commit/88411b79d13147583fcd96d2e8c94e6cc6839ee0 would help?

remarkked, 2024-02-02:
> Something like [LineageOS/android_packages_modules_adb@88411b7](https://github.com/LineageOS/android_packages_modules_adb/commit/88411b79d13147583fcd96d2e8c94e6cc6839ee0) would help?

Missing root folder in packages/modules/adb. Pulling adb module branch to this commit causes many build failures:

remarkked, 2024-02-02:
> Assuming this is `android-13.0` branch, Android Automotive build target, on Raspberry Pi 4. Can't recall experiencing similar issue, though.

Do Developer Options open normally on your build (android-13.0.0_r75 branch)?

KonstaT, 2024-09-07:
Closing old AOSP13 related issues as we're currently at Android 15.

Opening developer options works on Android Automotive builds from both `android-14.0.0_r67` and `android-15.0.0_r1` branches.

closed 2024-09-07 by KonstaT

MagicAndre1981, 2024-10-16:
> Opening developer options works on Android Automotive builds from both `android-14.0.0_r67` and `android-15.0.0_r1` branches.

for me it fails on Android TV 15 with android-15.0.0_r2. Compile for RPi4 works fine, [USB boot works after changing the fstab entries](https://github.com/raspberry-vanilla/android_local_manifest/wiki/USB-boot), enabling Developer Settings works, but when I tried to open developer settings to enable adb (over network) it closes.