raspberry-vanilla/android_local_manifest #57: AOSP14 can't boot

xulinkun, 2023-12-29:
Hi KonstaT, the AOSP14 on my side can't boot normally(rpi4 and rpi5 both). Its symptoms are repeated restarts on the "android" logo screen. Do you have any suggestion?

KonstaT, 2023-12-29:
Suggestions include posting `logcat` output that will tell exactly the device is not booting. Also please read the wiki (https://github.com/raspberry-vanilla/android_local_manifest/wiki) if there is something related (display, USB boot, etc).

yoyicue, 2024-01-03:
> Hi KonstaT, the AOSP14 on my side can't boot normally(rpi4 and rpi5 both). Its symptoms are repeated restarts on the "android" logo screen. Do you have any suggestion?

me too

KonstaT, 2024-01-03:
> me too，I have a hunch that the variant of RPI used in Mainland China might incorporate distinct devices.

I highly doubt it. There's no point in guessing in any case. Please post `logcat` output. Also please make note of the device (rpi4 or rpi5), branch (`android-14.0` in this case), and build target (tablet, Android TV, or Android Automotive) when reporting issues.

If you are using my personal builds from my site instead of building Raspberry Vanilla yourself, this is not quite the right place to discuss it. E.g. you could be facing https://github.com/raspberry-vanilla/android_local_manifest/issues/37 that already long fixed if you're building yourself. On my personal releases you would connect the display on the primary HDMI0 port (instead of the secondary HDMI1) to workaround this issues.

yoyicue, 2024-01-04:
I successfully accessed the computer through the HDMI2.

yoyicue, 2024-01-04:
> > me too，I have a hunch that the variant of RPI used in Mainland China might incorporate distinct devices.
> 
> I highly doubt it. There's no point in guessing in any case. Please post `logcat` output. Also please make note of the device (`rpi4 `or `rpi5`), branch (`android-14.0` in this case), and build target (tablet, Android TV, or Android Automotive) when reporting issues.
> 
> If you are using my personal builds from my site instead of building Raspberry Vanilla yourself, this is not quite the right place to discuss it. E.g. you could be facing #37 that is already long fixed if you're building Raspberry Vanilla yourself. On my latest AOSP14 releases you would connect the display on the primary HDMI0 port (instead of the secondary HDMI1) to workaround this issues for now.

I successfully accessed the computer through the HDMI0.

KonstaT, 2024-01-04:
Closing as likely duplicate of https://github.com/raspberry-vanilla/android_local_manifest/issues/37.

This is now resolved on Raspberry Vanilla sources and my latest AOSP14 release (20240104).

closed 2024-01-04 by KonstaT