raspberry-vanilla/android_local_manifest #58: DSI display not working with AOSP14 on Raspberry Pi 5

KonstaT, 2024-01-09:
There seems to be some Raspberry Pi 5 specific issue with DSI display on AOSP14. DSI display is reported working with AOSP14 on Raspberry Pi 4.

Bootanimation is displayed on the DSI display but Android UI fails to show.

```
01-05 16:29:33.760   372   372 E hwc-display: Failed to create DrmKmsPlan
01-05 16:29:33.760   387   387 E HWComposer: presentAndGetReleaseFences: present failed for display 0: BadConfig (1)
```

`logcat` on rpi5 (not working):
https://gist.github.com/mkl/30147ab52e761d8ff6945f206cab2b4d
https://gist.github.com/lewilou22/705eab3e1de68caee451c82d56c0263c

`logcat` on rpi4 (working):
https://gist.github.com/mkl/863b65fc59505ab2b3db53438ea663c9

Comment by Raspberry Pi engineer (https://forums.raspberrypi.com/viewtopic.php?t=362563#p2175370) suggests that Pi 5 creates separate cards for DSI displays. Likely something to do with the new RPI1 hardware on the Pi 5.

```
# ls -l /dev/dri
total 0
crw-rw-rw- 1 root graphics 226, 0 1970-01-01 00:00 card0
crw-rw-rw- 1 root graphics 226, 1 1970-01-01 00:00 card1
crw-rw-rw- 1 root graphics 226, 2 1970-01-01 00:00 card2
crw-rw-rw- 1 root graphics 226, 128 1970-01-01 00:00 renderD128
```

```
# ls -l /sys/devices/platform/axi/axi:gpu/drm
total 0
drwxr-xr-x 7 root root 0 1970-01-01 00:00 card1
lrwxrwxrwx 1 root root 0 2024-01-08 18:16 controlD65 -> card1
```

lewilou22, 2024-01-13:
"Android by Emteria" image that is included in the raspberry pi imager loads on the pi5 with this 7 inch DSI touch screen

here is the log cat log of it working on DSI0

https://gist.github.com/lewilou22/ea511c18a88c024bfc100a09ff096161

lewilou22, 2024-01-14:
Found infomation on DSI displays for PI5 
might be helpful 

https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003472-WP/Using-a-DSI-display.pdf

"On the Raspberry Pi 5, both ports have moved from the SoC to the Raspberry Pi RP1 device and are exposed. Both DSI0
and DSI1 are 4-lane and implement DSI 1.1."

KonstaT, 2024-01-15:
> "Android by Emteria" image that is included in the raspberry pi imager ...

"Android by Emteria" is closed source so this doesn't help much. So, we know that DSI displays can also work on Pi 5 but that's about it. Emteria seems to use the same/similar graphics implementation (Mesa, minigbm gralloc, drm_hwcomposer) as Raspberry Vanilla but it's impossible to say what exactly goes into that.

One thing I noted as I tested the Pi 5 Android 14 Emteria 2023-12-09 image from Raspberry Pi imager, it uses Linux kernel 6.1.61-v8-g2cf4312bc060-dirty which looks familiar (https://github.com/raspberry-vanilla/android_device_brcm_rpi5-kernel/commit/4d0a0e41e184099f0eacce3ef3667b1c2a0bafdc & https://github.com/raspberry-vanilla/android_kernel_brcm_rpi/tree/2cf4312bc060f2bed344535c3b69cf7192537d3e). This means that the kernel is based on my Raspberry Vanilla kernel but contains some uncommitted changes (as tagged as dirty). Linux kernel is licensed under GPLv2 which obligates anyone distributing a binary to provide complete corresponding source code as well.

> On the Raspberry Pi 5, both ports have moved from the SoC to the Raspberry Pi RP1 device ...

Yes, I know as commented at http://disq.us/p/2x7ofwq. Also suggested http://disq.us/p/2x862kj as this rather appears to be a drm_hwcomposer issue.

As far as I've understood this displays the bootanimation correctly but fails to display the Android UI, right? I don't have any DSI displays to test myself since I broke mine some time ago.

lewilou22, 2024-01-15:
Yes, Stuck on the main android boot logo. The anamation stops and is frozen on the screen.
Thanks for looking into it.

I can test things if needed

lewilou22, 2024-01-15:
not aure if this helps

https://github.com/emteria/kernel_brcm_rpi

JanKehren, 2024-01-17:
We accidentally pushed a local development branch instead of the production branch we used for our build. We corrected this today here https://github.com/emteria/kernel_brcm_rpi/tree/v6.1.61 and will add a solution to prevent this in the future. Thank you for pointing that out and helping us to improve on our end.  
We were surprised that the DSI Touchscreen is working for RPi5 because we cannot reproduce this. We tested the version we added to RPi Imager again and the Touchscreen is not working for us.  
Is it working for you out of the box when using our image?  
Which revision of the Touchscreen are you using?

lewilou22, 2024-01-17:
> We accidentally pushed a local development branch instead of the production branch we used for our build. We corrected this today here https://github.com/emteria/kernel_brcm_rpi/tree/v6.1.61 and [...]

I am using "Freenove 7 inch display for Pi DSI" it is a clone of the official pi screen. here is a photo of it running the image on my Pi5 from the raspberry imager 2023-12-09.

[image]

To keep the discussion on topic, here is an image of Android boot screen stuck on for AOSP14 on Pi 5 

[image]

JanKehren, 2024-01-18:
Is the screen working on both ports or just on one of the ports?

lewilou22, 2024-01-20:
> Is the screen working on both ports or just on one of the ports?

just one

simpz, 2024-03-08:
Ah I have just been trying this on my rpi5 with official touchscreen  for a few days, as per touchscreen instructions on the website, before I found this bug report. 

Will this be a relatively quick fix or will it likely take a long time? 
Not hassling, just wondering if I should start looking back at Emteria.

Darceus312, 2024-03-24:
I have the same issue with the official 7 inch touchscreen. It is stuck on the boot logo
Does someone know a fix for that?

jsln, 2024-04-14:
Hi @lewilou22 , @Darceus312 , what CAM/DISP port you connect the ribbon cable to? I can see an image in the official 7 inch touchscreen when I use stock raspbian image, however with this AOSP14 image I see a blank display (I am using CAM/DISP 1). The HDMI output is OK though.

simpz, 2024-04-15:
Personally I tried both CAM/DISP ports and neither work the AOSP14 image, they do work with Raspbian OS. 
That is the issue this bug is about.

My touch screen also works with EmteriaOS but this is also just a beta with it's own issue, no OTA updates being a big one for me.

WeebTrash-WeebTrash, 2024-04-20:
Just wanted to throw this out there. I was able to get the Official RPI 7" Display to display and stick on the logo. I also had the mini HDMI hooked up to a secondary Dell IPS monitor, and it appeared to have loaded lineage OS as the lock screen was showing and moving around the screen. I did also notice that i did not have any USB input. My mouse and keyboard were not working.

fraubush, 2024-05-05:
I am running into the same issue myself. Trying to get my Raspberry Pi 5 8Gb working with the official 7" touchscreen running AOSP14. This same setup (previous AOSP) was working for me with a 4B pi. Using @KonstaT's AOSP14 on the pi5 works perfectly for me over HDMI but when I put it on the official touchscreen I only get to the Android splash screen on the display itself.  I can still VNC into the pi when the display is stuck on the splash screen so I know the UI is fully loading in the background but all I see on the official 7" display is the Android logo.

fraubush, 2024-05-06:
Following up on my previous post about the issues with the official 7" touchscreen on my Raspberry Pi 5, I've managed to extract and run some relevant error logs through ChatGPT to help pinpoint the specific problems. Here are the critical errors that appear to be directly related to the display issues:

1. **DRM KMS Plan Failure:**
   - Error: `Failed to create DrmKmsPlan`
   ChatGPT: This suggests a problem in setting up the necessary display configurations at the kernel level.

2. **Hardware Composer Failure:**
   - Error: `presentAndGetReleaseFences: present failed for display 0: BadConfig (1)`
  ChatGPT:  This indicates that there's a configuration mismatch or error when the system tries to present the display output.

These errors are recurring and seem to pinpoint where things are going wrong with the official touchscreen display setup under AOSP14 on the Pi 5. Any insights or suggestions on how to address these specific errors would be greatly appreciated. 

@KonstaT, if you have any updates or patches that might help resolve these issues, please let us know!

fraubush, 2024-05-06:
Understood and thanks for the response. I have a feeling fixing it myself might be over my head but I'll do some reading and see if I can get anywhere. A higher resolution official display would be great. Thanks again

jsln, 2024-05-06:
@fraubush , can you please tell me what display port are you using? I am using DISP1 and I cannot even see the splash screen.

fraubush, 2024-05-06:
> @fraubush , can you please tell me what display port are you using? I am using DISP1 and I cannot even see the splash screen.

I am also using DISP1. Did you try toggling on 'Touchscreen' under the Raspberry Pi settings?

jsln, 2024-05-19:
@fraubush , I have just compiled the latest (android-14.0.0_r34) and I don´t see Raspberry Pi settings in Settings -> System. I can enable developer menu but nothing rpi specific in System. Are you able to open that? Is there any built time configuration needed?

Beridione, 2024-07-27:
Hello, i wanted to ask if anyone found a workaround for this problem? My RPI5 is stays stuck on an blackscreen after the boot logo on the DSI display.

KonstaT, 2024-09-03:
Should be fixed with https://github.com/raspberrypi/linux/pull/6314 which I've backported to Raspberry Vanilla's 6.1 kernel.

Updated binaries are available at https://github.com/raspberry-vanilla/android_device_brcm_rpi4-kernel/tree/android-14.0 & https://github.com/raspberry-vanilla/android_device_brcm_rpi5-kernel/tree/android-14.0 or you can compile one yourself https://github.com/raspberry-vanilla/android_kernel_manifest/tree/android-14.0.

I'll close this when someone who has to hardware confirms this resolves this issue.

Beridione, 2024-09-03:
> Should be fixed with [raspberrypi/linux#6314](https://github.com/raspberrypi/linux/pull/6314) which I've backported to Raspberry Vanilla's 6.1 kernel.
> 
> Updated binaries are available at https://github.com/raspberry-vanilla/android_device_brcm_rpi4-kernel/tree/android-14.0 & https://github.com/raspberry-vanilla/android_device_brcm_rpi5-kernel/tree/android-14.0 or you can compile one yourself https://github.com/raspberry-vanilla/android_kernel_manifest/tree/android-14.0.
> 
> I'll close this when someone who has the DSI display hardware confirms this resolves this issue.

Sorry i am new to this, can you tell me a good tutorial how i would use these files to update please?

KonstaT, 2024-09-05:
This should work on the latest AOSP 15 release for Pi 5.
https://konstakang.com/devices/rpi5/AOSP15/

Please see the FAQ on above page on how to use the DSI display on my personal builds.

I'd still appreciate if someone who has the hardware would also test this on AOSP 14 with 6.1 kernel.

MarshM4ll0w, 2024-10-01:
> I'd still appreciate if someone who has the hardware would also test this on AOSP 14 with the recent 6.1 kernel as commented above.

I have a pi 5, the official 7" touchscreen (v1.1), and am trying to use AOSP 14.  I am running the ASOP 14 img as found in _AOSP14-20240704-KonstaKANG-rpi5.zip_.  I can verify that the binaries linked above have resolved the issues with the official 7" touch screen.

When running the ASOP 14 pi5 image as it exists within the current zip, I get the same result as many above.  DISP 1 port displays the android splash screen and that is as far as it gets.

After removing the SD card from the pi, and overwriting the files in the boot partition with the updated binaries ( [https://github.com/raspberry-vanilla/android_device_brcm_rpi5-kernel/tree/android-14.0](https://github.com/raspberry-vanilla/android_device_brcm_rpi5-kernel/tree/android-14.0) ), the issue is gone and the touchscreen is now working as expected.  I have not experienced any negative side effects or seen any errors thus far.  I think this can be considered fixed and the changes can be merged into a new asop14-pi5 release.

Thanks @KonstaT !!

KonstaT, 2024-10-05:
@MarshM4ll0w Thank you for confirming.

The fix has been merged for quite some time. It will be included if/when there are any Android 14 based releases.

closed 2024-10-05 by KonstaT