raspberry-vanilla/android_local_manifest #60: Unable to goto TWRP recovery partition using "reboot recovery"

tvek, 2024-01-14:


tvek, 2024-01-14:
As an alternative, I have tried the TWRP recommendation proposed by the KonstaT.
https://konstakang.com/devices/rpi4/AOSP14/

I have build the latest TWRP package using [TWRP RPi Github URL](https://github.com/twrp-raspberry) and kept in /boot partition with name as ramdisk-recovery.img. But I was able to login TWRP by modifying the config.txt manually. But it was showing the error as /data,/system,...etc partition was not found. 

I was able to see my /system, /boot partitions were present inside /dev/block/mmcblk0p1...p4.

Please let me know if I need to make any modifications.

tvek, 2024-01-14:
Issue resolved after changing the TWRP-Rpi/device/brcm/rpi/recovery.fstab with /dev/block/mmcblk0p*

closed 2024-01-14 by tvek

mbt28, 2024-11-19:
Hello 

How did you get which mmcblk0p* refers to which mounting point? I have the same issue.

#<src>                       <mnt_point>  <type>  <mnt_flags and options>                             <fs_mgr_flags>
/dev/block/by-name/boot      /boot        vfat    ro                                                  wait,backup=1,wipeingui
/dev/block/by-name/system    /system      ext4    ro,barrier=1                                        wait,backup=1,wipeingui
/dev/block/by-name/vendor    /vendor      ext4    ro,barrier=1                                        wait,backup=1,wipeingui
/dev/block/by-name/userdata  /data        ext4    noatime,nosuid,nodev,nomblk_io_submit,errors=panic  wait,check,formattable,encryptable=footer,length=-16384,quota