raspberry-vanilla/android_local_manifest #63: AOSP 14 Build error related to gbm_mesa_driver

remarkked, 2024-01-30:
PLATFORM_VERSION=14
TARGET_PRODUCT=aosp_rpi4_car
TARGET_BUILD_VARIANT=userdebug
Environment has been established 100% properly twice.

```
FAILED: out/soong/.intermediates/external/boringssl/libcrypto/android_vendor.34_arm64_armv8-a_cortex-a72_shared/libcrypto.so.abidiff
(prebuilts/clang-tools/linux-x86/bin/header-abi-diff -target-version current -allow-unreferenced-changes -allow-unreferenced-elf-symbol-changes -lib libcrypto -arch arm64 -o out/soong/.intermediates/external/boringssl/libcrypto/android_vendor.34_arm64_armv8-a_cortex-a72_shared/libcrypto.so.abidiff -new out/soong/.intermediates/external/boringssl/libcrypto/android_vendor.34_arm64_armv8-a_cortex-a72_shared/libcrypto.so.lsdump -old prebuilts/abi-dumps/vndk/34/64/arm64/source-based/libcrypto.so.lsdump)|| (echo 'error: Please update ABI references with: $ANDROID_BUILD_TOP/development/vndk/tools/header-checker/utils/create_reference_dumps.py -l libcrypto' && (mkdir -p $DIST_DIR/abidiffs && cp out/soong/.intermediates/external/boringssl/libcrypto/android_vendor.34_arm64_armv8-a_cortex-a72_shared/libcrypto.so.abidiff $DIST_DIR/abidiffs/) && exit 1)
******************************************************
error: libcrypto's ABI has EXTENDING CHANGES. Please check compatibility report at: out/soong/.intermediates/external/boringssl/libcrypto/android_vendor.34_arm64_armv8-a_cortex-a72_shared/libcrypto.so.abidiff
******************************************************
error: Please update ABI references with: $ANDROID_BUILD_TOP/development/vndk/tools/header-checker/utils/create_reference_dumps.py -l libcrypto
16:05:23 ninja failed with: exit status 1

#### failed to build some targets (04:50:46 (hh:mm:ss)) ####
``` 

I faced the same fail building default clean AOSP 14 without raspberry-vanilla project, but updating ABI references as suggested did the trick and build was successfully completed.

But with raspberry-vanilla project source code included, attempt to update ABI references fails building gbm_mesa_driver:

```
FAILED:
external/minigbm/gbm_mesa_driver/Android.mk: error: "libgbm_mesa_wrapper (SHARED_LIBRARIES android-arm) missing libgbm_mesa (SHARED_LIBRARIES android-arm)"
You can set ALLOW_MISSING_DEPENDENCIES=true in your environment if this is intentional, but that may defer real problems until later in the build.
build/make/core/main.mk:1135: error: exiting from previous errors.
16:58:10 ckati failed with: exit status 1
Traceback (most recent call last):
  File "development/vndk/tools/header-checker/utils/create_reference_dumps.py", line 228, in <module>
    main()
  File "development/vndk/tools/header-checker/utils/create_reference_dumps.py", line 219, in main
    num_processed = create_source_abi_reference_dumps_for_all_products(args)
  File "development/vndk/tools/header-checker/utils/create_reference_dumps.py", line 148, in create_source_abi_reference_dumps_for_all_products
    exclude_tags)
  File "development/vndk/tools/header-checker/utils/create_reference_dumps.py", line 53, in make_libs_for_product
    exclude_tags)
  File "/home/remark/aosp14_rpi/development/vndk/tools/header-checker/utils/utils.py", line 169, in make_libraries
    exclude_tags, build=True)
  File "/home/remark/aosp14_rpi/development/vndk/tools/header-checker/utils/utils.py", line 264, in read_lsdump_paths
    make_targets(product, variant, [lsdump_paths_file_path])
  File "/home/remark/aosp14_rpi/development/vndk/tools/header-checker/utils/utils.py", line 157, in make_targets
    subprocess.check_call(make_cmd, cwd=AOSP_DIR)
  File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['build/soong/soong_ui.bash', '--make-mode', '-j', 'TARGET_PRODUCT=aosp_arm', 'TARGET_BUILD_VARIANT=userdebug', 'out/target/product/generic/lsdump_paths.txt']' returned non-zero exit status 1.
``` 

Found no solution yet. Please help

remarkked, 2024-02-01:
Update:
Built successfully after manually
```
build/soong/soong_ui.bash --make-mode -j TARGET_PRODUCT=aosp_arm TARGET_BUILD_VARIANT=userdebug
``` 
But not sure it's a proper way.

Also, AOSP 14 image from site doesn't boot at all: reboot after rainbow screen after reboot and so on. In FAQ underneath you said the reason is defined resolution, so I tried each possible resolution: 720x480, 1920x1080, 3440x1440, also tried deleting "resolution.txt" as my display reports 3440x1440 as recommended resolution, nothing works. Same issue happens to built AOSP 14 after first successful boot and rebooting.

KonstaT, 2024-02-21:
`aosp_arm` is not a valid build target for Raspberry Vanilla project (and 32-bit is probably something you shouldn't be targeting in 2024 in any case).

closed 2024-02-21 by KonstaT

remarkked, 2024-02-24:
@KonstaT ofc I just tried your lunch aosp_rpi4_car which targets 64-bit rpi, but somewhere during updating ABI references the exact command that I successfully ran manually ruins this. I didn't change the target, just ran the command I saw in error trace

KonstaT, 2024-02-24:
> I faced the same fail building default clean AOSP 14 without raspberry-vanilla project

So, it is not any issue related to Raspberry Vanilla so this is not the correct place for this.

Quick Google search finds
https://xdaforums.com/t/major-bug-in-android-14-breaks-building-aosp-on-debian-ubuntu.4636394/
https://forum.unity.com/threads/major-bug-in-android-14-breaks-building-aosp-for-debian-ubuntu-users.1506443/

I'm building this fine on Ubuntu 22.04 LTS anyway. Something to do with python/python2 deprecation? `sudo apt install python-is-python3` would help?