raspberry-vanilla/android_local_manifest #65: Can't overclock the GPU on a Pi 4B

erayrafet, 2024-02-17:
Well, running Android TV on the Pi 4B and overclocking the CPU and video decoders seems functional but v3d or gpu core clocks won't go above 450 MHz no matter what.
I did the config.txt edits on both Android TV 13 and 14.
[image]
[image]

KonstaT, 2024-02-21:
This is something that is entirely handled on the firmware level and there's nothing Android specific here. Please refer to the Raspberry Pi documentation on the subject.
https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking-options

closed 2024-02-21 by KonstaT

erayrafet, 2024-02-21:
Sure but overclocking on PiOS worked. That's what's got me puzzled.

KonstaT, 2024-02-21:
Something that could depend on e.g. Linux kernel version (6.1 vs. 5.15 in Android 14 & 13, respectively) and the boot firmware that matches the kernel version on my releases. I know that some of the options in your screenshot (e.g. `over_voltage`) are deprecated with recent Pi firmware.

erayrafet, 2024-02-21:
I can't find any documentation about the firmware update changelogs but the results haven't changed in either Android 13 or 14. PiOS is on kernel 6.1 but the config.txt is in /boot/firmware now.