raspberry-vanilla/android_local_manifest #71: Build “user” not working

alemig97, 2024-04-08:
Hi Konsta,

After running the command ‘lunch aosp_rpi4_car-user’, I created a new image, but it doesn’t work on the Raspberry Pi. Is a build supported for Raspberry Pi 4?

KonstaT, 2024-04-08:
Please follow the instructions on how to build.
https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0

No, `user`build variant is not supported as it disables various things that are needed for development purposes. IIRC this also requires enforcing SELinux policies.
https://source.android.com/docs/setup/create/new-device#build-variants

closed 2024-04-08 by KonstaT

alemig97, 2024-04-08:
“Thank you for the quick response. Regarding aosp_rpi4_car, only userdebug is available, correct? Is it possible to modify the build in any way?

KonstaT, 2024-04-09:
Well, yes, you have the source code so you can modify the build in any way you like.

alemig97, 2024-04-09:
Thank you for the response, but what are the builds supported by Android? userdebug and eng? Based on what you were saying, "user" build no, right?

KonstaT, 2024-04-09:
AOSP has `user`, `userdebug`, and `eng` build variants as documented in the previous link.

Raspberry Pi is not a device that Google supports in AOSP. That would be the purpose of Raspberry Vanilla project. As mentioned in this project's documentation (also previously linked), you would use `userdebug` build variant.

alemig97, 2024-04-15:
I am interested in creating the project with a user build configuration. Could you please provide guidance on where I can modify the build configuration (can we discuss which files need to be modified for the configuration for rpi4)?
Your help on this matter would be greatly.

KonstaT, 2024-04-15:
> I am interested in creating the project with a user build configuration.

Why?

> Could you please provide guidance on where I can modify the build configuration (can we discuss which files need to be modified for the configuration for rpi4)?

As mentioned earlier, main reason `user` build variant won't boot is because it doesn't allow permissive SELinux (https://cs.android.com/android/platform/superproject/+/master:system/core/init/Android.bp;l=136-137).
You would need to fix the SELinux policies (https://github.com/raspberry-vanilla/android_device_brcm_rpi4/tree/android-14.0/sepolicy) and enforce SELinux (https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-14.0/BoardConfig.mk#L70). You would use the `userdebug` build variant for the SELinux audit logs of course.

Further reading.
https://source.android.com/docs/security/features/selinux
https://source.android.com/docs/security/features/selinux/implement

alemig97, 2024-04-17:
> Why?

I want to have a more performance-optimized version for rpi4.

KonstaT, 2024-04-17:
In practice it doesn't performance in any way and no, `userdebug` doesn't contain extra logging (just ADB and serial console are enabled by default).