raspberry-vanilla/android_local_manifest #72: Build error android 14

rameshandroid, 2024-04-08:
Hello konstakang,
Thanks for build and documentation. 
we followed steps which are mentioned in [](https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0)

**step-1: we clone the repo using below commands** 
repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r30 --depth=1
curl -o .repo/local_manifests/manifest_brcm_rpi.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-14.0/manifest_brcm_rpi.xml --create-dirs
curl -o .repo/local_manifests/remove_projects.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-14.0/remove_projects.xml

**step-2 : repo sync**
repo synced successfully
**step-3 : source  build/envsetup.sh**
this is also success
**step -4 : lunch aosp_rpi4-ap1a-userdebug**

build/make/core/version_util.mk:29: warning: Invalid TARGET_PLATFORM_VERSION 'userdebug', must be one of
In file included from build/make/core/config.mk:359:
In file included from build/make/core/envsetup.mk:68:
In file included from build/make/core/version_defaults.mk:110:
build/make/core/version_util.mk:30: error: UP1A UP1B UP2A UP2B UD1A UD1B UD2A UD2B UQ1A UQ1B UQ2A UQ2B UQ3A UQ3B VP1A.
17:28:43 dumpvars failed with: exit status 1

build is not working.any idea whats going wrong here?

KonstaT, 2024-04-09:
Are you sure you have everything up-to-date? I just pushed `android-14.0.0_r30` yesterday. Try running `repo sync` again. I did double check and everything should be up-to-date in Raspberry Vanilla repos.

Also note that `android-14.0.0_r30` doesn't currently boot on `rpi4`. You would need to use `android-14.0.0_r22` instead.
https://github.com/raspberry-vanilla/android_local_manifest/issues/70
https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0.0_r22

jmfriedt, 2024-04-13:
I compiled for both the RPi4 and RPi5 according to these instructions: none seem to be executing, although the image is generated fine and the bootloader is properly detected by each RPi. I downloaded the AOSP14-20240104-KonstaKANG-rpi5.zip image from  https://konstakang.com/devices/rpi5/AOSP14/ and it executed fine on the RPi5, so the SD card, SD card reader, the dd procedure of the .img file and the RPi5 are functional.
Thanks

KonstaT, 2024-04-15:
@jmfriedt Not sure how it relates to this specific issue. At least it builds unlike in the original report.

It's known that AOSP14 QPR2 (`android-14.0.0_r29` and forward) fails to boot on `rpi4` build targets.  (https://github.com/raspberry-vanilla/android_local_manifest/issues/70). `rpi5` build targets should work just fine but impossible to say about your specific issue without seeing any logs.

KonstaT, 2024-05-06:
Closing as there's been no update in a month. Assuming resolved by following the build instructions.
https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0

closed 2024-05-06 by KonstaT