raspberry-vanilla/android_local_manifest #74: libcamera-hal.so missing

racsnet, 2024-04-11:
I try to build aosp_rpi5_car-ap1a-userdebug it stops with following error:

FAILED: ninja: 'media/android-auto/data/target/product/rpi5/obj_arm/MESON_LIBCAMERA/install/usr/local/lib/libcamera-hal.so', needed by '/media/android-auto/data/target/product/rpi5/obj_arm/SHARED_LIBRARIES/camera.libcamera_intermediates/LINKED/camera.libcamera.so', missing and no known rule to make it
14:43:13 ninja failed with: exit status 1
                                                                                               
#### failed to build some targets (07:45 (mm:ss)) ####

build OS: Ubuntu 22.04

KonstaT, 2024-04-11:
Did you follow the instruction on how to setup build environment including installing meson? Which meson version are you using?

meson version >= 0.60 is required to build libcamera.

racsnet, 2024-04-11:
here is my apt-get and pip:
`apt-get install git-core gnupg flex bison build-essential zip curl zlib1g-dev libc6-dev-i386 libncurses5 x11proto-core-dev libx11-dev lib32z1-dev libgl1-mesa-dev libxml2-utils xsltproc unzip fontconfig meson python3 python3-setuptools bc coreutils dosfstools e2fsprogs fdisk kpartx mtools ninja-build pkg-config python3-pip`

`pip3 install meson mako jinja2 ply pyyaml dataclasses`

meson -v gives me 0.61.2

KonstaT, 2024-04-15:
Have you set `OUT_DIR` parameter as `/media/android-auto/data/` seems odd? Mesa and libcamera are built with meson build system which doesn't support using custom output directory.

Also, I have meson 1.3.0 on Ubuntu 22.04 LTS.

KonstaT, 2024-05-06:
Closing as there's been no update in a month. Assuming resolved by not setting custom `OUT_DIR`.