raspberry-vanilla/android_local_manifest #76: android-14.0.0_r30 for RPi5 not booting

jmfriedt, 2024-04-16:
I should have opened a new issue rather than answering the unrelated https://github.com/raspberry-vanilla/android_local_manifest/issues/72, apologies.

I have followed all the steps using the shallow clone 
* repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r30 --depth=1
* curl -o .repo/local_manifests/manifest_brcm_rpi.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-14.0/manifest_brcm_rpi.xml --create-dirs
* curl -o .repo/local_manifests/remove_projects.xml -L https://raw.githubusercontent.com/raspberry-vanilla/android_local_manifest/android-14.0/remove_projects.xml
* repo sync
* . build/envsetup.sh
* lunch aosp_rpi5-ap1a-userdebug
* make bootimage systemimage vendorimage -j$(nproc)
* ./rpi5-mkimg.sh
and the ``out/target/product/rpi5/RaspberryVanillaAOSP14-20240416-rpi5.img`` is correctly generated. However when starting the RPi5 with this image, the screen connected to HDMI0 remains black and the serial port connected to GPIO14,15 remains silent (no boot message).
When repeating with the image AOSP14-20240104-KonstaKANG-rpi5.zip found at https://konstakang.com/devices/rpi5/AOSP14/ I do get the root shell console on the TTY and the screen displays the Android splash screen before displaying the time and the icons. I unfortunately cannot get more log messages for debugging.
Thanks.

KonstaT, 2024-04-16:
Serial console is on the dedicated UART connector instead of GPIO14/15. You would need to revert https://github.com/raspberry-vanilla/android_device_brcm_rpi5/commit/71b154a8bb0facac24c8a27a0bfff93d3fe1aaf5 to move it back to GPIO14/15.

KonstaT, 2024-04-16:
Please run `logcat` in the shell for Android system logs as the `dmesg` output (Linux kernel logs) doesn't tell much.

Also check the wiki (https://github.com/raspberry-vanilla/android_local_manifest/wiki) if you're using a display with resolution other than 1080p or booting from an USB.

jmfriedt, 2024-04-16:
I had not realized I had been given a prompt as the messages kept on repeating in the console, I had not thought of hitting [Enter] to be shown the prompt. 
I am booting the RPi5 from the SD card and disconnecting the screen from the HDMI port leads to the same result.
This ``logcat`` command is extremely verbose (unless it triggers a positive feedback loop of error messages). I have uploaded the ``minicom`` log of the terminal connected to the serial port to http://65.21.30.42/logcat.dump
I am not sure you will be able to decipher anything from this log as I believe it mixes Linux kernel and Android system messages. I am only familiar with the former.
Thanks.

KonstaT, 2024-04-16:
Here's your crash:
```
01-01 00:00:19.551   361   625 W EGL-MAIN: MESA-LOADER: failed to open v3d: dlopen failed: library "/vendor/lib64/dri/v3d_dri.so" not found (search paths /vendor/lib64/dri, suffix _dri)
01-01 00:00:19.551   361   625 W EGL-MAIN: Failed to open any DRM device
01-01 00:00:19.551   361   625 I MESA    : Using IMapper v4 API
01-01 00:00:19.552   361   625 W EGL-MAIN: Failed to open any DRM device
01-01 00:00:19.552   361   625 W libEGL  : eglInitialize(0xb400007c33a1c930) failed (EGL_NOT_INITIALIZED)
01-01 00:00:19.553   361   625 W RenderEngine: no suitable EGLConfig found, trying a simpler query
01-01 00:00:19.553   361   625 F RenderEngine: no suitable EGLConfig found, giving up
--------- beginning of crash
01-01 00:00:19.553   361   625 F libc    : Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 625 (surfaceflinger), pid 361 (surfaceflinger)
...
01-01 00:00:21.424   633   633 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
01-01 00:00:21.424   633   633 F DEBUG   : Build fingerprint: 'Raspberry/aosp_rpi5/rpi5:14/AP1A.240405.002/eng.jmfrie.20240416.101813:userdebug/test-keys'
01-01 00:00:21.424   633   633 F DEBUG   : Revision: '0'
01-01 00:00:21.424   633   633 F DEBUG   : ABI: 'arm64'
01-01 00:00:21.424   633   633 F DEBUG   : Timestamp: 1970-01-01 00:00:20.701542769+0000
01-01 00:00:21.424   633   633 F DEBUG   : Process uptime: 6s
01-01 00:00:21.424   633   633 F DEBUG   : Cmdline: /system/bin/surfaceflinger
01-01 00:00:21.424   633   633 F DEBUG   : pid: 361, tid: 625, name: surfaceflinger  >>> /system/bin/surfaceflinger <<<
01-01 00:00:21.424   633   633 F DEBUG   : uid: 1000
01-01 00:00:21.424   633   633 F DEBUG   : tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
01-01 00:00:21.424   633   633 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
01-01 00:00:21.424   633   633 F DEBUG   : Abort message: 'no suitable EGLConfig found, giving up'
01-01 00:00:21.424   633   633 F DEBUG   :     x0  0000000000000000  x1  0000000000000271  x2  0000000000000006  x3  0000007b8f413320
01-01 00:00:21.424   633   633 F DEBUG   :     x4  7568661f2b636d74  x5  7568661f2b636d74  x6  7568661f2b636d74  x7  7f7f7f7f7f7f7f7f
01-01 00:00:21.424   633   633 F DEBUG   :     x8  00000000000000f0  x9  0000007e244c4200  x10 0000000000000001  x11 0000007e2451aba0
01-01 00:00:21.424   633   633 F DEBUG   :     x12 0000007b8f413300  x13 0000000000000027  x14 0000007b8f4134b0  x15 0000000028e33b65
01-01 00:00:21.424   633   633 F DEBUG   :     x16 0000007e24593fc8  x17 0000007e24563160  x18 0000007b8e33c000  x19 0000000000000169
01-01 00:00:21.424   633   633 F DEBUG   :     x20 0000000000000271  x21 00000000ffffffff  x22 0000007b8f413f00  x23 b400007b93a16879
01-01 00:00:21.424   633   633 F DEBUG   :     x24 b400007d73a14a10  x25 0000007b8f413bb0  x26 0000007b8f413ef0  x27 00000000000fc000
01-01 00:00:21.424   633   633 F DEBUG   :     x28 00000000000fe000  x29 0000007b8f4133a0
01-01 00:00:21.424   633   633 F DEBUG   :     lr  0000007e2450be48  sp  0000007b8f413300  pc  0000007e2450be74  pst 0000000000001000
01-01 00:00:21.424   633   633 F DEBUG   : 10 total frames
01-01 00:00:21.424   633   633 F DEBUG   : backtrace:
01-01 00:00:21.424   633   633 F DEBUG   :       #00 pc 000000000006ae74  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: 68efd4bc2a1e51d40ffa7870daf0f520)
01-01 00:00:21.424   633   633 F DEBUG   :       #01 pc 000000000001125c  /system/lib64/liblog.so (__android_log_default_aborter+12) (BuildId: 2399ac8895bcb32ac78b028f29ff59a9)
01-01 00:00:21.424   633   633 F DEBUG   :       #02 pc 00000000000120e4  /system/lib64/liblog.so (__android_log_assert+292) (BuildId: 2399ac8895bcb32ac78b028f29ff59a9)
01-01 00:00:21.424   633   633 F DEBUG   :       #03 pc 000000000045a8a4  /system/bin/surfaceflinger (android::renderengine::skia::SkiaGLRenderEngine::chooseEglConfig(void*, int, bool)+500) (BuildId: ffc903e64ff399d39a6de7e5defd518a)
...
```

Something must have gone wrong with Mesa build in your image but I have no idea what and why. What Linux distro are you using to build and what meson version does it have?

jmfriedt, 2024-04-16:
The compilation sequence did not complain of a failed package. I am using Debian/GNU Linux testing, up to date, with 
```
$ meson --version
1.2.3
```
Thanks

KonstaT, 2024-04-17:
Are following libraries present in the image you've built?
```
/vendor/lib{64}/dri/*.so
/vendor/lib{64}/egl/*.so
/vendor/lib{64}/libgbm_mesa*.so
/vendor/lib{64}/libglapi.so
```

jmfriedt, 2024-04-17:
I am not familiar with the Android file system and how it is populated at boot.
After mounting the four partitions of the SD card in /dev/sdb[1-4], I cannot find any lib* in a vendor directory:
```
# ls /mnt/sdb*/vendor*/lib*
ls: cannot access '/mnt/sdb*/vendor*/lib*': No such file or directory
```
but on the running Android system I can find between all the kernel messages displaying continuously on the console:
```
$ ls -l /vendor/lib*
...
drwxr-xr-x 2 root shell    4096 2024-04-17 12:11 dri
drwxr-xr-x 2 root shell     4096 2024-04-17 12:11 egl
-rw-r--r-- 1 root root    56720 2024-04-16 11:45 libgbm_mesa.so
lrw-r--r-- 1 root root       14 2024-04-16 11:45 libgbm_mesa.so.1 -> libgbm_mesa.so
-rw-r--r-- 1 root root     4916 2024-04-16 11:46 libgbm_mesa_wrapper.so
-rw-r--r-- 1 root root   161988 2024-04-16 11:45 libglapi.so
lrw-r--r-- 1 root root       11 2024-04-16 11:46 libglapi.so.0 -> libglapi.so
console:/ $ ls -l /vendor/lib*/dri/*
-rw-r--r-- 1 root root  8657412 2024-04-16 11:45 /vendor/lib/dri/libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-16 11:46 /vendor/lib/dri/libgallium_dri.so.0 -> libgallium_dro
-rw-r--r-- 1 root root 13828616 2024-04-16 11:45 /vendor/lib64/dri/libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-16 11:46 /vendor/lib64/dri/libgallium_dri.so.0 -> libgallium_o
console:/ $ ls -l /vendor/lib*/egl/*
-rw-r--r-- 1 root root 132208 2024-04-16 11:45 /vendor/lib/egl/libEGL_mesa.so
lrw-r--r-- 1 root root     14 2024-04-16 11:46 /vendor/lib/egl/libEGL_mesa.so.1 -> libEGL_mesa.so
-rw-r--r-- 1 root root  13904 2024-04-16 11:45 /vendor/lib/egl/libGLESv1_CM_mesa.so
lrw-r--r-- 1 root root     20 2024-04-16 11:46 /vendor/lib/egl/libGLESv1_CM_mesa.so.1 -> libGLESv1_CMo
-rw-r--r-- 1 root root  34944 2024-04-16 11:45 /vendor/lib/egl/libGLESv2_mesa.so
lrw-r--r-- 1 root root     17 2024-04-16 11:46 /vendor/lib/egl/libGLESv2_mesa.so.2 -> libGLESv2_mesa.o
-rw-r--r-- 1 root root 230976 2024-04-16 11:45 /vendor/lib64/egl/libEGL_mesa.so
lrw-r--r-- 1 root root     14 2024-04-16 11:46 /vendor/lib64/egl/libEGL_mesa.so.1 -> libEGL_mesa.so
-rw-r--r-- 1 root root 133056 2024-04-16 11:45 /vendor/lib64/egl/libGLESv1_CM_mesa.so
lrw-r--r-- 1 root root     20 2024-04-16 11:46 /vendor/lib64/egl/libGLESv1_CM_mesa.so.1 -> libGLESv1_o
-rw-r--r-- 1 root root 133056 2024-04-16 11:45 /vendor/lib64/egl/libGLESv2_mesa.so
lrw-r--r-- 1 root root     17 2024-04-16 11:46 /vendor/lib64/egl/libGLESv2_mesa.so.2 -> libGLESv2_meso
...
```
so yes I believe these directories are all populated. I tried also https://github.com/raspberry-vanilla/android_local_manifest/wiki/HDMI-display and changing the display resolution to one I am sure to be found on the screen I am connected to but same result.
Thanks

KonstaT, 2024-04-17:
Symlinks are missing for some reason. It should look like this:
```
console:/ # ls -l /vendor/lib*/dri/*                                           
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/armada-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/exynos_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/gm12u320_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/hdlcd_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/hx8357d_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/ili9163_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/ili9225_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/ili9341_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/ili9486_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/imx-dcss_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/imx-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/imx-lcdif_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/ingenic-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/kirin_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/komeda_dri.so -> libgallium_dri.so
-rw-r--r-- 1 root root  8640440 2024-04-15 19:47 /vendor/lib/dri/libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/libgallium_dri.so.0 -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/mali-dp_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/mcde_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/mediatek_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/meson_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/mi0283qt_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/mxsfb-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/panel-mipi-dbi_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/pl111_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/rcar-du_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/repaper_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/rockchip_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/st7586_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/st7735r_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/sti_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/stm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/sun4i-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/udl_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/v3d_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib/dri/vc4_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/armada-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/exynos_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/gm12u320_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/hdlcd_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/hx8357d_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/ili9163_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/ili9225_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/ili9341_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/ili9486_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/imx-dcss_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/imx-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/imx-lcdif_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/ingenic-drm_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/kirin_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/komeda_dri.so -> libgallium_dri.so
-rw-r--r-- 1 root root 13578824 2024-04-15 19:47 /vendor/lib64/dri/libgallium_dri.so
[... cut ...]
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/udl_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/v3d_dri.so -> libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-15 19:47 /vendor/lib64/dri/vc4_dri.so -> libgallium_dri.so
```

As in your log:
`01-01 00:00:19.551   361   625 W EGL-MAIN: MESA-LOADER: failed to open v3d: dlopen failed: library "/vendor/lib64/dri/v3d_dri.so" not found (search paths /vendor/lib64/dri, suffix _dri)`

jmfriedt, 2024-04-17:
Good catch, I see the error now. However when trying to symlink manually 
```
$ console:/vendor/lib/dri $ ln -s libgallium_dri.so armada-drm_dri.so   
ln: cannot create symbolic link from 'libgallium_dri.so' to 'armada-drm_dri.so': Read-only file system
$ console:/vendor/lib64/dri $ ln -s libgallium_dri.so v3d_dri.so  
ln: cannot create symbolic link from 'libgallium_dri.so' to 'v3d_dri.so': Read-only file system
```
is there a startup script where I can create the symlink?
Thanks.

jmfriedt, 2024-04-17:
awkwardly enough in the build tree I can see
```
$ ls -l  ./out/target/product/rpi5/vendor/lib64/dri
...
-rwxr-xr-x 1 jmfriedt jmfriedt 13828616 Apr 16 11:45 libgallium_dri.so
...
lrwxrwxrwx 1 jmfriedt jmfriedt       17 Apr 16 11:43 v3d_dri.so -> libgallium_dri.so
```
so the symlinks exist on the host used for cross compiling but not on the target file system!

KonstaT, 2024-04-17:
Symlinks should be generated at build time (https://github.com/raspberry-vanilla/android_external_mesa3d/blob/android-14.0/android/mesa3d_cross.mk#L315-L320). Likely something with your build environment if that has failed.

Might be easier just to mount the sdcard on your computer if you want to add the symlinks manually.

On Android you would need to have rooted shell (which console is not by default - you could change [this line](https://android.googlesource.com/platform/system/core/+/refs/tags/android-14.0.0_r30/rootdir/init.rc#1276) to `user root`) and mount the partition writable (`mount -o rw,remount /vendor`).

jmfriedt, 2024-04-17:
I found the ``lib64`` dir being mounted in ``/vendor`` and created manually the symlink
```
-rw-r--r-- 1 root root 13828616 2024-04-16 11:45 libgallium_dri.so
lrw-r--r-- 1 root root       17 2024-04-16 11:46 libgallium_dri.so.0 -> libgallium_dri.so
lrwxrwxrwx 1 root root       17 2024-04-17 17:47 v3d_dri.so -> libgallium_dri.so
```
but still no luck. The new ``logcat`` is at http://65.21.30.42/newlogcat.dump, I cannot figure out how to read it.

KonstaT, 2024-04-17:
Rather try `make vendorimage -j$(nproc)` and create a new flashable image. I think you also need symlink for vc4_dri.so.

It's still crashing with the display:
```
01-01 00:00:35.071   347   347 I [minigbm:gbm_mesa_internals.cpp(233)]: Found GPU v3d
01-01 00:00:35.071   347   347 I [minigbm:gbm_mesa_internals.cpp(249)]: GPU require KMSRO entry, searching for separate KMS driver...
01-01 00:00:35.071   347   347 I [minigbm:gbm_mesa_internals.cpp(255)]: Found KMS dev vc4
01-01 00:00:35.073   347   347 E GBM-MESA-WRAPPER: Unable to create gbm device
01-01 00:00:35.073   347   347 E [minigbm:gbm_mesa_internals.cpp(297)]: Unable to create gbm_mesa driver
01-01 00:00:35.074   347   347 E [minigbm:gbm_mesa_internals.cpp(359)]: Failed to init gbm driver
01-01 00:00:35.074   347   347 E android.hardware.graphics.allocator@4.0-service.minigbm_gbm_mesa: Failed to create bo.
01-01 00:00:35.074   914   914 E GraphicBufferAllocator: Failed to allocate (1600 x 900) layerCount 1 format 1 usage 1b00: 5
```

Also it seems you've forced resolution to 1600x900 with `debug.drm.mode.force` property. It would be better try first with more usual display setup (1920x1080 resolution). Your display seems to support it at least.
```
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1600x900@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: add new mode 1600x900@60 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@59.940201 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@30.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@30.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@29.970100 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@50.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1920x1080@25.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1152x864@75.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1280x720@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1280x720@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1280x720@59.940201 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1280x720@50.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1024x768@75.028580 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 1024x768@60.003838 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 800x600@75.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 800x600@60.316540 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x576@50.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x576@25.000000 for display in connector HDMI-A-
1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x480@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x480@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x480@59.940060 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x480@59.940060 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x480@59.940060 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x480@30.001110 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x480@29.970030 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 640x480@75.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 640x480@60.000000 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 640x480@59.940475 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 640x480@59.940475 for display in connector HDMI-A-1
01-01 00:00:34.805   348   348 D hwc-drm-connector: supported mode 720x400@70.081665 for display in connector HDMI-A-1
01-01 00:00:34.806   348   348 I hwc-resource-manager: Attaching connector HDMI-A-1
01-01 00:00:34.806   348   348 I hwc-drm-display-pipeline: Ignoring cursor plane 280
01-01 00:00:34.806   348   348 I hwc-drm-two: Attaching pipeline 'HDMI-A-1' to the display #0 (Primary)
01-01 00:00:34.806   348   348 I hwc-display-configs: set mode 1600x900@60Hz for HEADLESS-MODE
01-01 00:00:34.806   348   348 I hwc-drm-connector: force mode 1600x900@0Hz for display in connector HDMI-A-1
```

jmfriedt, 2024-04-18:
I tried ``make clean`` before ``make vendorimage -j$(nproc)`` since https://github.com/sonyxperiadev/bug_tracker/issues/253 reports on missing symlinks on non-clean builds, and reran another ``make vendorimage -j$(nproc)`` but in all cases
```
# ls -l /mnt/sdb3/lib64/dri/
total 13340
-rw-r--r--. 1 root root 13828616 Apr 18 07:31 libgallium_dri.so
lrw-r--r--. 1 root root       17 Apr 18 07:31 libgallium_dri.so.0 -> libgallium_dri.so
```
symlinks remain absent.
Thanks.

Samaelson, 2024-04-18:
Am facing exactly the same issue with _r30 on rpi5. With _r22 it was working fine.

jmfriedt, 2024-04-18:
Great, I feel less lonely now.
I got it to work thanks to all the explanation (and in the mean time I learned a bit about Android compilation and filesystem): after ``dd if=out/target/product/rpi5/RaspberryVanillaAOSP14-20240418-rpi5.img of=/dev/sdb bs=8M status=progress`` I mount ``/dev/sdb3`` of the SD card into ``/mnt/sdb3``  and
```
cp -dr out/target/product/rpi5/vendor/lib64/egl/* /mnt/sdb3/lib64/egl/
cp -dr out/target/product/rpi5/vendor/lib/egl/* /mnt/sdb3/lib/egl/
cp -dr out/target/product/rpi5/vendor/lib/dri/* /mnt/sdb3/lib/dri/
cp -dr out/target/product/rpi5/vendor/lib64/dri/* /mnt/sdb3/lib64/dri/
```
and now the RPi5 completes the boot sequence and displays the Android splash screen on the HDMI port.
Thanks a lot.

Samaelson, 2024-04-18:
Thank you, I will try it out. But question still is, why these are missing in the image *-)

jabin-ma, 2024-04-21:
Sir, I'm having this issue too, any clues now?

KonstaT, 2024-04-25:
This is now fixed with https://github.com/raspberry-vanilla/android_external_mesa3d/commit/ab237fe4a15ccf3b466413e679bc987824f44b84 and https://github.com/raspberry-vanilla/android_external_mesa3d/commit/abacf7ccabbe88b03f0e188dcb1419c8f3ec9874.

Please sync your sources.

closed 2024-04-25 by KonstaT