raspberry-vanilla/android_local_manifest #79: Audio and Bluetooth not working when installing with PINN

k2helix, 2024-04-20:
I'm using [PINN](https://github.com/procount/pinn/) to be able to have two OSes installed in the same card. PINN includes LineageOS and its versions are avalaible to install without downloading it from https://konstakang.com/ (I'm aware there's a red "Do not mirror my builds" text, but I need to install more than one OS). I don't know if I should report this here or at PINN's issues.

I'm using a Raspberry Pi 5, and checked both on LOS 20 ATV and 21 ATV. There seems to be no HDMI audio, even when changing the audio output. Bluetooth does detect devices but when clicking the pair button will keep showing "Pairing" and ends not connecting to any device (tried with bluetooth headphones and controller). The same setup works flawlessly when in Raspbian.

If needed, I can provide logcat information (would be great if the log level and filter were specified)

k2helix, 2024-04-20:
I took first look at `adb logcat` and here's what's literally everywhere in the output:

```
04-20 20:31:03.893   227  1715 E audio_hw_rpi_hdmi: out_write err: No such device
04-20 20:31:03.915   227  1715 W StreamHAL: Error from HAL stream in function get_presentation_position: Operation not permitted
```

KonstaT, 2024-04-25:
This issue tracker is mainly intended for people who are building from Raspberry Vanilla source code.

I'm not supporting any multi-boot solutions. Please follow the installation instructions on my website and if you can reproduce the issues, you're welcome to report it.