raspberry-vanilla/android_local_manifest #83: Home UI not fully shown followed by reset with AOSP14 on Raspberry Pi 5

Samaelson, 2024-05-07:
Hello,

with latest build I observe the issue, that home UI comes up with black background and some submenus not working/shown.

After a while rpi5 runs a reset and reboots.

Anyone else has this issue?

[aosp_log.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/15235088/aosp_log.txt)

khvMX, 2024-05-07:
I have just built and tested out `android-14.0.0_r30` with `aosp_rpi5_car-ap1a-userdebug` (AAOS) on Raspberry Pi 5.
I am seeing similar behavior: boot animation completes fine, some System UI is shown, parts of the CarLauncher are seen, but incomplete. No apps can be started, but pi reacts fine to things like climate menu.
After a minute or so it reboots on its own.

@Samaelson the logs you got are not logcat. I believe we need logcat logs to see what keeps crashing on Java side and causes the reboot.

khvMX, 2024-05-07:
I have tried to connect to Pi via USB-A to USB-A cable, but do not see it under `adb devices` even when it's "booted".
Looking at other issues, it seems like we need to use UART to get `logcat`. I only have standard UART adapters with dupont cables, however Pi5 has this weird 3-pin connector, which I have just ordered. So it will take me some time until I can look at it.

In the meantime, @KonstaT, maybe you have a guess on what is missing? Are we doing something wrong, or was `aosp_rpi5_car-ap1a-userdebug` version simply never tested with `android-14.0.0_r30` and we're the first to try?

Samaelson, 2024-05-07:
New logs attached:
[aosp_log+cat.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/15237478/aosp_log%2Bcat.txt)

...
--------- beginning of crash
05-07 10:57:08.825  1037  1154 E AndroidRuntime: FATAL EXCEPTION: CarAudioService
05-07 10:57:08.825  1037  1154 E AndroidRuntime: Process: com.android.car, PID: 1037
05-07 10:57:08.825  1037  1154 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.util.SparseArray.size()' on a null object reference
05-07 10:57:08.825  1037  1154 E AndroidRuntime: 	at com.android.car.audio.CarAudioService.audioDevicesAdded(CarAudioService.java:3409)
05-07 10:57:08.825  1037  1154 E AndroidRuntime: 	at com.android.car.audio.CarAudioDeviceCallback.onAudioDevicesAdded(CarAudioDeviceCallback.java:42)
05-07 10:57:08.825  1037  1154 E AndroidRuntime: 	at android.media.AudioManager$NativeEventHandlerDelegate$1.handleMessage(AudioManager.java:10114)
05-07 10:57:08.825  1037  1154 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
05-07 10:57:08.825  1037  1154 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:232)
05-07 10:57:08.825  1037  1154 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:317)
05-07 10:57:08.825  1037  1154 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:68)
05-07 10:57:09.195     0     0 I binder  : undelivered transaction 13096, process died.
05-07 10:57:08.997  1203  1252 E AndroidRuntime: FATAL EXCEPTION: CarAudioService
05-07 10:57:08.997  1203  1252 E AndroidRuntime: Process: com.android.car, PID: 1203
05-07 10:57:08.997  1203  1252 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.util.SparseArray.size()' on a null object reference
05-07 10:57:08.997  1203  1252 E AndroidRuntime: 	at com.android.car.audio.CarAudioService.audioDevicesAdded(CarAudioService.java:3409)
05-07 10:57:08.997  1203  1252 E AndroidRuntime: 	at com.android.car.audio.CarAudioDeviceCallback.onAudioDevicesAdded(CarAudioDeviceCallback.java:42)
05-07 10:57:08.997  1203  1252 E AndroidRuntime: 	at android.media.AudioManager$Na[   80.248258] logd: Skipping entries from slow reader, pid 15152, fd 15, from LogBuffer::Prune()
tiveEventHandlerDelegate$1.handleMessage(AudioManager.java:10114)
05-07 10:57:08.997  1203  1252 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
05-07 10:57:08.997  1203  1252 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:232)
05-07 10:57:08.997  1203  1252 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:317)
05-07 10:57:08.997  1203  1252 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:68)
05-07 10:57:09.547     0     0 I binder  : undelivered TRANSACTION_COMPLETE
05-07 10:57:09.552     0     0 I binder  : undelivered transaction 14350, process died.
05-07 10:57:09.363  1287  1337 E AndroidRuntime: FATAL EXCEPTION: CarAudioService
05-07 10:57:09.363  1287  1337 E AndroidRuntime: Process: com.android.car, PID: 1287
05-07 10:57:09.363  1287  1337 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke virtual method 'int android.util.SparseArray.size()' on a null object reference
05-07 10:57:09.363  1287  1337 E AndroidRuntime: 	at com.android.car.audio.CarAudioService.audioDevicesAdded(CarAudioService.java:3409)
05-07 10:57:09.363  1287  1337 E AndroidRuntime: 	at com.android.car.audio.CarAudioDeviceCallback.onAudioDevicesAdded(CarAudioDeviceCallback.java:42)
05-07 10:57:09.363  1287  1337 E AndroidRuntime: 	at android.media.AudioManager$NativeEventHandlerDelegate$1.handleMessage(AudioManager.java:10114)
05-07 10:57:09.363  1287  1337 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
05-07 10:57:09.363  1287  1337 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:232)
05-07 10:57:09.363  1287  1337 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:317)
05-07 10:57:09.363  1287  1337 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:68)
05-07 10:57:09.727     0     0 I binder  : undelivered TRANSACTION_COMPLETE
05-07 10:57:09.732     0     0 I binder  : undelivered transaction 15336, process died.
05-07 10:57:09.738     0     0 I AidlLazyServ[   80.434999] logd: Skipping entries from slow reader, pid 15152, fd 15, from LogBuffer::Prune()
iceRegistrar: Process has 0 (of 1 available) client(s) in use after notification apexservice has clients: 0
05-07 10:57:09.751     0     0 I binder  : undelivered transaction 15364, process died.
05-07 10:57:09.757     0     0 I AidlLazyServiceRegistrar: Shutdown prevented by forcePersist override flag.
05-07 10:57:09.766     0     0 I binder  : undelivered TRANSACTION_COMPLETE
05-07 10:57:09.771   10:57:10.355  1600  1641 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:317)
05-07 10:57:10.355  1600  1641 E AndroidRuntime: 	at android.os.HandlerThread.run(HandlerThread.java:68)
05-07 10:57:10.752     0     0 I servicemanager: Could not find android.hardware.automotive.vehicle.IVehicle/default in the VINTF manifest. No alternative in[   81.076113] logd: Skipping entries from slow reader, pid 15152, fd 15, from LogBuffer::Prune()
stances declared in VINTF.
05
...

KonstaT, 2024-05-08:
> I have tried to connect to Pi via USB-A to USB-A cable, but do not see it under `adb devices` even when it's "booted". Looking at other issues, it seems like we need to use UART to get `logcat`. I [...]

You can revert https://github.com/raspberry-vanilla/android_device_brcm_rpi5/commit/71b154a8bb0facac24c8a27a0bfff93d3fe1aaf5 if you want to move the serial console to GPIO14/15. USB-A ports on Raspberry Pi are host mode only so it's not even technically possible to use ADB on those. You can use ADB on the USB-C port if you have outside power source (powered USB-C hub, 5V on the GPIO, etc).

No, I haven't tested all build variants on all tags and all devices. `tv` and `car` build targets are just side products of the tablet configuration as what AOSP is offering on these build targets doesn't provide much use outside development purposes.

I would assume Android Automotive is working on `android-14.0.0_r22` because of https://github.com/raspberry-vanilla/android_local_manifest/issues/33. AOSP 14 QPR2 caused all kinds of mayhem so `android-14.0.0_r22` is branched out in a separate manifest.
https://github.com/raspberry-vanilla/android_local_manifest/tree/android-14.0.0_r22

As for the issue itself, it looks like it's required to implement automotive audio control HAL and occupant audio zones (if using legacy routing policy doesn't solve this).

khvMX, 2024-05-10:
I have tried it and saw no difference in behavior. Unfortunately, I still could not get to `logcat`, so I did not proceed with debugging. I would expect that with this change it would trip elsewhere.
I'm building the `android-14.0.0_r22` now. Note that you need to use instructions from this specific [README.md](https://github.com/raspberry-vanilla/android_local_manifest/blob/android-14.0.0_r22/README.md)

khvMX, 2024-05-10:
I can confirm that `android-14.0.0_r22` is booting fine:
[image]
[image]

ramesh-m99, 2024-05-14:
> I can confirm that `android-14.0.0_r22` is booting fine: [...]

Thanks for update @khvMX . i tried with android-14.0.0_r34 branch but its not booting. 
when i am trying to switch the branch to android-14.0.0_r2( **### repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r22 --depth=1**) i am getting manifest default.xml not avaliable.

here is the complete log of swithching the branch: 
ramesh@ramesh: **repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r22 --depth=1**
repo: reusing existing repo client checkout in /home/manyantan/aosp14
fatal: manifest 'default.xml' not available
fatal: remove-project element specifies non-existent project: <remove-project name="platform/external/libdav1d"/>
================================================================================
Repo command failed: UpdateManifestError
	Unable to sync manifest default.xml

any help woule be appreciated

Thanks

khvMX, 2024-05-14:
@ramesh-m99 to avoid all sort of weird unexpected issues, I usually start from scratch in another directory. For instance, I started with `~/aosp/rpi-android-14.0.0_r34/`, and after that I created `~/aosp/rpi-android-14.0.0_r22/` where I'd execute `repo init -u https://android.googlesource.com/platform/manifest -b android-14.0.0_r22 --depth=1` followed by instructions from [this README](https://github.com/raspberry-vanilla/android_local_manifest/blob/android-14.0.0_r22/README.md).
Take care not to run out of space, at least remove `~/aosp/rpi-android-14.0.0_r34/out`, or whole `~/aosp/rpi-android-14.0.0_r34/` directory.

Samaelson, 2024-05-15:
@khvMX : Based on the diff of CarAudioService.java (r22<->r34) may be flag name changed and your proposed <bool name="audioUseDynamicRouting">false</bool> does not fit for r34 any more?

[image]

Samaelson, 2024-05-15:
runInLegacyMode obviously depends on mUseDynamicRouting and mUseCoreAusioRouting

[image]

KonstaT, 2024-09-07:
This was probably fixed AOSP upstream at some point. Android Automotive builds from both `android-14.0.0_r67` and `android-15.0.0_r1` branches boot to UI successfully.

closed 2024-09-07 by KonstaT