raspberry-vanilla/android_local_manifest #85: Android-14  raspberry pi 4  Struck at android logo/splash screen

devalapalli, 2024-05-08:
Hello @KonstaT ,

We tried booting android 14 stock images which are downloaded from below mentioned link(https://dlupload.com/filedetail/1138405769)
we are tried booting this from sdcard but its not going through(Zygore service is crashing).

Environment:

Android version: 14 
Board :  raspberry pi 4
Booting mode : Sdcard 

Please find attached logs.
[aosp_14_rp4_logs_sdcard.txt](https://github.com/raspberry-vanilla/android_local_manifest/files/15248887/aosp_14_rp4_logs_sdcard.txt)
please let me know if you need any information. 

Thanks

KonstaT, 2024-05-08:
This issue tracker is intended for people building from Raspberry Vanilla source code. You can find discussion for my personal builds on each release page.

Reading the FAQ on each release page is good place to start. In any case impossible to say what is the issue without seeing `logcat` output.

closed 2024-05-08 by KonstaT

ramesh-m99, 2024-05-14:
Thanks for response