raspberry-vanilla/android_local_manifest #87: Car on Pi5 same issues

striork, 2024-05-13:
ok, so  I just made this today.
android-14.0.0_r34 
lunch aosp_rpi5_car-ap1a-userdebug
###  Have the same issues. 
 
it boots up nicely,
Black homescreen

### interfaces that will open

- climate control interface
- backlight slider
- wifi - network internet settings prompt
- bluetooth - bluetooth settings prompt
- notification panel does pull down

if I touch the driver icon ion the top right corner the "feature not available while driving" toast does flash on screen

KonstaT, 2024-05-13:
If this is a "same issue", why is there a new issue opened?

Closing as a duplicate of https://github.com/raspberry-vanilla/android_local_manifest/issues/83.

closed 2024-05-13 by KonstaT

striork, 2024-05-14:
If this is a "same issue", why is there a new issue opened?
Because it's the same issue on a new branch/commit... but I could have
added it as a comment