raspberry-vanilla/android_local_manifest #88: raspi5 adb not working over USB

JitendraKarma, 2024-05-13:
Hi,

I falshed Android-14 in **raspi5** board and connected type-C cable to provide power the board and use ADB. I am not able to use ADB over USB in **raspi5** board. But over the wifi, I can use through adb connect <ip_address>

Do we have any solution or workaround for using adb over USB in raspi5 ?

**Note**:  Same code base I built for **raspi4** and able to use over USB and wifi both.

Thanks & Regards,
Jitendra Karma

KonstaT, 2024-05-13:
This is supported since January.
https://github.com/raspberry-vanilla/android_local_manifest/issues/51#issuecomment-1877361879

You'll likely need to update EEPOM. What version do you have (`vcgencmd bootloader_version`)?

JitendraKarma, 2024-05-15:
Thank you for quick response and solution.

Here is update

Output of `vcgencmd bootloader_version`

2024/02/16 15:28:41
version 4c845bd37c8a7c7ff79173cdc50dd3facf63996f (release)
timestamp 1708097321
update-time 1710515530
capabilities 0x0000007f

I am able to use adb over USB and here are steps which I followed

1. Download [raspi OS](https://www.raspberrypi.com/software/operating-systems/), after flashing , connected into raspi 5 board
2. Connected to wifi network to update EEPROM
3. Run `sudo rpi-eeprom-update ` and reboot the device

After that I flashed Android-14 image and start using adb over USB.

closed 2024-05-15 by KonstaT