raspberry-vanilla/android_local_manifest #90: [rpi4-14os][Linux version 6.1.80-g9cd07cf52250-dirty-v8] Kernel panic - not syncing: EXT4-fs (device mmcblk0p4): panic forced after error

thangarajpvaleo, 2024-05-18:
JBD2: I/O error when updating journal superblock for mmcblk0p4-8.
[  236.293983] EXT4-fs error (device mmcblk0p4): ext4_journal_check_start:83: comm kworker/u8:3: Detected aborted journal
[  236.305884] EXT4-fs (mmcblk0p4): I/O error while writing superblock
[  236.312422] **Kernel panic - not syncing: EXT4-fs (device mmcblk0p4): panic forced after error**
[  236.320990] CPU: 1 PID: 52 Comm: kworker/u8:3 Not tainted 6.1.80-g9cd07cf52250-dirty-v8 #1
[  236.329371] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
[  236.335284] Workqueue: events_unbound quota_release_workfn
**[  236.339815] EXT4-fs error (device mmcblk0p4): __ext4_find_entry:1684: inode #131113: comm com.android.car: reading directory lblock 0
[  236.339846] EXT4-fs (mmcblk0p4): I/O error while writing superblock
[  236.359366] Call trace:
[  236.361838]  dump_backtrace+0x110/0x130
[  236.365725]  show_stack+0x28/0x38
[  236.369079]  dump_stack_lvl+0x6c/0x88
[  236.372788]  dump_stack+0x20/0x44
[  236.376141]  panic+0x15c/0x348
[  236.379233]  ext4_handle_error+0x1fc/0x220
[  236.383385]  __ext4_error+0xec/0x200
[  236.387005]  __ext4_journal_start_sb+0xc8/0x1cc
[  236.391595]  ext4_release_dquot+0x78/0xc4
[  236.395656]  quota_release_workfn+0x168/0x228
[  236.400067]  process_one_work+0x1b8/0x3d4
[  236.404129]  worker_thread+0x26c/0x444
[  236.407925]  kthread+0xfc/0x1c8
[  236.411102]  ret_from_fork+0x10/0x20**
[  236.414723] SMP: stopping secondary CPUs
[  236.418697] Kernel Offset: 0x15b2000000 from 0xffffffc008000000
[  236.424693] PHYS_OFFSET: 0x0
[  236.427605] CPU features: 0x00,00040000,2013c080,0000421b
[  236.433074] Memory Limit: none
**[  236.436166] Rebooting in 5 seconds..**
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.80-g9cd07cf52250-dirty-v8 (build-user@build-host) (Android (10087095, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.2) #1 SMP PREEMPT Thu May  2 13:21:20 UTC 2024
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.5
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000ec00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000ec00000..0x000000002ebfffff ( 524288 KB ) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003f129e00..0x000000003f12a1ff ( 1 KB ) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003f12a240..0x000000003f12a275 ( 0 KB ) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s84392 r8192 d30296 u122880
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=D8:3A:DD:26:4C:A3 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 no_console_suspend root=/dev/ram0 rootwait androidboot.hardware=rpi4 androidboot.selinux=permissive androidboot.serialno=1000000078fa0141 androidboot.btmacaddr=11:22:78:fa:01:41
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[... 182 lines cut ...]
[    0.619607] usbcore: registered new interface driver zaurus
[    0.619663] usbcore: registered new interface driver cdc_ncm
[    0.619704] usbcore: registered new interface driver r8153_ecm
[    0.665769] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    0.665798] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    0.666344] xhci_hcd 0000:01:00.0: hcc params 0x002841eb hci version 0x100 quirks 0x0b00040000000890
[    0.667308] xhci_hcd 0000:01:00.0: xHCI Host Controller
[    0.667325] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[    0.667341] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
[    0.667584] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    0.667598] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.667608] usb usb1: Product: xHCI Host Controller
[    0.667616] usb usb1: Manufacturer: Linux 6.1.80-g9cd07cf52250-dirty-v8 xhci-hcd
[    0.667624] usb usb1: SerialNumber: 0000:01:00.0
[    0.668240] hub 1-0:1.0: USB hub found
[    0.668282] hub 1-0:1.0: 1 port detected

seen reboot issue with frequent intervals, the same has been seen from AOSP14-20240508-KonstaKANG-rpi4-6.6-kernel also. Please check and help to resolve.

also tried pinging using VLAN IP, error [connect: network unreachable] CONFIG_VLAN_8021Q=y

tcpdump showing test device[192.168.XX.XX] -> rpi4[192.168.YY.YY] receiving packets but no reply from rpi4.
rpi4->test device vlan ping not working

03:11:08.789131 IP 192.168.XX.XX > 192.168.YY.YY: ICMP echo request, id 28852, seq 94, length 64

thangarajpvaleo, 2024-05-18:
@https://github.com/raspberry-vanilla/android_local_manifest/issues/78

In rpi4-13os, also the same behavior for vlan ping. Please help to fix the issue.

KonstaT, 2024-06-03:
Isn't this rather two separate issues?

`EXT4-fs error (device mmcblk0p4)` would rather indicate a corrupted storage device. Try using a different sdcard.

`6.1.80-g9cd07cf52250-dirty-v8` is not part of Raspberry Vanilla and I have no idea what changes you've made. VLAN implementation was previously discussed at https://github.com/raspberry-vanilla/android_local_manifest/issues/78.

closed 2024-06-03 by KonstaT