raspberry-vanilla/android_local_manifest #94: Looking to attach external USB Bluetooth controller to Rpi (running Android-14) as host BT adapter

skanda-coder, 2024-05-29:
Hello,

I am working on Raspberry Pi 5 with Android-14 flashed on it. By default, the on-device BT chip is added as the Bluetooth adapter :
01-01 00:00:12.375     0     0 I Bluetooth: hci0: BCM: chip id 107
01-01 00:00:12.376     0     0 I pci 0000: 00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
01-01 00:00:12.380     0     0 I Bluetooth: hci0: BCM: features 0x2f
01-01 00:00:12.393     0     0 I Bluetooth: hci0: BCM4345C0
01-01 00:00:12.397     0     0 I Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
01-01 00:00:12.403     0     0 I Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.hcd' Patch

Now I am looking to replace or add another external device as a Bluetooth adapter (Nordic nRF52840 configured as a Bluetooth controller and attached to Rpi5 via USB). Right now this is detected as a USB device:

01-01 00:00:13.444     0     0 I usb 3-1 : New USB device found, idVendor=2fe3, idProduct=000b, bcdDevice= 3.05
01-01 00:00:13.452     0     0 I usb 3-1 : New USB device strings: Mfr=1, Product=2, SerialNumber=3
01-01 00:00:13.452     0     0 I usb 3-1 : Product: USB-DEV
01-01 00:00:13.467     0     0 I usb 3-1 : Manufacturer: ZEPHYR

and hence the new connection is managed by the USB host manager : 
05-29 06:23:44.634   725   944 D UsbHostManager: USB device attached: vidpid 2fe3:000b mfg/product/ver/serial ZEPHYR/USB-DEV/3.05/24FBA3DAF54721C1 hasAudio/HID/Storage: false/false/false
05-29 06:23:44.641   725   944 D UsbHostManager: Added device UsbDevice[mName=/dev/bus/usb/003/002,mVendorId=12259,mProductId=11,mClass=0,mSubclass=0,mProtocol=0,mManufacturerName=ZEPHYR,mProductName=USB-DEV,mVersion=3.05,mSerialNumberReader=com.android.server.usb.UsbSerialReader@c2fd7ab, mHasAudioPlayback=false, mHasAudioCapture=false, mHasMidi=false, mHasVideoCapture=false, mHasVideoPlayback=false, mConfigurations=[

But the host Rpi5 doesn't detect another Bluetooth adapter.

In linux, I see 'bluetoothctrl' or 'hciattach' commands, etc that can control the attached Bluetooth adapters.

However in Android I came across a configuration file "packages/modules/Bluetooth/system/conf/bt_did.conf" that seems to manage handling of new Bluetooth devices and register with the Bluetooth Service Discovery Protocol (SDP). But this is not enough to configure the new adapter.

Is there any better way of adding a USB Bluetooth controller as the host Bluetooth adapter in Android?

skanda-coder, 2024-06-02:
The kernel modifications were needed to support USB bluetooth device. This is resolved now.
Thanks

KonstaT, 2024-06-03:
It might be useful for other people to document what was needed to make this work.

Closing as complete in any case.

closed 2024-06-03 by KonstaT

hmtheboy154, 2025-05-09:
> The kernel modifications were needed to support USB bluetooth device. This is resolved now. Thanks

Hi, may I ask. Did you ended up resolve this by hacking the kernel like what Project Celadon did ?
https://github.com/projectceladon/linux-intel-lts2023/commit/abb10755f082c5534e132dcbc3640921e4e0be34