raspberry-vanilla/android_local_manifest #96: android-14.0.0_r22   #### failed to build some targets

striork, 2024-05-31:
`[  1% 2049/159672] bc: libclcore.bc <= frameworks/rs/driver/runtime/rs_core.c
FAILED: out/target/product/rpi5/obj/RENDERSCRIPT_BITCODE/libclcore.bc_intermediates/rs_core.bc
/bin/bash -c "PWD=/proc/self/cwd prebuilts/clang/host/linux-x86/clang-3289846/bin/clang -Iframeworks/rs/script_api/include -Iexternal/clang/lib/Headers -MD -DRS_VERSION=24 -std=c99 -c -O3 -fno-builtin -emit-llvm -target renderscript64-linux-android -fsigned-char  -Wno-deprecated -Werror  -Werror -Wall -Wextra -Iframeworks/rs/cpu_ref -DRS_DECLARE_EXPIRED_APIS  -DARCH_ARM64_HAVE_NEON -x renderscript frameworks/rs/driver/runtime/rs_core.c -o out/target/product/rpi5/obj/RENDERSCRIPT_BITCODE/libclcore.bc_intermediates/rs_core.bc"
prebuilts/clang/host/linux-x86/clang-3289846/bin/clang.real: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory`

16:58:01 ninja failed with: exit status 1

#### failed to build some targets (11 seconds) ####

Samaelson, 2024-05-31:
Hello,

Do you have libncurses5 installed on your build system?

sudo apt-get install libncurses5

striork, 2024-05-31:
`find /usr/lib/ -name *ncurses*`
/usr/lib/x86_64-linux-gnu/libncurses.so.6.4
/usr/lib/x86_64-linux-gnu/libncurses.so
/usr/lib/x86_64-linux-gnu/libncurses.a
/usr/lib/x86_64-linux-gnu/libncursesw.so.6.4
/usr/lib/x86_64-linux-gnu/libncursesw.so.6
/usr/lib/x86_64-linux-gnu/libncurses.so.6
/usr/lib/x86_64-linux-gnu/libncursesw.a
/usr/lib/x86_64-linux-gnu/libncursesw.so
/usr/lib/x86_64-linux-gnu/vlc/plugins/gui/libncurses_plugin.so
/usr/lib/x86_64-linux-gnu/libncurses++w.a
/usr/lib/x86_64-linux-gnu/libncurses++.a
/usr/lib/x86_64-linux-gnu/pkgconfig/ncursesw.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/ncurses++.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/ncurses.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/ncurses++w.pc
/usr/lib/valgrind/ncurses.supp

`sudo apt-get install libncurses5`
Results in:
`E: Unable to locate package libncurses5`

striork, 2024-06-01:
> Hello,
> 
> Do you have libncurses5 installed on your build system?
> 
> sudo apt-get install libncurses5

No sir.
No I do not.
To make matters worse, as I'm sure you are well aware, I cannot install it with apt...  or any other package manager for that matter.

Samaelson, 2024-06-03:
May be, this is helpfull:
https://stackoverflow.com/questions/17005654/error-while-loading-shared-libraries-libncurses-so-5

KonstaT, 2024-06-03:
Please follow the instruction on how to establish [Android build environment](https://source.android.com/setup/initializing).

Generally speaking, issues with your specific AOSP build environment are not due to changes in Raspberry Vanilla project.

closed 2024-06-03 by KonstaT

striork, 2024-06-05:
yes, I appreciate the assistance. It pointed me in the right direction. I did find answers regarding the libncurses lib on stack that straightened be out.  FYI there is no longer reference to this library on the aosp build env page or the raspberry-vanilla build instructions.  
     thanks again