raspberry-vanilla/android_local_manifest #98: Is Hfp working on android 14 rev34 on RPI4.

dhondgeswayam, 2024-06-04:
In AOSP 14 rev34 we have a Dialer application which is not opening, stating "no device is connected ,Connect to Bluetooth"  . But in the bluetooth settings I have  paired and connected a device with all the necessary permissions granted. From the logs it seems the hfp profile is not working. Could you please confirm this is the case and if this is the case could you please help to unable the same.
Attaching the log file :
[LOGS.txt](https://github.com/user-attachments/files/15568650/LOGS.txt)

KonstaT, 2024-06-06:
What are you exactly trying to achieve and why? What device and build variant are you targeting?

There isn't any Dialer application in regular tablet UI or Android TV build variants. As discussed in https://github.com/raspberry-vanilla/android_local_manifest/issues/83 Android Automotive build variant isn't booting to UI on android-14.0.0_r34.

dhondgeswayam, 2024-06-09:
I have built android automotive 14 rev34 for raspberry pi 4 device with build target aosp_rpi4_car-ap1a-userdebug.
The Phone application(CarDialer) present is not able to detect any connected mobile device via Bluetooth. In the bluetooth settings I have paired device showing with connected status. Please let me know if the build supports hfp bluwtooth profile.

KonstaT, 2024-06-12:
As discussed in https://github.com/raspberry-vanilla/android_local_manifest/issues/83 Android Automotive build variant isn't booting to UI on android-14.0.0_r34. I have no idea how you could launch Phone application in that case. You're welcome to send pull requests on the changes you've made to boot `aosp_rpi4_car-ap1a-userdebug` on `android-14.0.0_r34`.

https://github.com/raspberry-vanilla/android_device_brcm_rpi4/blob/android-14.0/aosp_rpi4_car.mk#L27 would indicate HFP is not enabled on automotive build target. Again, you're welcome to send pull requests in this regard.

closed 2024-06-12 by KonstaT