raspberry-vanilla/android_local_manifest #99: USB Audio Input not working on RPI5

PiratedKukreja, 2024-06-07:
I'm trying to use USB microphone and USB sound card but it doesn't seem to work. I used wireless adb to look into alsa files and it does detect both the devices as capture devices but whenever I try to use any application or website to record sound or join calls on skype there is no input from the microphone.

jsln, 2024-06-07:
Can you record audio with the alsa utilities?

I have a similar problem to you, I cannot get audio input from my HifiBerry DAC+ ADC Pro card with the applications, however when I try to use the alsa commands I can record mic input, with the caveat that I cannot actually use my microphone, it seems alsa does not let me configure the gain (I cannot find that control), but plugging an mp3 player in the mic input works, and I can play back the recorded audio:

alsa_arecord -f cd -D plughw:2,0 -d 5 rec.wav

also_aplay -D plughw:2,0 rec.wav

KonstaT, 2024-06-07:
@PiratedKukreja Seems rather an application or user/hardware issue. I just tested USB audio input with Logitech USB webcam and it works fine.

@jsin Not quite the same issue. USB audio uses generic Android USB audio HAL (https://source.android.com/docs/core/audio/usb, https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-14.0/device.mk#L28, https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-14.0/audio/audio_policy_configuration.xml#L99). USB audio output/input works exactly the same any other Android device.

Audio DACs on the other hand use device specific audio HAL (https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-14.0/audio/audio_hw.c) which doesn't support audio input in the first place since Raspberry Pi doesn't have any built-in hardware for audio input. It's rather simple to implement as I've done it to support I2S MEMS mics but can't be included by default.

Things working on Linux side (i.e. record using ALSA tools) doesn't really implicate much on Android side.

closed 2024-06-07 by KonstaT

JSin, 2024-06-07:
@jsln pinging you here due to a mistyping of an "i" instead of an "l"

KonstaT, 2024-06-07:
@JSin Oops, sorry.

KonstaT, 2024-06-07:
> Can you point me at your I2S MEMS mic implementation? Perhaps I can sort something out for the hifiberry card.

```
diff --git a/audio/audio_hw.c b/audio/audio_hw.c
index 82f8167..6f50f2a 100644
--- a/audio/audio_hw.c
+++ b/audio/audio_hw.c
@@ -50,14 +50,22 @@
 /* number of frames per short period (low latency) */
 #define PERIOD_SIZE (CODEC_BASE_FRAME_COUNT * PERIOD_MULTIPLIER)
 /* number of pseudo periods for low latency playback */
-#define PLAYBACK_PERIOD_COUNT 4
-#define PLAYBACK_PERIOD_START_THRESHOLD 2
+#define PERIOD_COUNT 4
+#define PERIOD_START_THRESHOLD 2
 #define CODEC_SAMPLING_RATE 48000
 #define CHANNEL_STEREO 2
-#define MIN_WRITE_SLEEP_US      5000
 
-struct stub_stream_in {
+struct alsa_stream_in {
     struct audio_stream_in stream;
+
+    pthread_mutex_t lock;   /* see note below on mutex acquisition order */
+    struct pcm_config config;
+    struct pcm *pcm;
+    bool unavailable;
+    int standby;
+    struct alsa_audio_device *dev;
+    int read_threshold;
+    unsigned int read;
 };
 
 struct alsa_audio_device {
@@ -97,6 +105,20 @@ static int get_pcm_device()
     return atoi(device);
 }
 
+static int get_pcm_in_card()
+{
+    char card[PROPERTY_VALUE_MAX];
+    property_get("persist.audio.pcm.in.card", card, "0");
+    return atoi(card);
+}
+
+static int get_pcm_in_device()
+{
+    char device[PROPERTY_VALUE_MAX];
+    property_get("persist.audio.pcm.in.device", device, "0");
+    return atoi(device);
+}
+
 /* must be called with hw device and output stream mutexes locked */
 static int start_output_stream(struct alsa_stream_out *out)
 {
@@ -108,8 +130,8 @@ static int start_output_stream(struct alsa_stream_out *out)
     /* default to low power: will be corrected in out_write if necessary before first write to
      * tinyalsa.
      */
-    out->write_threshold = PLAYBACK_PERIOD_COUNT * PERIOD_SIZE;
-    out->config.start_threshold = PLAYBACK_PERIOD_START_THRESHOLD * PERIOD_SIZE;
+    out->write_threshold = PERIOD_COUNT * PERIOD_SIZE;
+    out->config.start_threshold = PERIOD_START_THRESHOLD * PERIOD_SIZE;
     out->config.avail_min = PERIOD_SIZE;
 
     out->pcm = pcm_open(get_pcm_card(), get_pcm_device(), PCM_OUT | PCM_MMAP | PCM_NOIRQ | PCM_MONOTONIC, &out->config);
@@ -134,13 +156,13 @@ static uint32_t out_get_sample_rate(const struct audio_stream *stream)
 
 static int out_set_sample_rate(struct audio_stream *stream, uint32_t rate)
 {
[... 341 lines cut ...]
-    return;
+    free(stream);
 }
 
 static int adev_dump(const audio_hw_device_t *device, int fd)
diff --git a/audio/audio_policy_configuration.xml b/audio/audio_policy_configuration.xml
index b2647e8..337df49 100644
--- a/audio/audio_policy_configuration.xml
+++ b/audio/audio_policy_configuration.xml
@@ -63,12 +63,12 @@
                 </devicePort>
                 <devicePort tagName="Built-In Mic" type="AUDIO_DEVICE_IN_BUILTIN_MIC" role="source">
                     <profile name="" format="AUDIO_FORMAT_PCM_16_BIT"
-                             samplingRates="8000 11025 12000 16000 22050 24000 32000 44100 48000"
+                             samplingRates="48000"
                              channelMasks="AUDIO_CHANNEL_IN_MONO AUDIO_CHANNEL_IN_STEREO"/>
                 </devicePort>
                 <devicePort tagName="Wired Headset Mic" type="AUDIO_DEVICE_IN_WIRED_HEADSET" role="source">
                     <profile name="" format="AUDIO_FORMAT_PCM_16_BIT"
-                             samplingRates="8000 11025 12000 16000 22050 24000 32000 44100 48000"
+                             samplingRates="48000"
                              channelMasks="AUDIO_CHANNEL_IN_MONO AUDIO_CHANNEL_IN_STEREO"/>
                 </devicePort>
                 <devicePort tagName="BT SCO Headset Mic" type="AUDIO_DEVICE_IN_BLUETOOTH_SCO_HEADSET" role="source">
```

PiratedKukreja, 2024-06-08:
Hello @jsln I didn't try recording audio using alsa utilities. 

@KonstaT I tried using a USB Logitech web cam () and I still can't get this to work, devices i've tried until now are a generic sound card, a usb microphone, a Logitech web cam, bluetooth headsets. Software i've tried using are edge browser (mictests.com), Discord, Skype and maybe a few random sound recorders. 

PS: IDK if this is helpful but I've an orange pi too and all the above mentioned devices seem to work with it.