raspberry-vanilla/android_local_manifest #101: Hyperpixel 4 display.

steveiliop56, 2024-06-16:
Hello,

Is there any chance for the hyperpixel 4 inch display to be supported?

KonstaT, 2024-06-17:
What steps have you made to test the display? Please provide appropriate logs (`logcat`, `dmesg`) if the display is not working.

There has been reports that DPI displays are working but I don't have any hardware to test myself. You would configure the display the same way as any other OS on the Pi (https://github.com/pimoroni/hyperpixel4/issues/177) with the only Android specific configuration being changing the forced display resolution (https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-14.0/vendor.prop#L44).

closed 2024-06-17 by KonstaT

steveiliop56, 2024-06-19:
Ohh I didn't test I just asked if there are any drivers for it. I will try to use the 7inch option and see if it works. I will reopen the issue if it doesn't.