raspberry-vanilla/android_local_manifest #103: HifiBerry in RPi5 does not support FAST TRACK audio mode

jsln, 2024-06-20:
Hi @KonstaT ,

it seems FAST TRACK audio mode is not supported with HifiBerry in RPi5 (see https://github.com/google/oboe/issues/2040). Could you confirm this? Any hints at how these drivers could be ported from linux to AOSP?

Do you know if FAST TRACK is implemented on a RPi4?

Thanks!

KonstaT, 2024-06-20:
The documentation isn't exactly clear what needs to be implemented to support low latency audio on Android.
https://developer.android.com/ndk/guides/audio/audio-latency
https://source.android.com/docs/core/audio/latency/design

You could try adding `AUDIO_OUTPUT_FLAG_FAST` flag to the output configuration (https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-14.0/audio/audio_policy_configuration.xml#L22 ) and/or adding `android.hardware.audio.low_latency` permissions file to the image.

Audio configuration on Raspberry Pi is slightly complicated since we're using three separate audio HALs. One that's used for 3.5mm audio jack and audio DAC, another one for HDMI output devices, and a generic Android USB audio HAL for external audio devices (https://github.com/raspberry-vanilla/android_device_brcm_rpi5/tree/android-14.0/audio, https://github.com/raspberry-vanilla/android_device_brcm_rpi5/blob/android-14.0/device.mk#L20-L73).

jsln, 2024-06-21:
Thanks, @KonstaT , I will try with the output configuration change you mention.

KonstaT, 2024-08-12:
Closing as there has been no interaction in over a month.

closed 2024-08-12 by KonstaT