raspberry-vanilla/android_local_manifest #107: out/soong/hiddenapi/hiddenapi-flags.csv is empty

tzjb, 2024-07-03:
After a successful build, I noticed that `out/soong/hiddenapi/hiddenapi-flags.csv` has 0 byte, and in the running image the hidden APIs can be accessed without disabling [hidden API policy](https://developer.android.com/guide/app-compatibility/restrictions-non-sdk-interfaces).

I don't know if this is by design in this project, or an actual issue. The Google AOSP enforces this policy by default. Is there a way to enable this during build time? Below is what I used for building the image. Thanks!

```
. build/envsetup.sh
lunch aosp_rpi5-ap2a-userdebug
make bootimage systemimage vendorimage -j$(nproc)
./rpi5-mkimg.sh
```

KonstaT, 2024-07-03:
Raspberry Vanilla doesn't change AOSP behaviour in this regard.

Maybe hidden API policy isn't enforced on userdebug build type? Maybe `out/soong/hiddenapi/hiddenapi-flags.csv` doesn't get built when you're building partition images? You're welcome to investigate.

KonstaT, 2024-08-12:
Closing as there has been no interaction in over a month.

closed 2024-08-12 by KonstaT

hanwenhao77, 2025-01-16:
You can run "m out/soong/hiddenapi/hiddenapi-flags.csv" to build it.