raspberry-vanilla/android_local_manifest #109: get the Raspberry Pi into fastboot mode

yangfan01-spec, 2024-07-09:
Hello,
I now hope to discover the Raspberry Pi device using fastboot devices. How can I get the Raspberry Pi into fastboot mode?

Currently, after executing adb devices in Windows, the Raspberry Pi device can be recognized. However, after executing fastboot devices, 

the number of displayed devices is 0. Do you have any good suggestions?

KonstaT, 2024-07-14:
fastboot is a protocol used to interact with the device when booted to Android bootloader. Since Raspberry Pi is not an Android device to begin with, it doesn't have an Android bootloader (i.e. no fastboot).

You could look into implementing e.g. U-Boot but that is not currently in the scope of Raspberry Vanilla project.

closed 2024-07-14 by KonstaT

yangfan01-spec, 2024-07-16:
Thank you very much for your reply, it was very helpful to me!
I will try using U-Boot to solve my problem.

yangfan01-spec, 2024-07-16:
I would appreciate your guidance on whether there is any method to execute 'adb remount' and gain root access to the /system folder without utilizing the bootloader. Thank you for your assistance.

KonstaT, 2024-07-16:
https://github.com/raspberry-vanilla/android_local_manifest/issues/47#issuecomment-1829888921